ghdl/ghdl #1462: Use AVX instructions to speed up simmulation

go2sh, 2020-09-14:
Hi all,

I did some expirements with AVX instructions lately and used vhdl std_logic types as idea. I implemented all standard operations specified in the std_logic_1164 package and gained a 5-6x speed up comared to a naive LookupTable approach. The same could be done for numeric_std types and so on

It would be nice to have such a possibility in ghdl. My idea would be to emit ASM code directly whenever a certain stl_logic_vector (or what ever) operation is called. Do you think that is possible with the current ghdl design, @tgingold ? I tried to find the code involved generating/emitting the stuf to llvm/gcc.

eine, 2020-09-14:
@go2sh, I believe the main problem is that GHDL works on multiple platforms (it is used frequently on amd64, armv7, aarch64, ppc64...): Hence, any enhancement in this regard would imply conditional implementation. E.g., for aarch64, NEON might be used instead.

On the other hand, one of the main features of LLVM should be to allow optimization without actually implementing all the details. See https://github.com/ghdl/ghdl/issues/667#issuecomment-428429616. Yet, I don't know if any optimization is good enough for this use case.

tgingold, 2020-09-14:
That's very interesting!

You can replace std_logic_1164-body with your AVX implementation directly provided the interface and call convention is the same.
That could be something that is easy to test.

Is your implementation available on a public repo ?

go2sh, 2020-09-14:
I can upload it on github. It's written in c++, but the functions mostly use AVX intrinsics. I'll have a look into the vhdl-body call conventions and check if it fits. 

One thing, I maybe did change though, is the representation of std_logic. I split of the "logic level" from the state. This makes it easy to use AVX "or" and "and" instructions as the "logic bits" for "1" "0" "H" and "L" are always at the same position.

@eine I did some test with clang and llvm but they failed to optimize the code to AVX instructions. I would definitely use something like this as an opt in feature as it is heavily dependent on cpu and os support. So there should be any issues with other cpus/oss. Optimization for ARM neon might be similar to implement.

go2sh, 2020-09-14:
@tgingold I had a look at the llvm IR and i'am not sure about "array type". The type looks like
`{ [0 x i8]*, { { i32, i32, i8, i32 } }* }`. Do you know the elements of the struct after the array?

tgingold, 2020-09-15:
> One thing, I maybe did change though, is the representation of std_logic. I split of the "logic level" from the state. This makes it easy to use AVX "or" and "and" instructions as the "logic bits" for "1" "0" "H" and "L" are always at the same position.

Changing the representation will make some (rarely used) operations more difficult (like 'Pos or comparisons).

tgingold, 2020-09-15:
> @tgingold I had a look at the llvm IR and i'am not sure about "array type". The type looks like
> { [0 x i8]*, { { i32, i32, i8, i32 } }* }. Do you know the elements of the struct after the array?

The first field is a pointer to the array of std_logic, with one byte per element.

The second field contains the bounds: left limit, right limit, direction and length.

marzoul, 2021-09-07:
Hi,
I was following this issue with delighted expectations about optimized simulation code and great speedup ;-)
The issue was just closed, but I didn't see result updates or commits, did I miss something ?
So is the LLVM backend able to better use per-CPU instructions, is there any hope for mcode support, do we have to compile with manually-provided implementation of compiled standard libs ?
Regards,
Adrien

tgingold, 2021-09-07:
Ok, I will re-open it.