ghdl/ghdl #1464: GHDL Bug Occured when compiling OSVVM 2020.08 libraries

eabtioglu, 2020-09-15:
**Description**
I've encountered a GHDL bug when compiling OSVVM 2020.08.

**How to reproduce?**
I followed the compilation order in `OsvvmLibraries.pro`. You can see the order below:
```
include ./osvvm/osvvm.pro
include ./common/common.pro
include ./Axi4/common/common.pro
include ./Axi4Lite/Axi4Lite.pro
include ./Axi4/Axi4.pro
include ./AxiStream/AxiStream.pro
include ./UART/UART.pro
```
I've succesfully compiled upto `Axi4.pro`. When i tried to compile files stated in the `Axi4.pro`, i got the following error:
```
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4MasterComponentPkg.vhd
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4ResponderComponentPkg.vhd
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4MemoryComponentPkg.vhd
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4MonitorComponentPkg.vhd
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4Context.vhd
Analyzing package: /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4Master.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/emrah/Desktop/shell/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 --ghdl--work=osvvm_axi4 --ghdl--std=08 --ghdl-fsynopsys --ghdl--no-vital-checks --ghdl--warn-binding --ghdl--mb-comments --ghdl-fexplicit --ghdl-frelaxed-rules -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o Axi4Master.s /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4Master.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap6.adb:1155
Call stack traceback locations:
0x78aa7c 0x78a77f 0x78a8fa 0x7a1ab1 0x7ab975 0x7ab565 0x7b4cf4 0x79d442 0x77bb73 0x76da9b 0x76b628 0x680f48 0xc2024d 0x6774a5 0x68602e 0x7dbb89 0x6750cd 0x7f238a14e0b1 0x67c5dc 0xfffffffffffffffe
******************************************************************

Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : trans-chap6.adb:1155
Call stack traceback locations:
0x78aa7c 0x78a77f 0x78a8fa 0x7a1ab1 0x7ab975 0x7ab565 0x7b4cf4 0x79d442 0x77bb73 0x76da9b 0x76b628 0x680f48 0xc2024d 0x6774a5 0x68602e 0x7dbb89 0x6750cd 0x7f238a14e0b1 0x67c5dc 0xfffffffffffffffe
ghdl:error: compilation error
```
i used the command below
```
ghdl -a --work=osvvm_axi4 --std=08 -fsynopsys --no-vital-checks --warn-binding --mb-comments -fexplicit -frelaxed-rules /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4Master.vhd
```

**Context**
Please, provide the following information:

- OS: Kubuntu 20.04
- Origin:
  - [ ] Package manager: `v0.37 with GCC backend`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/emrah/Desktop/shell/
Command line:
/usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 --ghdl--work=osvvm_axi4 --ghdl--std=08 --ghdl-fsynopsys --ghdl--no-vital-checks --ghdl--warn-binding --ghdl--mb-comments --ghdl-fexplicit --ghdl-frelaxed-rules -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -quiet -o Axi4Master.s /home/emrah/git/OsvvmLibraries/AXI4/Axi4/src/Axi4Master.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap6.adb:1155
Call stack traceback locations:
0x78aa7c 0x78a77f 0x78a8fa 0x7a1ab1 0x7ab975 0x7ab565 0x7b4cf4 0x79d442 0x77bb73 0x76da9b 0x76b628 0x680f48 0xc2024d 0x6774a5 0x68602e 0x7dbb89 0x6750cd 0x7f238a14e0b1 0x67c5dc 0xfffffffffffffffe
******************************************************************

Execution of /usr/local/libexec/gcc/x86_64-pc-linux-gnu/10.2.0/ghdl1 terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : trans-chap6.adb:1155
Call stack traceback locations:
0x78aa7c 0x78a77f 0x78a8fa 0x7a1ab1 0x7ab975 0x7ab565 0x7b4cf4 0x79d442 0x77bb73 0x76da9b 0x76b628 0x680f48 0xc2024d 0x6774a5 0x68602e 0x7dbb89 0x6750cd 0x7f238a14e0b1 0x67c5dc 0xfffffffffffffffe
ghdl:error: compilation error
```

timothystotts, 2020-12-19:
I had the same issue with compiling OSVVM 2020.10 with GHDL nightly build.

`$ ghdl --version
GHDL 1.0-dev () [Dunoon edition]
 Compiled with GNAT Version: 7.5.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.`

Bash transcript when bug occurred:

`./ghdl-run.sh 
+ rm -f '*.o'
+ rm -f work-obj08.cf osvvm-obj08.cf osvvm_common-obj08.cf osvvm_axi4-obj08.cf
+ rm -f fpga_serial_acl_tester_tb
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/NamePkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/OsvvmGlobalPkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/VendorCovApiPkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/TranscriptPkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/TextUtilPkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/AlertLogPkg.vhd
../OSVVM/OsvvmLibraries/osvvm/AlertLogPkg.vhd:1387:46:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
../OSVVM/OsvvmLibraries/osvvm/AlertLogPkg.vhd:1437:7:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
../OSVVM/OsvvmLibraries/osvvm/AlertLogPkg.vhd:1574:7:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
../OSVVM/OsvvmLibraries/osvvm/AlertLogPkg.vhd:1838:16:warning: declaration of "alertcount" hides variable "alertcount" [-Whide]
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/MessagePkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/SortListPkg_int.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/RandomBasePkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd ScoreboardPkg_int.vhd ResolutionPkg.vhd TbUtilPkg.vhd OsvvmContext.vhd
+ ghdl -a --std=08 --work=osvvm ../OSVVM/OsvvmLibraries/osvvm/RandomPkg.vhd
+ for filename in NamePkg.vhd OsvvmGlobalPkg.vhd VendorCovApiPkg.vhd TranscriptPkg.vhd TextUtilPkg.vhd AlertLogPkg.vhd MessagePkg.vhd SortListPkg_int.vhd RandomBasePkg.vhd RandomPkg.vhd CoveragePkg.vhd MemoryPkg.vhd ScoreboardGenericPkg.vhd ScoreboardPkg_slv.vhd Scoreboa [... cut ...]
/usr/local/bin/ghdl1-llvm --std=08 --work=osvvm_axi4 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -fpic -o Axi4Master.o ../OSVVM/OsvvmLibraries/AXI4/Axi4/src/Axi4Master.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:859
Call stack traceback locations:
0x7fa7b8e257b1 0x47a178 0x4bfe70 0x5377dc 0x53a225 0x53a3ef 0x53b030 0x53b82c 0x53ba82 0x53bbd9 0x57413f 0x57b20a 0x5318f8 0x640e5f 0x642aca 0x644134 0x40a0ab 0x7fa7b7ed2bf5 0x4093a8 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error
+ exit 1`

tpcorrea, 2021-04-30:
Hi, I had the same issue with OSVVM 2021.02 in a Windows 10 machine and GHDL v0.37.

Paebbels, 2021-05-01:
OSVVM (the AXI models, ut not the main library) uses VHDL-2008 features not yet supported by GHDL.

umarcor, 2021-10-24:
As commented in https://github.com/ghdl/ghdl/issues/1532#issuecomment-950249077, I think that this issue might be fixed already.

/cc @JimLewis