ghdl/ghdl #1465: index constraint not allowed here

sckoarn, 2020-09-16:
**Description**
 index constraint not allowed here,   is the result of the attempt to create an attribute of std_logic_vector  constrained.
A constrained attribute is required to enable aggregate assignment.

Target LRM 2000 section 7.3.2.2 g

**Expected behaviour**
It should compile,  Aldec accepts this.

**How to reproduce?**

https://www.edaplayground.com/x/Y__K

The commented out line is the actual  end goal.  aggregate assignment to an attribute using others

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

eine, 2020-09-16:
```vhdl
  attribute initv: std_logic_vector(15 downto 0);  
  signal slv1 : std_logic_vector(15 downto 0);
  attribute initv of slv1: signal is (0=>'0',1=>'1',others => 'U');
```
[September 16, 2020 4:47 PM](https://gitter.im/ghdl1/Lobby?at=5f622579d993b837e0698f85)

tgingold, 2020-09-16:
Yes, they aren't.  That's according to LRM08 6.7

closed 2020-09-16 by tgingold