ghdl/ghdl #1466: Components missing in precompiled Xilinx libraries.

ZirconfleX, 2020-09-16:
**Description**
After compiling ALL Xilinx libraries I discovered that a lot of in teh XIlinx libraries available components are not compiled in teh GHDL libraries.
Maybe I'm doing something wrong> Here is what I did.

**Expected behaviour**
Using: Linux-Mint 2020.4
Using: Vivado_202.1

Compiled the libraries by adjusting the "config.sh" script.
    InstallationDirectories[XilinxVivado]="/opt/Xilinx/Vivado/2020.1"
    DestinationDirectories[XilinxVivado]="/opt/Ghdl/lib/ghdl/Xilinx93"
    SourceDirectories[XilinxVivado]="data/vhdl/src"
The run the script:
    compile-xilinx-vivado.sh --vhdl93 --all    (as described in the ReadTheDocs documentation)

After compilation finishes I do have Xilinx precompiled component libraries BUT a lot of components are completely missing.
Example: Missing in teh UNISIM library:
   All "AIE_PL_S_<componentName>
   ISERDESE2
   ISERDESE3
   All Ultrascale/Ultrascale+ IO and memory components (BITSLICE_CONTROL, ... URAM288, ...)
General writing:
   ALL COMPONENTS OF THE LATEST FPGA FAMILIES ARE MISSING

What am I doing wrong here?

Thanks

tgingold, 2020-09-16:
I think that many new components of the latest fpga families are now encrypted.

ZirconfleX, 2020-09-17:
Hi,

Can be the case, but then the simulator will become non-relevant because 
designs can no longer be simulated.
Lots of designs are SOC designs adn, as I discovered, all components of 
these FPGA families are not available for simulation.
Question is: why can other simulators use those components?

svenn71, 2020-09-17:
If I were the owner of Modelsim or Synopsys, I would make an agreement with Xilinx to make my customers able to use binary versions of those components in their simulations or synthesis. Money talks.

ZirconfleX, 2020-09-17:
Menotor (modelsim, Questa), Synopsis and other simulators all work 
flawless up to the latest FPGA families.
I don't think complicated things are playing here.
Xilinx has no interest in making it complicated for other tools to use 
their stuff.

Yes, I know of first hand, all simulation primitives of speciality 
functions in the latest FPGA families are encrypted.
There just must be a way to use them as they are, encrypted.
     In the Xilinx unisims folder a file "unisim_VCOMP.vhd" exists.
     That file contains ALL primitives of ALL FPGA families
     So I guess that file is used for encrypted primitives as kind of a 
wrapper.

ZirconfleX, 2020-09-17:
Some extra

In the /Xilinx/Vivado/<version>/data directory, one can find following 
sub-directories:

- pubkey
     Contains different .vhd files but non of the files contains vhdl 
code by a public pgp key.
     Maybe those keys are used to decode ans use encrypted models of 
primitives.
- secureip
     This directory conatins a lot of sub-directories.
     Each containing one or multiple encrypted primitive models
     example: One of the sub-directories is called oserdese2 and 
contains four files, the encrypted model of a OSERDES of the 7-Series 
family.
- ip
     Contains all IP available for Vivado.
- boards
     Information of all boards that can be accessed from within Vivado.
     If one has a design and want to put it on a Xilinx development 
board he can indicate this in Vivado.
     All the board information is sucked from this directory.

Maybe somebody with access to a commercial simulator can find out how 
the pre-compilation of their libraries works.

svenn71, 2020-09-17:
When using modelsim to simulate my xilinx fpga designs I used to rely on the info here : https://www.xilinx.com/support/answers/64083.html

tgingold, 2020-09-17:
I had a look today at Vivado 2019.

The unisim_vcomp package is in clear text and contains many component
declarations including the one for ISERDESE3.
So if you use that component, the design can be analyzed.

Now, the corresponding entity is declared in an encrypted file.
The problem with those files is that they are encrypted using xsim, modelsim, vcs, aldec and maybe a few other keys.

For sure, ghdl could publish a key so that vendors can encrypt for ghdl.  But I really doubt they will do it!

Paebbels, 2021-01-16:
> Xilinx has no interest in making it complicated for other tools to use their stuff.

@ZirconfleX I can tell you, that Xilinx has no interest in making it easy to use there code. They have an interest that you use their code with their tools and with a few payed tools - less is better, so no maintainance costs. Aldec tool support in addition to Mentor ModelSim was only added because customers creating pressure on Xilinx to add this.

The whole idea of encrypted code is that no one can read the internals. So if Xilinx would provide a key-pair for GHDL, GHDL would have the possibilities to write formatted VHDL code to disc. So you can unencrypt all files and then encyption is useless...

In addition, new and complex simulation models are nowadays written in SystemVerilog. They just provide components for interfacing.