ghdl/ghdl #1468: compile-altera.sh script hangs and never compiles ariaii_pcie_hip

GlenNicholls, 2020-09-18:
**Description**
When running `compile-altera.sh` script, it hangs while analyzing `ariaii_pcie_hip`. I am running with Quartus 17.0. It appears that `sim_lib/arriaii_pcie_hip_atoms.vhd` is the problem.

```
$./compile-altera.sh --all --vhdl93
...
Compiling library 'arriaii_pcie_hip'...
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:60:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:61:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:74:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:75:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:841:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:846:9:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:859:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:1005:19:warning: declaration of "s" hides constant "s" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_components.vhd:1018:19:warning: declaration of "s" hides constant "s" [-Whide]
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_atoms.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_atoms.vhd:514:6:warning: declaration of "init_use_crc_forwarding" hides signal "init_use_crc_forwarding" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/arriaii_pcie_hip_atoms.vhd:520:15:warning: declaration of "calc_k_ptr" hides function "calc_k_ptr" [-Whide]
```

**Context**
- OS: Windows
- Origin:
  - [x] Built from sources: ` 63fd2ca`

GlenNicholls, 2020-09-18:
I'm not too concerned about this problem since it compiles the main libraries I actually use like `altera_mf`, but I'm posting this for visibility and tracability

GlenNicholls, 2020-09-18:
When running with `--vhdl2008`, it makes it much further but fails later. There are also warnings about missing files:

```
Glen.Nicholls@CL1073 MINGW64 /mingw64/lib/ghdl/vendors
$ ./compile-altera.sh --all --vhdl2008
Not all Altera packages are VHDL-2008 compatible! Setting HALT_ON_ERROR to FALSE.
Vendor directory '/mingw64/lib/ghdl/vendors/altera' already exists.
Compiling library 'lpm'...
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/220pack.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4166:17:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4231:41:warning: declaration of "j" hides variable "j" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4257:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4278:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4300:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4323:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4594:49:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:4992:17:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5061:41:warning: declaration of "j" hides variable "j" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5088:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5109:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5131:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5154:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5425:49:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5795:17:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5856:41:warning: declaration of "j" hides variable "j" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5881:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5899:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5917:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:5936:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6205:49:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6603:17:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6672:41:warning: declaration of "j" hides variable "j" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6700:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6721:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6743:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:6766:37:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/220model.vhd:7037:49:warning: declaration of "i" hides variable "i" [-Whide]
Compiling library 'sgate'...
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/sgate_pack.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/sgate.vhd'
Compiling library 'altera'...
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_europa_support_lib.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_europa_support_lib.vhd:309:9:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_europa_support_lib.vhd:321:9:warning: declaration of "i" hides variable "i" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_europa_support_lib.vhd:360:9:warning: declaration of "i" hides variable "i" [-Whide]
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_primitives_components.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_primitives.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_standard_functions.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_syn_attributes.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/alt_dspbuilder_package.vhd'
Compiling library 'altera_mf'...
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_mf_components.vhd'
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_mf.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_mf.vhd:1420:49:warning: declaration of "output_counter_value" hides function "output_counter_value" [-Whide]
C:/intelFPGA/17.0/quartus/eda/sim_lib/altera_mf.vhd:1422:28:warning: declaration of "output_counter_value" hides function "output_counter_value" [-Whide]
[... cut ...]
C:/intelFPGA/17.0/quartus/eda/sim_lib/twentynm_hssi_components.vhd:5585:9:warning: declaration of "i" hides variable "i" [-Whide]
Analyzing package 'C:/intelFPGA/17.0/quartus/eda/sim_lib/twentynm_hssi_atoms.vhd'
C:/intelFPGA/17.0/quartus/eda/sim_lib/twentynm_hssi_atoms.vhd:5979:1:warning: entity "twentynm_hssi_gen3_x8_pcie_hip" was also defined in file "C:/intelFPGA/17.0/quartus/eda/sim_lib/twentynm_hip_atoms.vhd" [-Wlibrary]
--------------------------------------------------------------------------------
Compiling Altera Quartus libraries [FAILED]
```

GlenNicholls, 2020-09-18:
I guess now that I'm here, I should also comment that `./compile-xilinx-vivado.sh --all --vhdl2008` also fails with Vivado 2019.1 libraries

tgingold, 2020-09-18:
I fear it depends on the back you use.

ariaii_pcie_hip  is huge and you may simply need a lot of time when using gcc (or llvm) backends.

eine, 2020-09-18:
@GlenNicholls, when using VHDL 2008, did you note the following line?

> Not all Altera packages are VHDL-2008 compatible! Setting HALT_ON_ERROR to FALSE.

Regarding Vivado build script being likely outdated, see #693.

GlenNicholls, 2020-09-18:
>I fear it depends on the back you use.

@tgingold what do you mean by this?

>ariaii_pcie_hip  is huge and you may simply need a lot of time when using gcc (or llvm) backends.

Alright, I'll kick it off again and let it run for a few hours.

@eine yeah I did, but I thought that it would still be successful. I guess it's not a big deal since it compiles all the major things I ever use for each vendor. Out of curiosity, when I compile vendor libraries for ModelSim for example, I don't have to choose a VHDL language version and yet I can still use VHDL2008. Why is that?

GlenNicholls, 2020-09-23:
@tgingold so I started the compile script yesterday morning and when I signed out from work 8 hours later, it was still compiling that `ariaii_pcie_hip` library and it was still at the same point that I outlined above. However, I let it continue running and this morning it was finished! I'll close this out since it is not a bug.

closed 2020-09-23 by GlenNicholls

Paebbels, 2020-09-24:
Some models are to big for GHDL, so I added a filesize limit to some of the scripts e.g. skipping everything above 100 kB in VHDL.

VHDL-2008 support is broken in many vendor component libraries, that's why I added a switch to silently go on and try to compile as much as possible.

Like Xilinx, Intel also started to kick out libraries while adding new libs. In addition users can chose what to install. The problem is how to know or maintain lists of what could be compiled?