ghdl/ghdl #1470: Subprogram headers in vhdl 2008

alemuller, 2020-09-20:
```vhdl
type t1 is array (natural range<>) of std_logic_vector(7 downto 0);
type t2 is array (natural range<>) of std_logic_vector(3 downto 0);

function foo
    generic   (type t)
    parameter (signal p : in t) return std_logic_vector is
begin
    return p(p'high);
end function;

function f1 is new foo generic map (t => t1);
function f2 is new foo generic map (t => t2);
```

When trying to run this code I get:
```
error: 'return' expected
error: ';' is expected instead of 'generic'
error: 'begin' is expected instead of 'generic'
error: unexpected token 'generic' in a concurrent statement list
```
```
$ ghdl --version
GHDL 1.0-dev (v0.37.0-1020-g84649d66) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.
```

tgingold, 2020-09-21:
Yes, generic subprograms are not yet supported.  I haven't seen any user code that needs them.

alemuller, 2020-09-25:
I stumbled upon them when trying to parametrize functions with types. A MWE:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity e is
end entity;

architecture a of e is
    type t1 is array (natural range<>) of std_logic_vector(7 downto 4);
    type t2 is array (natural range<>) of std_logic_vector(3 downto 0);

    procedure p_shift_right
        generic   (type t)
        parameter (signal pvec  : inout t;
                   signal pelem : in    t'element) is
    begin
        pvec <= pvec(pvec'high-1 downto pvec'low) & pelem;
    end procedure;

    -- Overload
    procedure p_shift_right is new p_shift_right generic map (t => t1);
    procedure p_shift_right is new p_shift_right generic map (t => t2);

    -- Constraint
    procedure p_shift_right_1 is new p_shift_right generic map (t => t1);
    procedure p_shift_right_2 is new p_shift_right generic map (t => t2);

    -- ...
begin
    -- ...
end architecture;
```

Related discussion on [Stack Overflow](https://stackoverflow.com/questions/43324918/vhdl-function-procedure-for-any-type-of-array).

tmeissner, 2020-09-25:
I always use function / procedure overloads instead of this 08 feature. IMHO its difficult to read, the syntax is very clunky. Never tried if the vendor tools support that.