ghdl/ghdl #1471: Access to external variables

JoseDanielMunozFrias, 2020-09-21:
I'm trying to access to an internal signal name from a testbench using the external variables (vhdl-08), but it seems that is not implemented. The analysis is ok, but the elaboration fails with the following error:

$ ghdl -e --std=08 RegisterFile_tb
ranslate_name: cannot handle IIR_KIND_EXTERNAL_SIGNAL_NAME (RegisterFile_tb.vhd:55:21)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1020-g84649d66) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/daniel/Asignaturas/SistDig2/risc-v/vhdl/
Command line:
ghdl -e --std=08 RegisterFile_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x558c18dca2a1 0x558c18fc6d47 0x558c18f8c4d8 0x558c18f8fb02 0x558c18f7f1ff 0x558c18fe764b 0x558c18f4e3db 0x558c18f4b555 0x558c18f59b5e 0x558c18febd89 0x558c18f2de9d 0x558c18e65448 0x558c18fef697 0x558c18c1b47b 0x7f123a394b95 0x558c18c19d58 0xfffffffffffffffe
******************************************************************

eine, 2020-09-21:
Hi José! I think that feature is not supported for simulation yet. See #24.

JoseDanielMunozFrias, 2020-09-21:
This is what I was supposing... Thank to add the feature request.

tgingold, 2023-09-17:
Now supported (by mcode).

closed 2023-09-17 by tgingold