ghdl/ghdl #1473: Crash when a constant name from a package is redefined, VHDL 2008

targeted, 2020-09-23:
**Description**
GHDL crashes when compiling some invalid code.

**How to reproduce?**
Compile the following two files, not that VALUES from config is redefined in test.

```vhd :file: config.vhd
library ieee;

package config is
    constant VALUES: positive := 2;
end config;

package body config is
end config;
```

```vhd :file: test.vhd
library ieee;
use work.config.all;

entity test is
end test;

architecture test of test is

    type values_t is array (0 to VALUES - 1) of boolean;
    signal values: values_t;

    type elem_t is (a, b, c);
    type arr_t is array (0 to VALUES - 1) of elem_t;
    signal arr: arr_t;

begin
    arr <= (others => a);
end test;
```

```bat
C:/ghdl/1.0-dev-mingw64-llvm/bin/ghdl.exe -a --std=08 config.vhd
C:/ghdl/1.0-dev-mingw64-llvm/bin/ghdl.exe -a --std=08 test.vhd
```

**Context**
- OS: Windows 10
- Origin: GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
test.vhd:13:38: no function declarations for operator "-"
eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (test.vhd:13:26)
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-w64-mingw32
Command line:
C:\ghdl\1.0-dev-mingw64-llvm\bin\ghdl1-llvm.exe --std=08 -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ieee\v08\ -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ -c -o test.o test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x4a1d2d 0x4ed8f5 0x4ec798 0x4eca45 0x4ec7ee 0x4ecb5b 0x529abd 0x52a406 0x52ddb6 0x52e09f 0x52e1e1 0x564b2b 0x56b2c3 0x524b5f 0x633024 0x634cac 0x636531 0x402691 0x40138a 0x4014b9 0x7ff85f157bd2 0x7ff86056ce4f
******************************************************************```

targeted, 2020-09-23:
Upd: I tried to further strip the example to bare minimum, it seems that just one file is enough, not two as per initial report, and the problem is just around repeated declaration of VALUES.

tmeissner, 2020-09-23:
I see no definition of `VALUES`. Where is it defined? Without this definition the code makes no sense.

targeted, 2020-09-23:
@tmeissner 
The code is invalid, indeed, but that still doesn't warrant a crash.

eine, 2020-09-23:
@targeted, it is hard to guess whether the crash is produced because the name is redefined (as the title suggests) or because the missing declaration. Would you mind adding both cases? That is, just copy the block above, but define `VALUES` (e.g. as a `constant`).

eine, 2020-09-23:
With the MWE above, I get:

```
# ghdl -a test.vhd 
mytest.vhd:8:34: no declaration for "values"
mytest.vhd:12:38: no function declarations for operator "-"
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl -a --std=08 test.vhd
mytest.vhd:8:34: no declaration for "values"
mytest.vhd:12:38: no function declarations for operator "-"
eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (test.vhd:12:26)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1020-g84649d66) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o mytest.o test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x1400a6f2f 0x1400eef25 0x1400eddac 0x1400ee059 0x1400ede02 0x1400ee16f 0x14016e88d 0x14016f1d6 0x140172b86 0x140172e6f 0x140172fb1 0x140132fa7 0x14013974d 0x140120ebf 0x140234dc6 0x140236a55 0x140238325 0x1400026ea 0x1400013bf 0x1400014f4 0x7ffaa30c6fd2 0x7ffaa3e3cebf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

If I define `constant VALUES: natural := 8;`, I get:

```
# ghdl -a test.vhd
mytest.vhd:11:12: identifier "values" already used for a declaration
mytest.vhd:8:14: previous declaration: constant "values"
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl -a --std=08 test.vhd
mytest.vhd:11:12: identifier "values" already used for a declaration
mytest.vhd:8:14: previous declaration: constant "values"
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

targeted, 2020-09-24:
@eine 
> the crash is produced

the original report that I posted initially had two files, and the VALUES constant was defined in a separate "config" package. Still crashed. Then I tried to simplify it by removing the second package, and it crashed visibly in the same way (didn't check the magic hex of course), so for simplicity I removed the config, leaving VALUES hanging.

eine, 2020-09-24:
@targeted, the point, as shown in my previous message, is that the crash is not produced by a name being redeclared. Hence, the title of this issue is not accurate. Instead, the crash seems to be produced when a missing declaration is used, in VHDL 2008 only.

targeted, 2020-09-24:
@eine Your judgement is way better then mine there. IMO, as a compiler user, all I can do is to provide a simple example that crashes it. Why exactly that happens, I can't say. 
OTOH, like I said, it still crashes if VALUES is defined *in a separate package* and imported from there. If you feel that the title of the bug is misleading, I can set it to anything you say.

eine, 2020-09-24:
> it still crashes if VALUES is defined in a separate package and imported from there

We'd need to, please, have an example of that package. A minimal version of it. Does the following match what you mean?

```vhdl
package pkg is
	constant VALUES: integer := 8;
end package;

package body pkg is
end package body;

library ieee;

entity test is
end test;

use work.pkg.VALUES;

architecture test of test is

    type values_t is array (0 to VALUES - 1) of boolean;
    signal values: values_t;

    type elem_t is (a, b, c);
    type arr_t is array (0 to VALUES - 1) of elem_t;
    signal arr: arr_t;

begin
    arr <= (others => a);
end test;
```

```sh
# ghdl -a --std=08 pkg.vhd

# ghdl -a --std=08 test.vhd
test.vhd:14:38: no function declarations for operator "-"
eval_discrete_type_length: cannot handle IIR_KIND_SIMPLE_NAME (test.vhd:14:26)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1020-g84649d66) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o test.o test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x1400a6f2f 0x1400eef25 0x1400eddac 0x1400ee059 0x1400ede02 0x1400ee16f 0x14016e88d 0x14016f1d6 0x140172b86 0x140172e6f 0x140172fb1 0x140132fa7 0x14013974d 0x140120ebf 0x140234dc6 0x140236a55 0x140238325 0x1400026ea 0x1400013bf 0x1400014f4 0x7ffaa30c6fd2 0x7ffaa3e3cebf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

Note that the crash is produced when defining `arr_t`. However, the following variation works as expected:

```vhdl
package pkg is
	constant VALUES: integer := 8;
end package;

package body pkg is
end package body;

library ieee;

entity test is
end test;

use work.pkg.VALUES;

architecture test of test is

    --type values_t is array (0 to VALUES - 1) of boolean;
    --signal values: values_t;

    type elem_t is (a, b, c);
    type arr_t is array (0 to VALUES - 1) of elem_t;
    signal arr: arr_t;

begin
    arr <= (others => a);
end test;
```

```vhdl
# ghdl -a --std=08 pkg.vhd 

# ghdl -a --std=08 test.vhd
```

targeted, 2020-09-24:
@eine 
this is almost exactly what I had, with the minor difference in my case it was
use work.pkg.all;

commit 7efd2f56 (2020-09-29): vhdl-sem_types: improve error handling on index subtypes.  Fix #1473
closed 2020-09-29 by tgingold