ghdl/ghdl #1474: Unconstrained std_logic_vector array crashes in package but not in entity

FranzForstmayr, 2020-09-24:
I'm aware that VHDL08 support is not complete. Just close the issue if it's a duplicate. 

**Description**
ghdl crashes, when a unconstrained std_logic_vector array is used in package but not in entity

**Expected behaviour**
ghdl should not crash

**How to reproduce?**
The following code is working, when using the vector-arrays in a entity.

```vhd :file: ent_work.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end entity;
    
architecture a of ent  is
    
    type vector_array_t is array(natural range <>) of std_logic_vector;
    
    signal test : vector_array_t(7 downto 0)(7 downto 0);
    signal test2 : std_logic_vector(63 downto 0);
    
    function concatenate(arr : vector_array_t) return std_logic_vector is
		constant ARR_SIZE : natural := arr'length;
		constant VEC_SIZE : natural := arr(arr'low)'length;
		variable ret : std_logic_vector(ARR_SIZE * VEC_SIZE - 1 downto 0);
	begin
		for r in arr'range loop
			ret((r+1) * VEC_SIZE - 1 downto r * VEC_SIZE) := arr(r);
		end loop;
		return ret;
	end function;

begin
	test2 <= concatenate(test);
end;
```

**How to reproduce?**
This does not work:

```vhd :file: pkg.vhd
library ieee;
use ieee.std_logic_1164.all;

package pkg is

	type vector_array_t is array(natural range <>) of std_logic_vector;
	
	function concatenate(arr : vector_array_t) return std_logic_vector;
end package;

package body pkg is

    function concatenate(arr : vector_array_t) return std_logic_vector is
		constant ARR_SIZE : natural := arr'length;
		constant VEC_SIZE : natural := arr(arr'low)'length;
		variable ret : std_logic_vector(ARR_SIZE * VEC_SIZE - 1 downto 0);
	begin
		for r in arr'range loop
			ret((r+1) * VEC_SIZE - 1 downto r * VEC_SIZE) := arr(r);
		end loop;
		return ret;
	end function;

end package body;
```

```vhd :file: ent_fail.vhd
library ieee;
use ieee.std_logic_1164.all;

library work;
use work.pkg.all;

entity ent is
end entity;
    
architecture a of ent is
	signal test : vector_array_t(7 downto 0)(7 downto 0);
	signal test2 : std_logic_vector(63 downto 0);

begin
	test2 <= concatenate(test);
end;

```

```sh :image: ghdl/ghdl:buster-mcode
# this should work
ghdl -a ent_work.vhd
ghdl --elab-run ent

# this not
ghdl -a --std=08 pkg.vhd
ghdl -a --std=08 ent_fail.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x] Built from sources: `34666ab2`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37-dev (v0.36-99-g34666ab2) [Dunoon edition]
Compiled with GNAT Version: 7.4.0
Target: x86_64-linux-gnu
In directory: /home/ecs/Workspaces/vhdl/axis-arith/
Command line:
/usr/local/bin/ghdl1-llvm --std=08 -P/usr/local/lib/ghdl/ieee/v08/ -P/usr/local/lib/ghdl/ -c -o pkg.o pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans-chap3.adb:3001
Call stack traceback locations:
0x617cb9 0x5bc14e 0x5bda41 0x620226 0x56c965 0x62f3e0 0x630aa4 0x6323b7 0x40861b 0x7fca681f5bb9 0x4079e8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

eine, 2020-09-24:
@FranzForstmayr, the version of GHDL you are using is rather old, and several enhancements to unconstrained arrays were pushed since then. Yet, I can still reproduce a crash by analyzing the package alone:

```sh
# ghdl -a --std=08 pkg.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1020-g84649d66) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o pkg.o pkg.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1674
Call stack traceback locations:
0x14009cac2 0x140187c1d 0x1401888b4 0x14018b03d 0x140189584 0x140189877 0x1401ff870 0x140200ef6 0x1401ecb02 0x1401d56e8 0x1402350f8 0x140236a55 0x140238325 0x1400026ea 0x1400013bf 0x1400014f4 0x7ffaa30c6fd2 0x7ffaa3e3cebf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

Regarding VHDL 1993, which you say it should work, this is the output:

```sh
# ghdl -a pkg.vhd
pkg.vhd:6:31: array element of unconstrained array type "std_logic_vector" is not allowed before vhdl08
pkg.vhd:11:14: package "pkg" was not analysed
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

FranzForstmayr, 2020-09-24:
> Regarding VHDL 1993, which you say it should work, this is the output:

My fault, i tested with --std=08 of course, I'll update the description.

> @FranzForstmayr, the version of GHDL you are using is rather old, and several enhancements to unconstrained arrays were pushed since then. Yet, I can still reproduce a crash by analyzing the package alone:

Yeah I know, had troubles building ghdl from sources today, but I think this is another issue.

eine, 2020-09-24:
On Ubuntu 20 or on Windows, you can use these: https://github.com/ghdl/ghdl/releases/tag/nightly.

FranzForstmayr, 2020-09-25:
> On Ubuntu 20 or on Windows, you can use these: https://github.com/ghdl/ghdl/releases/tag/nightly.

I'm using Ubuntu 18, however I was not aware of the '--default-fip' option for configure

commit d8817ae3 (2021-09-07): testsuite/gna: add a test for #1474

Close #1474
closed 2021-09-07 by tgingold