ghdl/ghdl #1476: Unexpected

alemuller, 2020-09-25:


eine, 2020-09-25:
@alemuller, unfortunately, I cannot reproduce:

```sh
# ghdl -a issue.vhdl -e e
C:\msys64\mingw64\bin\ghdl.exe:warning: no option expected after files, use ./-e [-Wunexpected-option]
unknown option '-e'
error: cannot open e
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl --version
GHDL 1.0-dev (v0.37.0-1026-g73d2b4ce) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

alemuller, 2020-09-25:
It seems that is more complicated to reproduce than I previously thought.

It needs to do some mise en place.

`file1.vhdl`
```vhdl file1.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity e is
end entity;

architecture a of e is
    signal s : unsigned(31 downto 0) := (others=>'0');
begin
    assert s = 3;
end architecture;
```

Analyze and elaborate:
```
$ ghdl -a file1.vhdl
$ ghdl -e e
```

Remove `work-obj98.cf`. (Keep the exectuable `e`).
```
$ rm work-obj93.cf
```

----

Now. It should be able to reproduce:

`file2.vhdl`
```vhdl file2.vhdl
entity e is
end entity;

architecture a of e is
begin
end architecture;
```

```
$ ghdl -a file2.vhdl -e e
```

```
ghdl:warning: no option expected after files, use ./-e [-Wunexpected-option]
unknown option '-e'
e:1:1: control character that is not CR, LF, FF, HT or VT is not allowed
e:1:2: missing entity, architecture, package or configuration
e:1:5: control character that is not CR, LF, FF, HT or VT is not allowed
e:1:6: control character that is not CR, LF, FF, HT or VT is not allowed
...
error: error limit reached
ghdl:error: compilation error
```

tgingold, 2020-09-25:
Ok but this is not very different from:  ghdl -a /bin/sh

You are trying to analyze a binary file.  But there is a control on the number of errors.

alemuller, 2020-09-25:
I see.

The command ignores the `-e` and tries to analyze both `file2.vhdl` and `e`. 

I stumbled upon this when replacing a flag from the last ran synth command (`^--synth^-a`).

I'm closing this. The new revisted CLI syntax https://github.com/ghdl/ghdl/issues/1204#issuecomment-641687628 would address my particular use case.

closed 2020-09-25 by alemuller

eine, 2020-09-25:
@alemuller, actually, it seems that you were expecting the single command synth syntax to be supported for simulation too. As you found in that comment, I proposed it some weeks ago. However, GHDL did never really work like that, so it's a new feature to be  added. Fortunately, there was some recent [activity](https://github.com/ghdl/ghdl/issues/1204#issuecomment-696068890) related to it.