ghdl/ghdl #1477: type of a shared variable must be a protected type (ERROR, using -frelaxed, with STD 02 and 08)

rodrigomelo9, 2020-09-26:
Hi. I am using the last `ghdl/synth:beta` with a ISE example ([simple_dual_two_clocks.vhd.tar.gz](https://github.com/ghdl/ghdl/files/5286083/simple_dual_two_clocks.vhd.tar.gz)). It is solved with a shared variable. It works with VHDL03, but it fails using VHDL02 or 08, also usign the -frelaxed option. It was working some months ago (I have GDHL from March installed in my system and it works). Steps to reproduce the problem:
```
$ DOCKER_CMD="docker run --rm -it -v /$(pwd)://wrk -w //wrk ghdl/synth:beta"
$ $DOCKER_CMD ghdl -a -fsynopsys -fexplicit -frelaxed --std=93 simple_dual_two_clocks.vhd
$ echo $?
0
$ $DOCKER_CMD ghdl -a -fsynopsys -fexplicit -frelaxed --std=02 simple_dual_two_clocks.vhd
simple_dual_two_clocks.vhd:24:18:error: type of a shared variable must be a protected type
simple_dual_two_clocks.vhd:24:18:note: (you can use -frelaxed to turn this error into a warning)
$ secho $?
1
$ $DOCKER_CMD ghdl -a -fsynopsys -fexplicit -frelaxed --std=08 simple_dual_two_clocks.vhd
simple_dual_two_clocks.vhd:24:18:error: type of a shared variable must be a protected type
simple_dual_two_clocks.vhd:24:18:note: (you can use -frelaxed to turn this error into a warning)
$ echo $?
1
```
Regards,
Rodrigo

tgingold, 2020-09-26:
Put --std=XX option before -fexplicit and -frelaxed.

eine, 2020-09-26:
@tgingold, is that something that we need to document or is it fixable?

tgingold, 2020-09-26:
It would be simpler to document it!

rodrigomelo9, 2020-09-26:
It works :-) which is great, but it seems something broken, because:
* It worked a few months ago.
* It finishes with an ERROR but is not clearly indicated.
* It proposes to use -frelaxed, when it was already used.

So, I think that this note in the documentation will be not enough (seems more a workaround). I can try to find when it broke and also try to fix it (without guaranties haha).

tgingold, 2020-09-26:
The behaviour has slightly changed when --std=93c became exactly like --std=93 -frelaxed

eine, 2020-09-26:
@rodrigomelo9 see #980. It's not the same as this issue, but I think it is related, since it's about reworking `--std`. Therefore documenting that "*--std must be the first option*" is indeed a workaround while we wait for that, or some other fix.

commit 189cec2a (2020-09-28): doc: clarify position of option --std.  Fix #1477
closed 2020-09-28 by tgingold