ghdl/ghdl #1478: gtkwave crashes with unknown type 37

FranzForstmayr, 2020-09-28:
**Description**
I'm using unconstrained std_logic_vector arrays in vhdl. Simulation runs without errors, however gtkwave is unable to show the waveforms. Already created a bug in https://github.com/gtkwave/gtkwave/issues/31, they suggested to create another one here too.

**Expected behaviour**
Do not crash :) 

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: pkg.vhd
library ieee;
use ieee.std_logic_1164.all;

package pkg is

	type vector_array_t is array(natural range <>) of std_logic_vector;
end package;

package body pkg is
end package body;
```

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library work;
use work.pkg.all;

entity ent is
    generic (
        REGISTERS           : natural  := 1
        );
    port (
        read_reg      : in    vector_array_t(REGISTERS downto 0)(31 downto 0) := (others => (others => '0'))
    );
end entity;

architecture arch of ent is
begin
end architecture;
```

```vhd :file: tb.vhd
library ieee;
use ieee.std_logic_1164.all;

library work;
use work.pkg.all;

entity tb is
end entity;

architecture arch of tb is
begin
    dut: entity work.ent 
    generic map(
        REGISTERS => 4
    ) port map(
        read_reg => open
    );
    
    
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 pkg.vhd
ghdl -a --std=08 ent.vhd
ghdl -a --std=08 tb.vhd
ghdl --elab-run --std=08 tb --wave=out.ghw
gtkwave out.ghw
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ x ] Built from sources: `GHDL 1.0-dev (v0.37.0-1026-g73d2b4ce)`

**Additional context**
gtkwave output:
```
GTKWave Analyzer v3.3.103 (w)1999-2019 BSI

ghw_read_type: unknown type 37
ghw_read_base: error in section TYP
Error in ghw file 'out.ghw'.
```

FranzForstmayr, 2020-09-28:
Interesting fact:
I was able to reproduce the error with an unconstrained std_logic_vector too. This was definitely working before.
I updated the issue description accordingly.

tgingold, 2020-09-28:
I think you are using a too old version of gtkwave.  You should build it from very recent sources.

Or use an older version of ghdl.

FranzForstmayr, 2020-09-29:
Thank you.
Already tried 3.3.103, but 3.3.106 is working!

closed 2020-09-29 by FranzForstmayr