ghdl/ghdl #1479: synth: assume statement expecting a clock directive

alemuller, 2020-09-28:
The code should work without the clock directive.

```vhdl issue.vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity e is
    port (lhs, rhs : unsigned(7 downto 0));
end entity;

architecture a of e is
begin
    assume rhs < rhs;
    assert lhs > rhs;
end architecture;
```
```
$ ghdl --synth --std=08 issue.vhdl -e e
issue.vhdl:11:5:error: no clock for PSL directive
```
Note: No error is thrown when only the assert statement is used.

tmeissner, 2020-09-28:
GHDL doesn't support unclocked PSL directives at the moment. If you use a simple assert, no PSL is used, instead it is a simple VHDL assertion (in >= VHDL-08).

eine, 2020-09-28:
@tmeissner, please consider adding some reference to `--no-formal` in the same PR. It has been in my todo since June... https://github.com/ghdl/ghdl/commit/667ab51811b612da68524043874277e6484f3392#commitcomment-39945312.