ghdl/ghdl #1481: synth: operation on types with upward range

alemuller, 2020-09-28:
`upto.vhdl`
```vhdl file:upto.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity e is
    port (i : in  std_logic_vector(3 to 0);
          o : out std_logic);
end entity;

architecture a of e is
begin
    o <= xor(i);
end architecture;
```
```
$ ghdl synth --std=08 upto.vhdl -e e > syn_upto.vhdl
```
`syn_upto.vhdl`
```vhdl file:syn_upto.vhdl
library ieee;
use ieee.std_logic_1164.all;
entity e is
  port (
    i: in std_logic_vector (3 to 0);
    o: out std_logic
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of e is
  signal wrap_i: std_logic_vector (-1 downto 0);
  signal wrap_o: std_logic;
  signal n1_o : std_logic;
begin
  wrap_i <= i;
  o <= wrap_o;
  wrap_o <= n1_o;
  -- upto.vhdl:11:10
  n1_o <= wrap_i; -- reduce xor
end rtl;
```
```
$ ghdl -a syn_upto.vhdl
syn_upto.vhdl:23:11:error: can't match 'wrap_i' with type std_ulogic
ghdl:error: compilation error
```

tgingold, 2020-09-28:
It is not really a problem with upward ranges, but with null ranges.
The vector (3 to 0) has no elements.

alemuller, 2020-09-28:
Ops. Fixed the title.

commit da729b5a (2020-09-28): netlists-disp_vhdl: handle null vectors for reducation operators.

Fix #1481
closed 2020-09-28 by tgingold