ghdl/ghdl #1483: -frelaxed option doesnt work with elaboration

IshFx, 2020-10-02:
**Description**
When i set `-frelaxed` option while running `ghdl --elab-run`,  ghdl doesn't take it into account and exists with an error and a note `note: (you can use -frelaxed to turn this error into a warning)`.

Here is the command and its output :

```
DEBUG - Started process with pid=8: '/usr/local/bin/ghdl --elab-run --std=08 --work=* --workdir=* -P* -o * tb_mic_interface tb -frelaxed -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : *,tb path : *,use_color : true --assert-level=error'
/work/hdl/common/src/rtl_true_dp_ram.vhdl:50:21: type of a shared variable must be a protected type
/work/hdl/common/src/rtl_true_dp_ram.vhdl:50:21:note: (you can use -frelaxed to turn this error into a warning)
/usr/local/bin/ghdl: compilation error
  DEBUG - Process with pid=8 terminated with code=1
```

The same option works as expected with analysis `ghdl -a`, meaning it only print a warning.

**Expected behaviour**
Only print a warning and not throw an error.

**How to reproduce?**
Here is the vhdl file causing this error.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

------------------------------------------------------------------------------------------
-- Entity
------------------------------------------------------------------------------------------

entity true_dp_ram is

    generic (
        DATA_WIDTH : natural;
        ADDR_WIDTH : natural
    );

    port (
        a_clock    : in std_logic;
        a_address  : in natural range 0 to 2**ADDR_WIDTH - 1;
        a_data     : in std_logic_vector((DATA_WIDTH-1) downto 0);
        a_write_en : in std_logic;
        a_data_out : out std_logic_vector((DATA_WIDTH -1) downto 0);

        b_clock    : in std_logic;
        b_address  : in natural range 0 to 2**ADDR_WIDTH - 1;
        b_data     : in std_logic_vector((DATA_WIDTH-1) downto 0);
        b_write_en : in std_logic;
        b_data_out : out std_logic_vector((DATA_WIDTH -1) downto 0)
    );

end true_dp_ram;

------------------------------------------------------------------------------------------
-- Architecture
------------------------------------------------------------------------------------------

architecture rtl_true_dp_ram of true_dp_ram is

    subtype word_t is std_logic_vector((DATA_WIDTH-1) downto 0);
    type memory_t is array(2**ADDR_WIDTH-1 downto 0) of word_t;

    shared variable ram : memory_t;

begin

    -- Port A
    process(a_clock)
    begin
    if(rising_edge(a_clock)) then 
        if(a_write_en = '1') then
            ram(a_address) := a_data;
        end if;
        a_data_out <= ram(a_address);
    end if;
    end process;

    -- Port B
    process(b_clock)
    begin
    if(rising_edge(b_clock)) then 
        if(b_write_en = '1') then
            ram(b_address) := b_data;
        end if;
        b_data_out <= ram(b_address);
    end if;
    end process;

end rtl_true_dp_ram;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**
Please, provide the following information:

- Docker image : ghdl/ext:ls-vunit - b431f2356f9c

**Additional context**
I had no error with Modelsim.

Thanks in advance!

eine, 2020-10-02:
@IshFx, it's ok, it's just a syntax error. The syntax for elab-run is `ELABORATION_OPTIONS TOP_UNIT [ARCH] RUN_OPTIONS`. `-frelaxed` is an Analysis or Elaboration option, not a Run option. Hence, you need to change your command from:

```sh
ghdl --elab-run --std=08 --work=* --workdir=* -P* -o * tb_mic_interface tb -frelaxed -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : *,tb path : *,use_color : true --assert-level=error'
```

to

```sh
ghdl --elab-run --std=08 -frelaxed --work=* --workdir=* -P* -o * tb_mic_interface tb -grunner_cfg=active python runner : true,enabled_test_cases : ,output path : *,tb path : *,use_color : true --assert-level=error'
```

Note that `-frelaxed` was moved from after `tb` to before.

However, I am really curious about your use case. It seems that you did not write the command yourself. You are using VUnit instead. Therefore, the problem is likely in your `run.py` file. See https://github.com/VUnit/vunit/blob/master/examples/vhdl/array/run.py#L28-L29.

IshFx, 2020-10-02:
You are right! i misused vunit options. i was passing the -frelaxed option to `ghdl.vsim_flags` instead of `ghdl.elab_flags`.

this fixed my issue !
[image]

Thank you for your help, and sorry for wasting your time on such mistake.

eine, 2020-10-02:
Oh, no worries mate! We will never get used to this CLI param diversity 😆

closed 2020-10-02 by eine