ghdl/ghdl #1484: Attempting to build on Centos (GCC) fails with error in src copy step

stfarley, 2020-10-03:
**Description**
Following the instructions to build on a Centos 7 machine I get the following 
make copy-sources
if test -d ../.git && desc=`cd ..; git describe --dirty`; then GHDL_DESC="$desc"; fi; \
sed -e "s/[(].*[)]/($GHDL_DESC)/" -e "s/@VER@/1.0-dev/" < ../src/version.in > version.tmp; \

if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
cp: skipping file ‘version.tmp’, as it was replaced while being copied
make: *** [Makefile:173: version.ads] Error 1
 

**Expected behaviour**
I assume that there should be no error.

**How to reproduce?**
Followed the instructions to build for GCC.

tgingold, 2020-10-03:
Do you start from a clean state ?  What is the output of `ls -l version*` ?

stfarley, 2020-10-03:
Hi Thanks for the response.

It is from a fresh checkout

ls -l version*
returns
-rw-rw-r--. 1 stevefarley stevefarley 163 Oct  2 13:22 version.tmp

Thanks,
Steve

tgingold, 2020-10-04:
As far as I understand, the '.' after '-rw-rw-r--' means there is a security context.  That's unusual and might be the source of the problem.

Can you manually copy the file: `cp version.tmp version.ads` ?
Once done, you should be able to continue the make.

Fatsie, 2020-10-04:
See #1423
TL;DR No right version of gcc with ADA support is available packaged for Centos7. I had to compile own gcc with ADA support to get it compiled.

eine, 2020-10-04:
> TL;DR No right version of gcc with ADA support is available packaged for Centos7. I had to compile own gcc with ADA support to get GHDL compiled.

That's unfortunately correct. See also #1118.

stfarley, 2020-10-04:
Hi Everybody,

Thanks for the responses.

The initial problem turns out to be a strange file system issue. (Building on a usb mounted drive through parallels)
Moved to a native Centos machine on an NFS mount an the file copy problems went away.

I was able to get most of the way through the build (Using gcc-gnat, installed with yum), However I get an error near the end. Basically indicates that ADA needs to be newer.
So I removed gcc-gnat and downloaded the gnat 2020 installer and manually installed that. (Modified my env to point to it)

Now I get this error.
checking whether fflush_unlocked is declared... 0
configure: error: Oops, mp_limb_t doesn't seem to work
yes
checking whether fgetc_unlocked is declared... make[1]: *** [configure-gmp] Error 1

Thanks
Steve

tgingold, 2020-10-05:
Did you try to first build with the mcode backend ?  That's much simpler.

stfarley, 2020-10-05:
Hi,

I haven’t tried it this time with MCODE. I will give that a try though. I was trying to get it through the GCC backend as I am looking for the performance improvement.

Thanks,
Steve

eine, 2020-10-05:
@stfarley performance of LLVM should be comparable or better than GCC. My suggestion is to try mcode first, and then LLVM. GCC is only useful if you need debugging or code coverage.

stfarley, 2020-10-10:
OK
 I can get mode to build.

LLVM builds fail with
Unhandled version llvm 3.4.2
using 
scl enable llvm-toolset-7 bash
doesn’t seem to make any difference.

GCC build fails at the make -j2 && make install step with
/projects/gcc-9.3.0/libiberty/gettimeofday.c:22:1: error: conflicting types for 'gettimeofday'
   22 | gettimeofday (struct timeval *tp, void *tz)
      | ^~~~~~~~~~~~

tgingold, 2020-10-12:
Yes, llvm 3.4 is too old. But ghdl supports later versions.  What is the error message ?

> GCC build fails at the make -j2 && make install step with
> /projects/gcc-9.3.0/libiberty/gettimeofday.c:22:1: error: conflicting types for 'gettimeofday'
> 22 | gettimeofday (struct timeval *tp, void *tz)
> | ^~~~~~~~~~~~

It looks like gcc-9.3.0 doesn't build on your computer.  That also prevents from building ghdl with the gcc backend.

stfarley, 2020-10-14:
Ah thanks for all of the responses.

I managed to get it to compile with GCC. It seems that it doesn’t matter which version of GCC I use to compile it, The crucial one seems to be the version of GCC source code I compile it with.
i.e. I can do "set enable devtoolset-X bash” with any available version and it compiles just fine.
What I had to do is use GCC 5.5.0 source code to compile against.

for LLVM even after setting LLVM to a newer version I still get the message complaint about LLVM 3.4… not handled.

Thanks again
Steve

tgingold, 2020-10-14:
For llvm, you should specify which llvm-config you are using.  If your system compiler is 3.4, then the default llvm-config would be for 3.4.  You need to install a later version (like 7.x) and do:
$ configure --with-llvm-config=llvm-config7.0

rodrigomelo9, 2021-02-08:
Hi. I was recently helping to compile GHDL for Centos7, with the GCC backend, and it works. I installed from a fresh centos7 docker image. The main problem is gnat and its graphical installer XD. Following, the steps I used:

* Go to https://www.adacore.com/community, click on `download`, select the Linux installer (bin).
* `chmod +x *.bin` and execute it to install.
* I created the following Dockerfile:
```
FROM centos:centos7

RUN yum update -y
RUN yum -y install centos-release-scl
RUN yum -y install gcc-gnat bash

RUN yum install -y centos-release-scl && yum install -y devtoolset-8
RUN gcc --version
SHELL [ "/usr/bin/scl", "enable", "devtoolset-8" ]
RUN gcc --version

RUN yum -y install git curl bzip2 wget flex texinfo fontconfig libX11 zlib-devel
```
**Note:** it is not the best Dockerfile in the world, I know :P was a PoC about to install GHDL-GCC in Centos7.
* Run the Docker (having access to the installed GNAT) and execute
```
export PATH="<GNAT_INSTALL_PATH>/bin:$PATH"

mkdir ghdl && curl -L https://github.com/ghdl/ghdl/archive/master.tar.gz | tar -xz -C ghdl --strip-components=1
mkdir gcc && curl -L https://codeload.github.com/gcc-mirror/gcc/tar.gz/releases/gcc-8.3.0 | tar -xz -C gcc --strip-components=1
cd gcc && sed -i.bak s/ftp:/http:/g contrib/download_prerequisites && ./contrib/download_prerequisites && cd ..

mkdir build7 && cd build7 && ../ghdl/configure --with-gcc=../gcc --prefix=/usr/local && make copy-sources
mkdir gcc-objs && cd gcc-objs
../../gcc/configure --prefix=/usr/local --enable-languages=c,vhdl --disable-bootstrap --disable-lto --disable-multilib --disable-libssp --disable-libgomp --disable-libquadmath
make -j2 && make install && cd .. && make ghdllib
```
* And finally,  `make install` as root.

Could be useful to put somewhere? To solve this issue? @stfarley ? 

Regards,

eine, 2021-02-08:
As commented in #1118 a batch installation of AdaCore's GNAT is required, so that we can add this to ghdl/docker. In the macOS CI jobs we do install AdaCore's GNAT (see https://github.com/ghdl/ghdl/blob/master/scripts/macosx/install-ada.sh). A similar procedure might work.

closed 2021-09-07 by tgingold