ghdl/ghdl #1486: Complex constants in when statement

hutch31, 2020-10-05:
**Description**
GHDL flags constant expressions if they use numeric conversions

**Expected behaviour**
Expressions should be considered to be locally static as long as they can be resolved to a constant at compile time

**How to reproduce?**

```library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.ALL;

entity TestcaseStatic is
  port (
  SEL    : in std_logic_vector(1 downto 0);
  DATA0  : in std_logic_vector(7 downto 0);
  DATA1  : in std_logic_vector(7 downto 0);
  DATA2  : in std_logic_vector(7 downto 0);
  DATA3  : in std_logic_vector(7 downto 0);
  RESULT : out std_logic_vector(7 downto 0)
  );
end;

architecture test of TestcaseStatic is

  constant SEL_ZERO : std_logic_vector(1 downto 0) := "00"; 

begin

  process(SEL, DATA0, DATA1, DATA2, DATA3)
    begin
      case (SEL) is
	when SEL_ZERO => 
          RESULT <= DATA0;
	when std_logic_vector(unsigned(SEL_ZERO)+1) => 
          RESULT <= DATA1;
	when std_logic_vector(unsigned(SEL_ZERO)+2) => 
          RESULT <= DATA2;
	when std_logic_vector(unsigned(SEL_ZERO)+3) => 
          RESULT <= DATA3;
      end case;
    end process;

end test;

```

**Context**
Should be platform-independent, but ran on Ubuntu 20, GHDL 0.37

tgingold, 2020-10-05:
> Expressions should be considered to be locally static as long as they can be resolved to a constant at compile time

Sorry, but this is not what the LRM says.

However, in VHDL 2008, your test should pass, so I keep the issue.

hutch31, 2020-10-05:
I understand this may not be according to the LRM, but the fragment above was extracted from working code that passed through Aldec-HDL and Synplify.

eine, 2020-10-05:
@hutch31, in Aldec-HDL and Synplify, was it analysed as VHDL 1993 or VHDL 2008?

hutch31, 2020-10-05:
Aldec was compiled as LRM-2002 with "relax LRM requirements" set.  Synplify compiles with VHDL-2008 checked and without (presumably 1993).

diogratia, 2020-10-05:
You're code example is missing a library clause making `ieee` visible as well as the command line arguments to ghdl.

The -2008 standard incorporates packages previously found in other standards:

&nbsp;1. Overview  
 
1.1 Scope  

>This standard revises and enhances the VHDL language reference manual (LRM) by including a standard C language interface specification; specifications from previously separate, but related, standards IEEE Std 1164-1993 [B16],1 IEEE Std 1076.2-1996 [B11], and IEEE Std 1076.3-1997 [B12]; and general language enhancements in the areas of design and verification of electronic systems.

Here the standards of interest are 1164 and 1076.3:

9.4.2 Locally static primaries

>An expression is said to be locally static if and only if every operator in the expression denotes an implicitly defined operator or an operator defined in one of the packages STD_LOGIC_1164, NUMERIC_BIT, NUMERIC_STD, NUMERIC_BIT_UNSIGNED, or NUMERIC_STD_UNSIGNED in library IEEE, and if every primary in the expression is a locally static primary, where a locally static primary is defined to be one of the following: 
>>a) A literal of any type other than type TIME  
>>  
>>b) A constant (other than a deferred constant) explicitly declared by a constant declaration with a locally static subtype or with an unconstrained or partially constrained composite subtype for which the applicable constraints are locally static, and initialized with a locally static expression  
>>...

As a result expressions involving operators (e.g. `unsigned(SEL_ZERO)+2`) from these packages can be considered locally static .

In previous revisions of the standard (here -2002) this was not the case:

7.4.1 Locally static primaries
  
>An expression is said to be locally static if and only if every operator in the expression denotes an implicitly defined operator whose operands and result are scalar and if every primary in the expression is a locally static primary, where a locally static primary is defined to be one of the following:
>> ...

You'd find that currently with --std=08 command line argument ghdl doesn't treat your expressions as locally static representing an unimplemented -2008 feature.

hutch31, 2020-10-06:
Thank you for the explanation, however I'm still unclear as to why older revisions of the standard would still not treat this as a constant, unless unsigned() is not "an implicitly defined operator".  The results of unsigned(FOO)+1 are still scalar.

Also, my version of GHDL (0.37 release) does not understand an --std argument, is this a new feature?

alemuller, 2020-10-06:
*Type conversion* is a *basic operation* (5.1), not an *operator* (9.2).

*Operators* are only the functions defined with *operator symbol* (such as "+", "and", etc).

Anyhow, type conversions are locally static expression (9.4.2).  Even if they weren't, it is possible to reproduce your issue without them.

I'm not familiar with the older std.

> Also, my version of GHDL (0.37 release) does not understand an --std argument, is this a new feature?

The std has to be after `-a`.
```
ghdl -a --std=08 filename
```

diogratia, 2020-10-06:
> Thank you for the explanation, however I'm still unclear as to why older revisions of the standard would still not treat this as a constant, unless unsigned() is not "an implicitly defined operator". The results of unsigned(FOO)+1 are still scalar.

As alemuller points out a type conversion is not an implicitly defined operator it's a basic operation.  The semantics of type conversion are found in 9.3.6 Type conversions.

Also `unsigned(FOO)+1` does not provide a scalar value, it's value type is unsigned which is an array type intended to be used numerically through the use of operator overloads.

The function declaration for the "+" overload is found in IEEE package numeric_std:

```vhdl
function "+" (L : UNRESOLVED_UNSIGNED; R : NATURAL) return UNRESOLVED_UNSIGNED;
```
(The function L parameter and result type in IEEE Std 1076.3-1996 package numeric_std were of type UNSIGNED, -2008 has changes involving resolution.)

The reason why the overloaded operator with locally static parameters is not considered locally static in previous revisions is that the package was not part of the 1076 standard itself. The -2008 standard revision does not allow interface declarations nor functional semantics to be changed for the packages mentioned in 9.4.2. See 16.7 Standard multivalue logic package, 16.8.5 Packages for arithmetic using bit and standard logic values, 16.8.5.2 Allowable modifications.

JimLewis, 2021-11-23:
Ran into this too where I have constants that are computed:
library ieee ;
use ieee.std_logic_1164.all ;
use ieee.numeric_std_unsigned.all ;

```vhdl
package CpuCheckerPkg is
    subtype  CAddrType is std_logic_vector(28 downto 0) ; 

    constant CHECK_BASE            : CAddrType := (others => '0') ;
    
    constant CHECK_NO_NRDY         : CAddrType := CHECK_BASE + 1 ; 
    constant CHECK_2X_NRDY         : CAddrType := CHECK_BASE + 2 ; 
    constant CHECK_IMMEDIATE_NRDY  : CAddrType := CHECK_BASE + 3 ; 
    constant CHECK_STRAY_NRDY      : CAddrType := CHECK_BASE + 4 ; 
```

And used in a case statement:
``` vhdl
      case? Addr is 
        when CHECK_CYCLE_RANGE =>
          for i in 2 to to_integer(Addr(4 downto 0)) loop
            wait until Clk = '1' ;
          end loop ; 
          DriveData(Addr) ; 
          DriveRdy ; 

        when CHECK_NO_NRDY =>
          wait until Clk = '1' ; 
          DriveData(CHECK_NO_NRDY) ; 
          -- DriveRdy ;  -- not driven in this test
```

I have VHDL-2008 selected.    

Errors are:
```
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:114:14: choice must be locally static expression
        when CHECK_CYCLE_RANGE =>
             ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:121:14: choice must be locally static expression
        when CHECK_NO_NRDY =>
             ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:126:14: choice must be locally static expression
        when CHECK_2X_NRDY =>
             ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:131:14: choice must be locally static expression
        when CHECK_IMMEDIATE_NRDY =>
             ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:141:14: choice must be locally static expression
        when CHECK_STRAY_NRDY =>
             ^
C:/SynthWorks/Dev/_lab_build/VhdlTest_prototype/Tb2_extras/MemIO_Tb_CpuChecker.vhd:151:14: choice must be locally static expression
        when CHECK_TIMING_RANGE =>
             ^
C:\tools\msys64\mingw64\bin\ghdl.exe: compilation error
child process exited abnormally
```

thmslg, 2022-11-16:
> You'd find that currently with --std=08 command line argument ghdl doesn't treat your expressions as locally static representing an unimplemented -2008 feature.

@diogratia I am not sure to understand this sentence.  Is the issue confirmed ?

BTW, I confirm the same behavior  with version 2.0.0.

As previously mentioned in similar ticket, _case_ replacement with _if/elseif_ solve the ghdl issue. Could be helpful as a workaround in some situations.

tgingold, 2022-11-16:
This test is now working correctly with the head.

commit d3f4d087 (2022-11-17): testsuite/gna: add a test and close #1486
closed 2022-11-17 by tgingold