ghdl/ghdl #1490: Crash when referencing an entity from a separate file

targeted, 2020-10-10:
**Description**
Compiler crashes compiling the attached code.

**How to reproduce?**
Compile this:

```vhd :file: ent.vhd
entity ent is
end ent;
```

```vhd :file: arc.vhd
use work.ent.ent; -- ***this second .ent causes the crash***
architecture foo of ent is
begin
end foo;
```

```sh
ghdl.exe -a ent.vhd
ghdl.exe -a arc.vhd
```

**Context**
Please, provide the following information:

- OS: Windows 10
- Origin: GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]
Compiled with GNAT Version: 9.1.0
Target: x86_64-w64-mingw32
Command line:
C:\ghdl\1.0-dev-mingw64-llvm\bin\ghdl1-llvm.exe -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ieee\v93\ -PC:\ghdl\1.0-dev-mingw64-llvm\lib/ghdl\ -c -o arc.o arc.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:859
Call stack traceback locations:
0x157fba6 0x472b41 0x4f7257 0x4f7478 0x56aa46 0x56ad09 0x56b239 0x524b5f 0x633024 0x634cac 0x636531 0x402691 0x40138a 0x4014b9 0x7ff982af7bd2 0x7ff982d8ce4f
******************************************************************
```

eine, 2020-10-10:
@targeted, even though the crash should not happen, `use work.ent.ent;` does not make sense to me. On the one hand, there is no `ent` package in the library. On the other hand, you don't need to "import" an entity for declaring the architecture in a different file.

Anyway, I confirm the crash:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1026-g73d2b4ce) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe -PC:\msys64\mingw64\lib/ghdl\ieee\v93\ -PC:\msys64\mingw64\lib/ghdl\ -c -o arch.o arch.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:859
Call stack traceback locations:
0x141287496 0x140073ed1 0x1400f8f4d 0x1400f916e 0x140139510 0x1401397d3 0x140139d03 0x1401214ff 
0x140235436 0x1402370c5 0x140238995 0x1400026ea 0x1400013bf 0x1400014f4 0x7ffc43f66fd2 0x7ffc4425cebf
******************************************************************
```

targeted, 2020-10-10:
@eine thank you :) Because ghdl is so fast, I usually run it even after smaller changes, during refactoring, to see where things break. The syntax may be incoherent, and if it goes well, compiler complains and I fix the problem. But sometimes it crashes, as you could see from my last reports.

eine, 2020-10-10:
@targeted, that's ok. However, it is important to be accurate in the descriptions. If you know that some example is not valid VHDL, please explain it explicitly. In this case, the title says that the error is produced when referencing an entity from a separate file. That's not exactly true because `work.ent.ent` is not how you do it. Doing such reference properly does not produce a crash. Therefore, there are potentially thousands of invalid uses of the language which might crash GHDL but which are very low priority compared to legit usages that are unsupported yet. It would be helpful for other users to know what you wanted to achieve (the title) and whether you could achieve it (which I think you did).

closed 2020-10-10 by tgingold

tgingold, 2020-10-11:
Now fixed.  Thank you for the issue.

For refactoring, I have also found that VS Code + vhdl language server are handy.

commit f62a2c0b (2020-10-10): vhdl-sem_scopes: avoid a crash on invalid selected name.  Fix #1490