ghdl/ghdl #1494: Can multiple programs interact with libghdl simultaneously?

m-kru, 2020-10-19:
Can multiple programs interact with libghdl simultaneously? Is it safe?
Imagine following scenario:

[image]

There are 2 processes that interact with libghdl.
Process 0 starts working as the first one. It analyzes 2 libraries foo and bar using libghdl.
Then Process 1 starts working.
It wants to analyze the same version of library foo, however different version of library bar.
Library bar version 2.0 has the same design units as bar version 1.0, however they behave in different way.
Can something bad happen in this scenario?
If I understand correctly libghdl stores analysis results internally. Can you confirm this?
Does analyzing library foo twice make any sense?
Do processes working with libghdl have any context associated with them within libghdl?

tgingold, 2020-10-25:
Each processes work independently.  Unless you try to save a library (which is not common), there cannot be any conflict between processes.

m-kru, 2020-10-25:
What does it mean to `save a library`? Where is it saved?

tgingold, 2020-10-26:
When libghdl is used, there is only one possible side effect: writing the library file (the .cf file).
This side effect can modify an existing library or create a new one that would be visible from other users of libghdl.

To save a library, you need to call a procedure which (IIRC) is not currently exported to the python interface.

closed 2021-09-07 by tgingold