ghdl/ghdl #1495: Attempting to compile UVVM (2020.10.06) fails with a Segmentation fault

stfarley, 2020-10-22:
**Description**
Compile failure

**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
run the compile_uvvm script

**Context**

- OS: Centos 7
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ X] Built from sources: `commit SHA`
b27832d75291c3288f59d648cfa83b74152adc38
ghdl --version
GHDL 1.0-dev (v0.37.0-1060-gb27832d) [Dunoon edition]
 Compiled with GNAT Version: Community 2020 (20200818-93)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
/projects/UVVM_All/uvvm_util/script/../src/methods_pkg.vhd:6772:0: internal compiler error: Segmentation fault
     return v_byte_array(0 to v_num_bytes-1);
 ^
0xa89f1f crash_signal
	/projects/gcc-5.5.0/gcc/toplev.c:381
0x862417 size_binop_loc(unsigned int, tree_code, tree_node*, tree_node*)
	/projects/gcc-5.5.0/gcc/fold-const.c:1786
0x830792 get_inner_reference(tree_node*, long*, long*, tree_node**, machine_mode*, int*, int*, bool)
	/projects/gcc-5.5.0/gcc/expr.c:6868
0x8639eb fold_unary_loc(unsigned int, tree_code, tree_node*, tree_node*)
	/projects/gcc-5.5.0/gcc/fold-const.c:8011
0x864218 fold_build1_stat_loc(unsigned int, tree_code, tree_node*, tree_node*)
	/projects/gcc-5.5.0/gcc/fold-const.c:14361
0x7002ce trans__chap7__translate_expression
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap7.adb:4371
0x7064ea trans__chap7__translate_expression__2
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap7.adb:4272
0x72373d trans__chap8__translate_return_statement
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap8.adb:251
0x730cde trans__chap8__translate_statement
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap8.adb:4976
0x730fae trans__chap8__translate_statements_chain_has_return
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap8.adb:5079
0x71668b trans__chap2__translate_subprogram_body
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap2.adb:681
0x721559 trans__chap4__translate_declaration_chain_subprograms
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap4.adb:2571
0x71777b trans__chap2__translate_package_body
	/projects/gcc-5.5.0/gcc/vhdl/trans-chap2.adb:979
0x70df9b translation__translate
	/projects/gcc-5.5.0/gcc/vhdl/translation.adb:306
0x73caac ortho_front__do_compile
	/projects/gcc-5.5.0/gcc/vhdl/ortho_front.adb:406
0x73caac ortho_front__parse
	/projects/gcc-5.5.0/gcc/vhdl/ortho_front.adb:656
0x6d053d lang_parse_file
	/projects/gcc-5.5.0/gcc/vhdl/ortho_gcc.adb:125
0x5f1849 ortho_parse_file
	/projects/gcc-5.5.0/gcc/vhdl/ortho-lang.c:483

**Additional context**
UVVM version 2020.10.06.
hash 160ce4f28766eb4b7f711c7e0c4948eac4d2e447

Paebbels, 2021-01-13:
See also #1593.

avelure, 2021-03-25:
This is not related to the issue in #1593 (except for both appearing in UVVM), but is instead related to VHDL 2008 unconstrained arrays. The issue is only present in the GCC backend. Tested on `Nightly GHDL 2.0.0-dev (v1.0.0-89-g7722e309)`

MWE:
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
end entity ent;

architecture beh of ent is
  type t_slv_array is array (natural range <>) of std_logic_vector;
  function test
    return t_slv_array is
    variable v_num_bytes   : integer := 2;
    variable v_byte_array  : t_slv_array(0 to v_num_bytes-1)(7 downto 0);
  begin
    return v_byte_array(0 to v_num_bytes-1);
  end function;
begin
end architecture beh;
```

```sh :image: ghdl/ghdl:buster-gcc
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent
```

```
/tmp/test.vhd: In function 'work__test5__ARCH__beh__test':
/tmp/test.vhd:15: internal compiler error: Segmentation fault
   15 |         return v_byte_array(0 to v_num_bytes-1);
      |
0xb9bce3 crash_signal
        ../../gcc-srcs/gcc/toplev.c:326
0x91170e size_binop_loc(unsigned int, tree_code, tree_node*, tree_node*)
        ../../gcc-srcs/gcc/fold-const.c:1905
0x8d9713 get_inner_reference(tree_node*, poly_int_pod<1u, long>*, poly_int_pod<1u, long>*, tree_node**, machine_mode*, int*, int*, int*)
        ../../gcc-srcs/gcc/expr.c:7290
0x912b52 fold_unary_loc(unsigned int, tree_code, tree_node*, tree_node*)
        ../../gcc-srcs/gcc/fold-const.c:7942
0x912f1c fold_build1_loc(unsigned int, tree_code, tree_node*, tree_node*)
        ../../gcc-srcs/gcc/fold-const.c:12347
0x79af92 trans__chap7__translate_expression
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap7.adb:4373
0x7a135a trans__chap7__translate_expression__2
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap7.adb:4274
0x778f3f trans__chap8__translate_return_statement
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap8.adb:249
0x778f3f trans__chap8__translate_statement
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap8.adb:4974
0x77a43e trans__chap8__translate_statements_chain_has_return
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap8.adb:5077
0x7665eb trans__chap2__translate_subprogram_body
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap2.adb:679
0x78f524 trans__chap4__translate_declaration_chain_subprograms
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap4.adb:2571
0x781142 trans__chap9__translate_block_subprograms
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap9.adb:1120
0x76a053 trans__chap1__translate_architecture_body
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/trans-chap1.adb:315
0x7485b4 translation__translate
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/translation.adb:315
0x73a4cc ortho_front__do_compile
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/ortho_front.adb:404
0x73a4cc ortho_front__parse
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/ortho_front.adb:656
0x738069 lang_parse_file
        /work/build-gcc-9.3.0/gcc-srcs/gcc/vhdl/ortho_gcc.adb:123
0x64b9f9 ortho_parse_file
        ../../gcc-srcs/gcc/vhdl/ortho-lang.c:482
```

The following removes the issue:
- Changing v_num_bytes to a constant instead of a variable
- Setting the declaration of v_byte_array to have a fixed range `variable v_byte_array  : t_slv_array(0 to 2)(7 downto 0);`

If the return is changed to have a fixed range  `return v_byte_array(0 to 1);` you get another backtrace
```
/tmp/test.vhd: In function 'work__test5__ARCH__beh__test':
/tmp/test.vhd:15: internal compiler error: Segmentation fault
   15 |         return v_byte_array(0 to 2);
      |
0xb9bce3 crash_signal
        ../../gcc-srcs/gcc/toplev.c:326
0x911708 size_binop_loc(unsigned int, tree_code, tree_node*, tree_node*)
        ../../gcc-srcs/gcc/fold-const.c:1900
0x97077d gimplify_compound_lval
        ../../gcc-srcs/gcc/gimplify.c:2934
0x96ae9b gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12461
0x96bd52 gimplify_addr_expr
        ../../gcc-srcs/gcc/gimplify.c:6076
0x96bd52 gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12556
0x969a53 gimplify_modify_expr
        ../../gcc-srcs/gcc/gimplify.c:5671
0x96b3fa gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12509
0x96e0da gimplify_stmt(tree_node**, gimple**)
        ../../gcc-srcs/gcc/gimplify.c:6719
0x96c1cb gimplify_statement_list
        ../../gcc-srcs/gcc/gimplify.c:1794
0x96c1cb gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12938
0x96e0da gimplify_stmt(tree_node**, gimple**)
        ../../gcc-srcs/gcc/gimplify.c:6719
0x96ea7f gimplify_bind_expr
        ../../gcc-srcs/gcc/gimplify.c:1362
0x96bae9 gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12710
0x96e0da gimplify_stmt(tree_node**, gimple**)
        ../../gcc-srcs/gcc/gimplify.c:6719
0x96c1cb gimplify_statement_list
        ../../gcc-srcs/gcc/gimplify.c:1794
0x96c1cb gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12938
0x96e0da gimplify_stmt(tree_node**, gimple**)
        ../../gcc-srcs/gcc/gimplify.c:6719
0x96ea7f gimplify_bind_expr
        ../../gcc-srcs/gcc/gimplify.c:1362
0x96bae9 gimplify_expr(tree_node**, gimple**, gimple**, bool (*)(tree_node*), int)
        ../../gcc-srcs/gcc/gimplify.c:12710
```

avelure, 2021-09-14:
I believe this can be closed, my MWE no longer fails and the compile_uvvm script passes.

tgingold, 2021-09-15:
Thank you for the status update!

commit efcf7034 (2021-09-15): testsuite/gna: add a test and close #1495
closed 2021-09-15 by tgingold