ghdl/ghdl #1496: Problems with generic circuits and aggregates in 2008 mode

ti6wb, 2020-10-22:
```VHDL
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity adder is
generic (N: integer := 4);  
port( a,b : in  std_logic_vector(N-1 downto 0);
      cin : in  std_logic;
      sum : out std_logic_vector(N-1 downto 0); 
      cout: out std_logic
      );
end;
architecture rtl2008 of adder is
begin
  (cout, sum) <= std_logic_vector(unsigned('0' & a) + unsigned('0' & b) + cin);
end;  
```
Simulation works, but synthesis with yosys gives me an error (see below). 
If I use fixed arrays, e.g. (3 downto 0) everything is ok.

I use the following yosys synthesis script:

synth.ys:    ghdl --std=08 -gN=4 adder

Starting the yosys synthesis:

yosys -m ghdl synth.ys 

results in:

Yosys 0.9+3541 (git sha1 04d5692a, clang 6.0.1-svn334776-1~exp1~20190309042730.123 -fPIC -Os)

-- Executing script file `synth.ys' --

1. Executing GHDL.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-821-g99b542c8) [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-pc-linux-gnu

Command line:

Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-objtypes.adb:365
******************************************************************
ERROR: vhdl import failed.

ti6wb, 2020-10-22:
Of course there is a workaround, but I was already interested to find out where the problem lies. For a beginner it might also be difficult to find out if his code does not comply with the VHDL2008 specification or if it is a compiler bug. 

Work around:
```VHDL
architecture rtl2008 of adder is
  signal r: std_logic_vector(N downto 0);
begin
  r    <= std_logic_vector(unsigned('0' & a) + unsigned('0' & b) + cin);
  sum  <= r(N-1 downto 0);
  cout <= r(N);
end;  
```

dalex78, 2026-08-20:
# Issue #1496 — Crash: generic circuits + aggregates in VHDL-2008 mode

Analysis by an AI agent.

## High-level summary

An adder using a target aggregate combining a scalar carry-out with a generic-sized vector sum (`(cout, sum) <= std_logic_vector(unsigned('0' & a) + unsigned('0' & b) + cin);`, with `a`/`b`/`sum` sized by a generic `N`) crashed `ghdl --synth` with an internal `TYPES.INTERNAL_ERROR` at `synth-objtypes.adb:365`. The reporter noted using a fixed-size array instead of a generic worked fine.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth --std=08 -gN=4 adder` now produces a complete, correct netlist (full carry-out adder logic: two zero-extended additions, the carry bit extracted from bit 4, the sum from bits 3 downto 0) instead of crashing. This must have been fixed as part of general improvements to generic-sized aggregate-target handling in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1496/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design (with `-gN=4`, matching the report) and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this ticket be closed ? (PR merged)

closed 2026-08-24 by tgingold