ghdl/ghdl #1499: PSL error on clock name

Bcuzeau, 2020-10-24:
When the clock is named "clock", this poses a problem to ghdl :
--psl default clock is rising_edge(Clock);
-> _design.vhd:41:36: unexpected token 'clock' in a primary_
but the workaround below works :
--psl default clock is rising_edge(Clk);

tgingold, 2020-10-24:
Well, clock is a PSL keyword.

Bcuzeau, 2020-10-24:
After default clock, there must be a boolean *. 
rising_edge(Clock) is a boolean since VHDL can use a PSL keyword for a signal.
* I think the LRM says : 
_In the VHDL flavor, any expression that **VHDL allows** to be used as the condition in an if statement can be used as a clock expression_
In any case, Questa (& other simulators) has no problem with rising_edge(Clock) as default clock.
So I think it would be better if it worked in GHDL (though I'm more interested in seeing PSL operators support by simulation ;-)

manu38gre, 2020-10-25:
Hello,
I think PSL keywords have become part of the langage since VHDL 2008.
Embedding PSL statements using comments is still a very common practice, but it was a kind of workaround when PSL support was introduced in commercial simulators (and considered as a different language). Now we should consider that PSL language is a VHDL 2008 construct when embedded in VHDL source code, even if embedded comments are used.
I assume that you are testing GHDL on an existing code base and don't want to modify entities interfaces. Perhaps using an alias or an intermediate signal should help you ?

Bcuzeau, 2020-10-25:
Hi,
I indeed had no problem working around this issue (with changes in the code), but the matter is whether it is legitimate or not to reject this statement and force the user in modifying his code.
Note that I still have to find a single simulator that does support PSL code in VHDL without the pragma... (indeed with -2008 compilation active). You know one ? 
anyway, at this point, it's an academic point since GHDL cannot simulate PSL (and, in fact, most commercial simulators except Questa can't really either). IMO, using assertions (ABV) today means in practice using SVA (which is absolutely integrated in SV and supported by all commercial simulators).

So I think it would be a huge service to the VHDL users community if GHDL could simulate PSL ! And it would make absolute sense with GHDL not having a gui nor integrated waveform viewer since assertions do not need them (just an assertion coverage report) and they provide a lot of verification info.

tmeissner, 2020-10-25:
@Bcuzeau Sorry, that is not correct. You can use PSL for simulation & synthesis with GHDL. Only a few constructs cannot be used in simulation, PSL functions for example. In synthesis, some of these are implemented, as it is much simpler, even I without much knowledge of the GHDL codebase could do that (see #1353 .

I know both languages, SVA & PSL. I only use PSL at work (with Questa), I never had a case where it could not be solved in PSL. I already used GHDL (with SymbiYosys) for formal verification of code at work. Most of the time you cannot use the same assertions from simulation for your formal runs, no matter if you use SVA or PSL. You have to simplify them, as even commercial tools don't support all constructs of PSL & SVA for formal verification.

Bcuzeau, 2020-10-25:
I don't agree with you about the methodology. 

I don't see how you can use properties in formal tools that you are not able to simulate...
(what a waste of time to try and prove formally assertions that fail with a test bench !)

Likewise, vunits are good for some use of assertions (and obviously for black box testing), but the only way in practice to have enough (valid + density) assertions checking the internal RTL code is to locate them closest to what they check.

Apart from Questa, the $$$ simulators focus on where there are $$$ (ASICs) and it's where SV (& SVA & UVM) is predominant, and where PSL in the noise floor (Wilson Research 2020) and VHDL is very low.

btw : SVA is PSL (Sugar) rolled into SV syntax, same foundation layer & engine etc. Only the syntax differ and a few extra features. If you know one, you know the other. What I wrote is that SVA is better supported by most simulators than PSL.

I tested GHDL PSL support a bit further and it was a good surprise that it seems to support the temporal layer as long as functions are not used (rose, fell, prev etc) :
--psl default clock is rising_edge(Clk);
--psl Rstshort : assert always (Reset -> next_e[1 to 5] (not Reset));
--psl Unt: assert always (Reset -> (Cnt= 0) until_ not Reset);
--psl Bef: assert always (Reset -> next (Load before Reset) );
So it's even more frustrating that standard PSL functions are not supported by GHDL, since I suppose the temporal operators are more complex to implement.

You say prev() or Stable are difficult to implement in simulation, I fail to see why :
PrevCnt <= Cnt when rising_edge(Clk); -- that what synthesis does, no ?
Stable <= PrevCnt=Cnt;
I just don't want to hack my code to support the limitation of a tool.

I keep hoping, always the optimist :-)

closed 2020-10-25 by Bcuzeau

tmeissner, 2020-10-25:
@Bcuzeau Don't get me wrong, I also not always satisfied with the PSL support in the vendor tools 😉 

I have a repo which tries to summarize and test the PSL features GHDL implements: [PSL with GHDL](https://github.com/tmeissner/psl_with_ghdl).

What I meant with different assertions for formal and simulation are two things:

#### Complex properties which can't be verified by the formal solvers, but can be simulated very easily

Some complex properties, for example with replication or complex data checks can be simulated (maybe with slow performance), but can't be verified by the formal solvers in finite time. You have to rewrite them in simpler form, maybe split them in more properties.

#### Syntax which is supported in simulation. but not by the formal tools

AFAIR even Questa doesn't support `forall` in the formal apps, for example. I evaluated their Propcheck tool some time ago. It wasn't possible to use my existing AXI PSL-checkers, as they included constructs supported by Questa Sim, but not by Questa Formal. I don't know if this is the case for SVA also.

eine, 2020-10-25:
> but the matter is whether it is legitimate or not to reject this statement and force the user in modifying his code.
> Note that I still have to find a single simulator that does support PSL code in VHDL without the pragma...

I feel that the attention was brought elsewhere and this issue was closed before answering these points:

- Is it legitimate to reject the statement because `clock` is a reserved word?
  - Do other vendors support it anyway? If so, is this to be supported as a relaxed option?
- Is GHDL the only simulator known to support https://ghdl.github.io/ghdl/using/ImplementationOfVHDL.html#psl-annotations-vhdl-2008-only and/or https://ghdl.github.io/ghdl/using/ImplementationOfVHDL.html#psl-vunit-files-vhdl-2008-synthesis-only ?

> You say prev() or Stable are difficult to implement in simulation, I fail to see why :
> PrevCnt <= Cnt when rising_edge(Clk); -- that what synthesis does, no ?
> Stable <= PrevCnt=Cnt;
> I just don't want to hack my code to support the limitation of a tool.

You might want to have a look at the latest PRs regarding PSL: https://github.com/ghdl/ghdl/pulls?q=is%3Apr+sort%3Aupdated-desc+is%3Aclosed+label%3A%22Feature%3A+PSL%22