ghdl/ghdl #1500: Array assignment with a single value and others misbehaves w/o a comma

targeted, 2020-10-28:
**Description**
The code compiles and produces invalid result.

**Expected behaviour**
It should either fail to compile or assign the value to the first element of the array.

**How to reproduce?**
Compile and run the attached code. The problem is the missing comma in the array assignment. 

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test is
end test;

architecture test of test is

    subtype byte is unsigned (7 downto 0);
    type arr_t is array (0 to 1) of byte;
    signal arr: arr_t;

begin

    process is
    begin
        arr <= (B"11111111" others => B"00000000");
        --               ^^^^ no comma
        wait for 1 ns;
        report to_string(arr(0)); -- reports 00000000
        wait;
    end process;

end test;
```

```sh
ghdl -a --std=08 test.vhd
ghdl --elab-run test
```
**Context**
Please, provide the following information:

- OS: Windows
- GHDL 1.0-dev (v0.37.0-996-g498ee9b6) [Dunoon edition]

tgingold, 2020-10-29:
Yes, I confirm the issue.

closed 2020-10-29 by tgingold

tgingold, 2020-10-29:
This is now fixed.  Thank you for the report.

commit 997d378b (2020-10-29): vhdl-parse: do not skip token in case of error.  Fix #1500