ghdl/ghdl #1502: synth: raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-expr.adb:830

endofexclusive, 2020-10-31:
**Description**
The example below triggers assertion with `ghdl --synth`.

**Expected behaviour**
It was expected that `ghdl --synth` would output a VHDL netlist or an error message.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: theunit.vhd
library ieee;
use ieee.std_logic_1164.all;

entity passthrough is
  port (
    wdata : in  std_ulogic_vector(1 downto 0);
    rdata : out std_ulogic_vector(1 downto 0)
  );
end;

architecture rtl of passthrough is
begin
  rdata <= wdata;
end;

library ieee;
use ieee.std_logic_1164.all;

entity theunit is
  port (
    inx   : in    std_ulogic_vector(1 downto 0);
    utx   : out   std_ulogic_vector(1 downto 0)
  );
end;

architecture rtl of theunit is
  type tdata_t is record
    hello : std_ulogic_vector(123 downto 0);
  end record;

  subtype tdatavec_t is std_ulogic_vector(1 downto 0);

  function encodex(u : tdata_t) return std_ulogic_vector is
    variable ret : tdatavec_t;
  begin
    ret(1) := u.hello(101);
    ret(0) := u.hello(100);
    return ret;
  end;

  signal wtdata  : tdata_t;

begin
  wtdata.hello <= (others => '0');

  ram: entity work.passthrough
  port map (
    wdata => encodex(wtdata),
    rdata => utx
  );
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a theunit.vhdl
ghdl --synth theunit
```

**Context**
Please, provide the following information:

- OS: Debian GNU/Linux 10 (buster)
- Origin:
  - [x] Built from sources: `9207342e0b89ba07c123eb5e20a360035ad5ebb0`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1068-g9207342e) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/home/asdf/repos/ghdl/sbug5/
Command line:
ghdl --synth -v theunit
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-expr.adb:830
Call stack traceback locations:
0x7f6dbbada4f1 0x5598154d2544 0x5598154ac616 0x5598154ac98f 0x5598154acef0 0x5598154ad00c 0x5598154bfcfd 0x5598154bfdfb 0x5598154ae8b1 0x5598154aea18 0x5598154a7b5a 0x55981552c0df 0x5598154374a7 0x559815535b0f 0x5598152ddfc5 0x7f6dbb6ea099 0x5598152dcdf8 0xfffffffffffffffe
******************************************************************
```

**Additional context**
```
$ ghdl --elab-run theunit
$ echo $?
0
```

tgingold, 2020-10-31:
Yes, the conversion function is not handled.  I will add that.

patrickrst, 2021-05-23:
I get the same error from synth when using conversion functions for records.

I started looking at the GHDL codebase, mostly under `src/synth`. Do you have an idea how big of a change would it be to add this?

patrickrst, 2021-10-05:
New exception, but same problem I believe

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r813.g36d8c01d) [Dunoon edition]
Compiled with GNAT Version: Community 2021 (20210519-103)
Target: x86_64-pc-linux-gnu
/home/user/ghdl-bug/
Command line:
ghdl --synth theunit
Exception ADA.ASSERTIONS.ASSERTION_ERROR raised
Exception information:
raised ADA.ASSERTIONS.ASSERTION_ERROR : synth-vhdl_expr.adb:831
Call stack traceback locations:
0x7b6cc0 0x6af813 0x6c419b 0x6c4b60 0x6c50cf 0x6c520a 0x6a4256 0x6a4389 0x6c6c08 0x6c6d20 0x6bf8e1 0x6f553e 0x628b8e 0x79aa2b 0x408915 0x7f565a7c3553 0x4070be 0xfffffffffffffffe
******************************************************************
```

dalex78, 2026-08-19:
# Issue #1502 — `ASSERT_FAILURE : synth-expr.adb:830`

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

`ghdl --synth` crashed with an internal `ASSERT_FAILURE` at `synth-expr.adb:830` synthesizing a design where a function (`encodex`) extracts two bits from a record field (`u.hello(101)`/`u.hello(100)`) into a small return array, used directly as an actual in a port map association.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth theunit` now produces a complete, valid netlist (the function call is inlined into two single-bit extractions from `wtdata` concatenated back into a 2-bit vector for the `passthrough` instance's `wdata` port) instead of crashing. This must have been fixed as part of general improvements to function-call inlining / record-field extraction in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1502/` (the exact repro from the report), whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist (the standard `testsuite/synth` sanity check: the netlist itself must be valid VHDL). This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold

dalex78, 2026-08-20:
Address the second part of https://github.com/ghdl/ghdl/issues/1502 that was missed previously.

I made a mistake on some of the tickets for which I opened PRs. When letting the AI agent work on it, I didn't realize that it was not looking at the conversations several people has in the thread. Therefore, I asked him to go back and re-analyze the tickets.

# Issue #1502 — Crash on a conversion function in a port association

Analysis by an AI agent.

## High-level summary

A port association whose actual is produced by a conversion function crashed `ghdl --synth` with an internal `ASSERT_FAILURE` at `synth-expr.adb:830`. The maintainer confirmed the cause immediately on the thread:

> Yes, the conversion function is not handled. I will add that.

A second reporter later hit the same crash from a different direction:

> I get the same error from synth when using conversion functions for records.

and, some months on, a further exception from the same area (`synth-vhdl_expr.adb:831`).

## Current status: fixed on master

Both shapes work on current `master`.

The report's design — the actual being an array taken from a record field, produced by a function call — synthesizes into a complete, valid netlist that re-analyzes cleanly.

The record variant from the thread — a conversion function whose parameter and result are both records, used on a port association — also synthesizes cleanly. It was reconstructed here, since that reporter did not attach code.

No source change is needed. Conversion functions on associations are handled in `src/synth` now, which is what the maintainer said would be added.

## What this commit does

No source fix. `testsuite/synth/issue1502/` covers both shapes from the thread rather than only the original report:

- `theunit.vhdl` — the report's design; synthesized and the netlist re-analyzed.
- `rec.vhdl` — the record-conversion variant; synthesized and checked for the absence of the crash. It is not re-analyzed because the netlist still refers to the package declaring the record types, which synthesis does not emit.

Both pass on mcode, llvm and gcc.