ghdl/ghdl #1504: 'unaffected' crashes ghdl synth

eduardosm, 2020-11-11:
**Description**
Using `unaffected` in a `when` statement crashes `synth`.

**How to reproduce?**

Minimal example:
```vhd :file: test1.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        clk: in std_ulogic;
        i: in std_ulogic;
        o: out std_ulogic
    );
end entity;

architecture arch of ent is
begin
    o <= i when rising_edge(clk) else unaffected;
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl synth test1.vhd -e ent
```

**Context**

- OS: Manjaro Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit g267ceb3b`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1044-g267ceb3b) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/tmp
Command line:
ghdl synth test1.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-stmts.adb:75
Call stack traceback locations:
0x55eece4cc8e1 0x55eece4cfeef 0x55eece4ddbe2 0x55eece4dde1d 0x55eece50749a 0x55eece507627 0x55eece50007c 0x55eece539b37 0x55eece454b28 0x55eece5f1e74 0x55eece201ff0 0x7f4562b46150 0x55eece20072c 0xfffffffffffffffe
******************************************************************
```

eine, 2020-11-11:
I think this is a legit bug; however, you could work around it with `o <= i when rising_edge(clk);`.

dalex78, 2026-08-19:
# Issue #1504 — `unaffected` crashes `ghdl --synth`

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

`o <= i when rising_edge(clk) else unaffected;` crashed `ghdl --synth` with an internal `TYPES.INTERNAL_ERROR` at `synth-stmts.adb:75`.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl synth test1.vhd -e ent` now produces a correct netlist, recognizing the `rising_edge`-gated conditional assignment with an `unaffected` else-branch as an edge-triggered process:

```vhdl
process (wrap_clk)
begin
  if rising_edge (wrap_clk) then
    n3 <= wrap_i;
  end if;
end process;
```

instead of crashing. This must have been fixed as part of general improvements to `unaffected`/clock-edge handling in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1504/` (the exact repro from the report), whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold