ghdl/ghdl #1505: Invalid ROM generated with 2-dimensional array with non power of two dimensions

eduardosm, 2020-11-11:
**Description**
Definind a 2-dimension array constant whose dimensions are not power of two, the synthesized code is invalid.

**How to reproduce?**

Minimal example:
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        x: in integer range 0 to 9;
        y: in integer range 0 to 9;
        o: out std_ulogic
    );
end entity;

architecture arch of ent is
    type arr is array (0 to 9, 0 to 9) of std_ulogic;
    constant rom: arr := (
        ('0', '0', '1', '0', '0', '0', '0', '1', '0', '0'),
        ('0', '0', '0', '1', '0', '0', '1', '0', '0', '0'),
        ('1', '0', '0', '0', '1', '1', '0', '0', '0', '1'),
        ('1', '0', '1', '1', '1', '1', '1', '1', '0', '1'),
        ('1', '1', '1', '0', '1', '1', '0', '1', '1', '1'),
        ('0', '1', '1', '1', '1', '1', '1', '1', '1', '0'),
        ('0', '0', '1', '1', '1', '1', '1', '1', '0', '0'),
        ('0', '0', '1', '0', '0', '0', '0', '1', '0', '0'),
        ('0', '0', '1', '0', '0', '0', '0', '1', '0', '0'),
        ('0', '0', '0', '1', '0', '0', '1', '0', '0', '0')
    );
begin
    o <= rom(y, x);
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl synth ent.vhd -e ent
```

The `synth` output is:
```vhd
library ieee;
use ieee.std_logic_1164.all;
entity ent is
  port (
    x: in integer range 0 to 9;
    y: in integer range 0 to 9;
    o: out std_ulogic
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of ent is
  subtype typwrap_x is std_logic_vector (3 downto 0);
  signal wrap_x: typwrap_x;
  subtype typwrap_y is std_logic_vector (3 downto 0);
  signal wrap_y: typwrap_y;
  signal wrap_o: std_logic;
  signal n2_o : std_logic_vector (3 downto 0);
  signal n5_o : std_logic_vector (3 downto 0);
  signal n11_o : std_logic_vector (7 downto 0);
  signal n12_data : std_logic;
begin
  wrap_x <= std_logic_vector(to_unsigned(x, 4));
  wrap_y <= std_logic_vector(to_unsigned(y, 4));
  o <= wrap_o;
  wrap_o <= n12_data;
  -- test3.vhd:27:13
  n2_o <= std_logic_vector (unsigned'("1001") - unsigned (wrap_x));
  -- test3.vhd:27:13
  n5_o <= std_logic_vector (unsigned'("1001") - unsigned (wrap_y));
  -- test3.vhd:8:9
  process (n11_o) is
    type n10_oport_type is array (0 to 99)
      of std_logic;
    variable n10_oport : n10_oport_type :=
      (99 => '0',
       98 => '0',
       97 => '1',
       96 => '0',
       95 => '0',
       94 => '0',
       93 => '0',
       92 => '1',
       91 => '0',
       90 => '0',
       89 => '0',
       88 => '0',
       87 => '0',
       86 => '1',
       85 => '0',
       84 => '0',
       83 => '1',
       82 => '0',
       81 => '0',
       80 => '0',
       79 => '1',
       78 => '0',
       77 => '0',
       76 => '0',
       75 => '1',
       74 => '1',
       73 => '0',
       72 => '0',
       71 => '0',
       70 => '1',
       69 => '1',
       68 => '0',
       67 => '1',
       66 => '1',
       65 => '1',
       64 => '1',
       63 => '1',
       62 => '1',
       61 => '0',
       60 => '1',
       59 => '1',
       58 => '1',
       57 => '1',
       56 => '0',
       55 => '1',
       54 => '1',
       53 => '0',
       52 => '1',
       51 => '1',
       50 => '1',
       49 => '0',
       48 => '1',
       47 => '1',
       46 => '1',
       45 => '1',
       44 => '1',
       43 => '1',
       42 => '1',
       41 => '1',
       40 => '0',
       39 => '0',
       38 => '0',
       37 => '1',
       36 => '1',
       35 => '1',
       34 => '1',
       33 => '1',
       32 => '1',
       31 => '0',
       30 => '0',
       29 => '0',
       28 => '0',
       27 => '1',
       26 => '0',
       25 => '0',
       24 => '0',
       23 => '0',
       22 => '1',
       21 => '0',
       20 => '0',
       19 => '0',
       18 => '0',
       17 => '1',
       16 => '0',
       15 => '0',
       14 => '0',
       13 => '0',
       12 => '1',
       11 => '0',
       10 => '0',
       9 => '0',
       8 => '0',
       7 => '0',
       6 => '1',
       5 => '0',
       4 => '0',
       3 => '1',
       2 => '0',
       1 => '0',
       0 => '0');
  begin
    n12_data <= n10_oport(to_integer (unsigned (n11_o)));
  end process;
  n11_o <= n5_o & n2_o;
  -- test3.vhd:27:17
end rtl;
```

The `n11_o <= n5_o & n2_o` should be something like `n11_o <= n5_o * 10 + n2_o`.

**Context**

- OS: Manjaro Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit g267ceb3b`

eine, 2020-11-11:
@eduardosm, `n11_o <= n5_o & n2_o` is equivalent to `n11_o <= n5_o * 16 + n2_o`. I didn't look further into the output; but does that match your expectations? Other than that, did you actually simulate the output to check the behaviour?

eduardosm, 2020-11-11:
> @eduardosm, `n11_o <= n5_o & n2_o` is equivalent to `n11_o <= n5_o * 16 + n2_o`. I didn't look further into the output; but does that match your expectations?

But the dimension is 10, not 16, so it should be multiply by 10, not 16.

> Other than that, did you actually simulate the output to check the behaviour?

I found this bug while synthesizing with the ghdl-yosys plugin and testing on a FPGA.

eine, 2020-11-12:
I synthesised `ent.vhd` and changed the entity name of the output to `synth`. I then tried the following testbench:

```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity tb is
end entity;

architecture arch of tb is

  signal x, y : integer range 0 to 9;
  signal r, s : std_ulogic;

begin

  process
  begin
    for ix in 0 to 9 loop
      x <= ix;
      for iy in 0 to 9 loop
        y <= iy;
        wait for 10 ns;
      end loop;
    end loop;
    wait;
  end process;

  ref: entity work.ent
  port map (
    x => x,
    y => y,
    o => r
  );

  synth: entity work.synth
  port map (
    x => x,
    y => y,
    o => s
  );

end architecture;
```

Simulation fails with:

```vhdl
# ./tb --wave=wave.ghw
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
../../src/ieee/v93/numeric_std-body.vhdl:2098:7:@0ms:(assertion warning): NUMERIC_STD.TO_INTEGER: metavalue detected, returning 0
T:\vunit\vunit\tmp\tb.exe:error: index (153) out of bounds (0 to 99) at synth.vhd:140
in process .tb(arch).synth@synth(rtl).P6
T:\vunit\vunit\tmp\tb.exe:error: simulation failed
```

eduardosm, 2024-12-15:
Test added in https://github.com/ghdl/ghdl/pull/2825

closed 2024-12-15 by eduardosm