ghdl/ghdl #1506: Running a tb with ghdl -r says "use --ieee=synopsys"

philtomson, 2020-11-13:
I added the following line to my VHDL testbench:
    use IEEE.std_logic_textio.all;

Then compiled analyzed and elaborated the tb like so:

$ ghdl -a --ieee=synopsys -frelaxed-rules hplc_tb.vhd
$ ghdl -e --ieee=synopsys -frelaxed-rules hplc_tb

Being sure to use --ieee=synopsys as std_logic_textio requires it.

Then when I try to run the testbench:

$ ghdl -r hplc_tb  --wave=hplc_tb.ghw 

I see: 

hplc_tb.vhd:33:10:error: unit "std_logic_textio" not found in library "ieee"
hplc_tb.vhd:33:10:error:  (use --ieee=synopsys or --std=08 for this non-standard synopsys package)

Why would I get these errors if the design was analyzed and elaborated using the --ieee=synopsys option?

$ ghdl --version
GHDL 0.36-dev (v0.35-128-gc72963ed) [Dunoon edition]

eine, 2020-11-13:
Hi @philtomson! The version of GHDL you are using is rather outdated (from march 2018). Can you please try with a more recent version?

BTW, how did you install GHDL? Did you build it from sources or you used the package manager of your distribution?

eine, 2020-11-13:
As explained in [November 13, 2020 7:44 AM](https://gitter.im/ghdl1/Lobby?at=5fae2b5f7cac87158fa33a2b), the issue is that the flags used for analysis and elaboration are not passed to the run command:

```sh
$ ghdl -r --ieee=synopsys -frelaxed-rules hplc_tb --wave=hplc_tb.ghw
```

closed 2020-11-13 by eine