ghdl/ghdl #1508: an integration for ubuntu ?

Aqueuse, 2020-11-14:
Could it be possible to have a package for ubuntu, with dependencies solving, etc ? I had real pain to install it last day and I was obligated to change for Fedora. And it's really not pratical to need to do this :-S

eine, 2020-11-14:
@Aqueuse, please let us know what steps did you follow, command by command, and which issues you found.

- GHDL is actively built and tested on Ubuntu 18 and 20: https://github.com/ghdl/docker/runs/1392267265?check_suite_focus=true.
- Pre-built packages are available in https://github.com/ghdl/ghdl/releases/tag/nightly.
- Specific build/runtime dependencies are listed in https://github.com/ghdl/docker/blob/master/build_debian.dockerfile and https://github.com/ghdl/docker/blob/master/run_debian.dockerfile.
- Although not recommended, GHDL is also available through `apt`/`apt-get`.

Hence, it sounds really strange that you were obligated to use Fedora. Fedora works too, but neither better nor worse than Ubuntu.

Aqueuse, 2020-11-14:
Hi @eine 

I have made a traditional apt-get update, then apt-get install ghdl and has I was not finding it, I have search "ghdl" with apt-cache search, with no result at all. I'm in ubuntu 18.04 via WSL.

I finaly managed to install it via the prebuild package I have find here, but it worked very bad and as I was executing a test for a friend, I went to this message in the console : 

```
llo@:/mnt/c/Users/Megaport/Desktop/LibreGates_20201108$ sh build.sh

Execution terminated by unhandled exception
Exception name: STORAGE_ERROR
Message: stack overflow or erroneous memory access
Call stack traceback locations:
0x7f2464dd288e 0x7ffffbd9d6f3
ghdl: compilation error
llo@:/mnt/c/Users/Megaport/Desktop/LibreGates_20201108$ sh run_tests.sh

This script will run exhaustive self-tests and will
take "a while", depending on your CPU and software.
(about one minute on my i7 PC)
You may interrupt it (CTRL+C) if needed. Please report errors.

--> going to run_tests.sh: 28: run_tests.sh: pushd: not found

Execution terminated by unhandled exception
Exception name: STORAGE_ERROR
Message: stack overflow or erroneous memory access
Call stack traceback locations:
0x7f14bdfd288e 0x7fffd70f2f13
ghdl: compilation error
<-- back to run_tests.sh: 30: run_tests.sh: popd: not found

```

I then install GHDL traditionnaly in a Fedora and the test execute flawless (it log very verbious but if you insist, I will copy you a log file). My hypothesis is that it lacked a package dependency in the ubuntu version, that was correctly managed in fedora, has it use a package manager.

eine, 2020-11-14:
> I have made a traditional apt-get update, then apt-get install ghdl and has I was not finding it, I have search "ghdl" with apt-cache search, with no result at all. I'm in ubuntu 18.04 via WSL.

GHDL is available through apt/apt-get on Ubuntu 20.04, but not on Ubuntu 18.04: https://packages.ubuntu.com/search?keywords=ghdl.

> I finaly managed to install it via the prebuild package I have find here, but it worked very bad

Which of the prebuilt packages did you install? Which specific dependencies did you install besides extracting the package?

> ```
> llo@:/mnt/c/Users/Megaport/Desktop/LibreGates_20201108$ sh build.sh
> llo@:/mnt/c/Users/Megaport/Desktop/LibreGates_20201108$ sh run_tests.sh
> ...
> --> going to run_tests.sh: 28: run_tests.sh: pushd: not found
> ...
> <-- back to run_tests.sh: 30: run_tests.sh: popd: not found
> ```

It is very difficult to understand what's going on without knowing what's the content of `build.sh` and `run_tests.sh`. My guess is that `pushd`/`popd` errors are related to WSL, not to GHDL.

> I then install GHDL traditionnaly in a Fedora and the test execute flawless (it log very verbious but if you insist, I will copy you a log file).

Maintainers of Fedora packages keep an active communication with the development of GHDL. That's why RPM packages exist for many versions of Fedora, and that's why all the required dependencies are installed automatically. However, once installed, GHDL should behave equally on any distro.

> My hypothesis is that it lacked a package dependency in the ubuntu version, that was correctly managed in fedora, has it use a package manager.

Unlike Fedora packagers, Debian/Ubuntu packagers work decoupled. As a result, DEB packages are not as consistent. The tarballs you downloaded from 'nightly' release are not DEB packages. Hence, you need to install the dependencies explicitly. That's why I asked about the specific set of commands/steps that you used for installing/extracting GHDL.

Aqueuse, 2020-11-14:
I share with you the files that I have execute (they are shell script of course, not txt, but github doesn't want that I share sh files so I have change the extension).

[run_tests.txt](https://github.com/ghdl/ghdl/files/5541053/run_tests.txt)
[build.txt](https://github.com/ghdl/ghdl/files/5541054/build.txt)

As for ubuntu, I have used Fedora in WSL. I have install with this file : https://github.com/ghdl/ghdl/releases/download/v0.37/ghdl-0.37-ubuntu16-llvm-3.9.tgz

I have tried to install the dependencies but with no success and it was an headache to do :'-( (libgnat for example), at the end, I dropped. Is there no way to make the process of installation for ubuntu a little easier ? Maybe an easy access to the list of dependencies and resulting files to install ?

At least, if it work well with ubuntu 20, it's a good news. I haven't find GHDL in raspbian too but I use an old version, so...

closed 2020-11-14 by Aqueuse

eine, 2020-11-14:
> For ubuntu, I have install GHDL with this file : [`v0.37` ghdl-0.37-ubuntu16-llvm-3.9.tgz (download)](https://github.com/ghdl/ghdl/releases/download/v0.37/ghdl-0.37-ubuntu16-llvm-3.9.tgz)

That tarball is for Ubuntu 16. Note the name. For Ubuntu 18, the equivalent would be: https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-18.04-llvm.tgz. The dependencies are: `libgnat-7`, `zlib1g-dev` and `libllvm5.0` (see https://github.com/ghdl/docker/blob/master/run_debian.dockerfile).

> I have tried to install the dependencies but with no success and it was an headache to do :'-( (libgnat for example), at the end, I dropped.

While I understand it is not "plug and play", dependencies are two only (zlib and gnat), plus LLVM (if you use that backend).

> Is there no way to make the process of installation for ubuntu a little easier ? Maybe an easy access to the list of dependencies and resulting files to install ?

It is possible to create non-official DEB files, just as packages for MSYS2 are created. However, our resources are limited and that's not a priority for active contributors. Hence, someone would need to step forward and help maintining it. See:

- #901
- #888 (coordination for Fedora)
- #507 (coordination for Arch)

List of dependencies are kept up to date in ghdl/docker, because GHDL is tested on several versions of Debian and Ubuntu.

> At least, if it work well with ubuntu 20, it's a good news.

Be aware that some packagers build GHDL with openieee libraries, instead of ieee. As a result, some common features are not available. I'm not sure whether that's on Debian only or on Ubuntu 20 too.

> I haven't find GHDL in raspbian too but I use an old version, so...

GHDL does work on raspbian, as I have used it both natively and through ARM containers. However, I'm not aware about pre-built packages being availables. I use multiarch images from [dbhi/docker](https://github.com/dbhi/docker).