ghdl/ghdl #1509: Synthesis produces incorrect representation for integers with certain ranges.

eduardosm, 2020-11-15:
**Description**

When there is an `integer range -X to X`, where `X` is a power of two, `ghdl synth` will create a wrong representation for this type.

If `X = 2^N`, a vector of length `N+1` is used to represent the type. This length would be correct to represent a range of `-4 to 3` or `0 to 4`, but not `-4 to 4`.

**Expected behaviour**

A vector of length `N+2` should be used.

**How to reproduce?**

Minimal example:

```vhd :file: test.vhd
entity pass is
    port (
        i: in integer range -4 to 4;
        o: out integer range -4 to 4
    );    
end entity;

architecture arch of pass is
begin
    o <= i;
end architecture;
```

Synthesized with:
```sh :image: ghdl/ghdl:buster-mcode
ghdl synth test.vhd -e pass
```

Produces the following code, where `wrap_i` and `wrap_o` have length 3 instead of 4:
```
entity pass is
  port (
    i: in integer range - 4 to 4;
    o: out integer range - 4 to 4
  );
end entity;

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

architecture rtl of pass is
  subtype typwrap_i is std_logic_vector (2 downto 0);
  signal wrap_i: typwrap_i;
  subtype typwrap_o is std_logic_vector (2 downto 0);
  signal wrap_o: typwrap_o;
begin
  wrap_i <= std_logic_vector(to_signed(i, 3));
  o <= to_integer (signed (wrap_o));
  wrap_o <= wrap_i;
end rtl;
```

**Context**
- OS: Manjaro Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit g31ca0d8a`

eduardosm, 2020-11-15:
I think the issue is in this function:
https://github.com/ghdl/ghdl/blob/31ca0d8a687f0b3cc5615407e2ee6160e4355baa/src/synth/synth-objtypes.adb#L123-L157

where `Width (Clog2 (Uns64 (Hi)))` should be `Width (Clog2 (Uns64 (Hi) + 1))`

eine, 2020-11-16:
`range -4 to 4` should hit L130-L131, so `Lo := -4` amd `Hi := 4`. Then, `Lo <= Hi`, `Lo < 0`, `Lo /= Int64'First` and `Hi >= 0`. Therefore, L149-L154 are executed. `Clog2 (Uns64 (-(-4)))` and `Clog2 (Uns64 (4))` should produce the same result: 2. Last, `2 + 1 = 3`, so `2^3=8`, which is less than the 9 values in the range.

I agree that there is a bug, but I think it's because the range is not `left-right`|`right-left`, but `left-right+1`|`right-left+1`. That might need to be fixed in L141, L147 and L149-L154.

eine, 2020-11-16:
A test script:

```py
from math import log, ceil

def clog2(x):
    return ceil(log(x, 2))

def Discrete_Range_Width(
    Left,
    Right,
    Dir
):
    Lo = Left if Dir == 'to' else Right
    Hi = Right if Dir == 'to' else Left
    if Lo > Hi:
        print(' Null range')
        return 0
    elif Lo >= 0:
        print(' Positive')
        return clog2(Hi+1)
    elif Hi < 0:
        print(' Negative only')
        return clog2(-Lo)+1
    else:
        Wl = clog2(-Lo)
        Wh = clog2(Hi)
        return max(Wl,Wh) + 1

def Test_Range(Left, Right, Dir):
    print('\n%d %s %d' % (Left, Dir, Right))

    r = Right - Left +1 if Dir == 'to' else Left - Right +1
    print(' - %d [%d]' % (r, clog2(r)))

    w = Discrete_Range_Width(Left, Right, Dir)
    wr = 2**w
    print(' - %d [%d] %s' % (wr, w, '!' if r > wr else ''))

for x in range(1, 6):
    Test_Range(0, x, 'to')
    Test_Range(x, 0, 'downto')

    # crashes because it tries to execute `Wh = clog2(Hi)` where `Hi = 0`
    #Test_Range(-x, 0, 'to')
    #Test_Range(0, -x, 'downto')

    Test_Range(-1, x, 'to')
    Test_Range(x, -1, 'downto')

    Test_Range(-2, x, 'to')
    Test_Range(x, -2, 'downto')

    if x > 2:
        Test_Range(-x, x, 'to')
        Test_Range(x, -x, 'downto')
```

To be investigated:

- `? to 0`
- `0 to ?`
- `-x to x`
- `x downto -x`
- `-1 to 5`
- `5 downto -1`
- `-2 to 5`
- `5 downto -2`

tgingold, 2020-11-16:
I agree with @eduardosm, the fix looks correct.

Thank you for the issue (and the suggestion).

commit e5b000fa (2020-11-16): synth-objtypes: fix width computation.  Fix #1509
closed 2020-11-16 by tgingold