ghdl/ghdl #1513: GHDL-Bug when trying to run a testbench

Sir-Photch, 2020-11-18:
**Description**
I am trying to do an exercise form my VHDL-lecture, with a testbench provided by our professor. I am a novice when it comes to vhdl, and not sure wether my code, the testbench or ghdl itself is the problem. Note that I am not here to get my homework done, but rather to know if ghdl is the problem.

**Expected behaviour**
I wanted to run a testbench on my code, an error occurs though.

**How to reproduce?**
I provide my code below.
Go to the directory and:
ghdl -c *.vhd
ghdl -m nbit_adder_tb
ghdl -r nbit_adder_tb

- OS: Debian Buster
- Origin:
  - [ ] Package manager: apt

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.36-dev (Debian 0.35+git20181129+dfsg-3) [Dunoon edition]
Compiled with GNAT Version: 8.2.0
Target: x86_64-linux-gnu
In directory: /home/christoph/ownCloud/Studium/Einführung in VHDL/02/
Command line:
/usr/bin/ghdl-mcode -r nbit_adder_tb
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap5.adb:806
Call stack traceback locations:
0x7f89f22524f1 0x55fd6484f531 0x55fd6484fad6 0x55fd6484882f 0x55fd648afcae 0x55fd648afe1d 0x55fd648b1d44 0x55fd6481d162 0x55fd6481a25b 0x55fd64827bc5 0x55fd648b2645 0x55fd647fc5a0 0x55fd6478464c 0x55fd648b53f8 0x55fd645ba45d 0x7f89f1e44099 0x55fd645b90f8 0xfffffffffffffffe
******************************************************************

**fulladder.vhd**
`entity fulladder is
  port (a,b,cin : in bit; s,c : out bit);
end entity fulladder;

architecture arch of fulladder is

  component halfadder
    port(a,b: in bit; s,c: out bit);
  end component;

  signal s0,c0,c1: bit;

begin

  HA0: halfadder
  port map(a,b,s0,c0);

  HA1: halfadder
  port map(cin,s0,s,c1);

  -- doenst work
  -- HA0: halfadder
  -- port map(a => a, b => b, c => c0, s => s0);
  -- FUNKTIONIERT NICHT???
  -- HA1: halfadder
  -- port map(a => cin, s => s0, s => s, c => c1);

  c <= c0 or c1;

end architecture;`

**halfadder.vhd**
`entity halfadder is
  generic(delay : natural);
  port (a,b : in bit; s,c : out bit);
end entity halfadder;

--- Process Signal Asignments
--architecture arch_proc of halfadder is

--begin

  --trigger : process(a,b)
  --begin

    --s <= transport a xor b after delay + 20 ns;
    --c <= transport a and b after delay + 15 ns;
  --  s <= a xor b;
  --  c <= a and b;

  --end process;

--end architecture arch_proc;

-- Concurrent Signal Asignments
architecture arch_concurrent of halfadder is

begin

  --s <= transport a xor b after delay + 20 ns;
  --c <= transport a and b after delay + 15 ns;
  s <= a xor b;
  c <= a and b;

end architecture arch_concurrent;`

**nbit_adder.vhd**
`entity nbit_adder is
  generic (n: positive);
  port (cin: in bit;
        a, b: in bit_vector(n-1 downto 0);
        cout: out bit;
        s: out bit_vector(n-1 downto 0));
end entity nbit_adder;

-- n-bit Ripple-Carry Adder
architecture rca_adder of nbit_adder is

  component fulladder
    port(a,b,cin: in bit; s,c: out bit);
  end component;
  signal carry: bit_vector(n-1 downto 0);

begin
  FA0: for i in n-1 downto 0 generate
    FA1: if (i = n-1) generate
      BITPAIR_HIGH: fulladder port map (a(n-1),b(n-1),carry(n-2),s(n-1),cout);
    end generate;
    FA2: if (i > 0) and (i < n-1) generate
      BITPAIR_MIDDLE: fulladder port map (a(i),b(i),carry(i-1),s(i),carry(i));
    end generate;
    FA3: if (i = 0) generate
      BITPAIR_LOW: fulladder port map (a(i),b(i),cin,s(i),carry(i));
    end generate;
  end generate;

end architecture rca_adder;`

**nbit_adder_tb.vhd**
`library ieee;
    use ieee.numeric_bit.to_unsigned;   -- Integer->Bit_Vector conversion functions

-- Entity: testbench has no ports
entity nbit_adder_tb is
end entity;

-- Test architecture:
--    1. create Unit Under Test (UUT)
--    2. while (not all input pattern tested)
--      2a. setup input pattern
--      2b. test output pattern
architecture structure of nbit_adder_tb is

  component nbit_adder is
    generic (n: positive);
    port    (cin:  in  bit; a, b: in  bit_vector(n-1 downto 0);
             cout: out bit; s:    out bit_vector(n-1 downto 0));
  end component;

  -- bitwidth for testing
  constant n: positive := 3;

  -- UUT connections
  signal a, b, s: bit_vector(n-1 downto 0);  
  signal cin, cout: bit;
  
  -- testpattern
  signal  expected : bit_vector(n downto 0);

begin

  -- Unit Under Test
  uut: nbit_adder
    generic map (n)
    port map(a=>a, b=>b, cin=>cin, s=>s, cout=>cout);

  -- setup all possible inputs and test the resulting output
  pTB: process is
    variable a_dec, b_dec, c_dec, s_dec : natural;
    begin
      report "Starting test..." severity note;            
        
      for a_dec in 0 to 2**n-1 loop
        a <= bit_vector(to_unsigned(a_dec, n));
        
        for b_dec in 0 to 2**n-1 loop
          b <= bit_vector(to_unsigned(b_dec, n));
          
          for c_dec in 0 to 1 loop                
            if (c_dec = 0) then cin <= '0'; else cin <= '1'; end if;
            
            -- expected output
            expected <= bit_vector(to_unsigned(a_dec + b_dec + c_dec, n+1));           
          
            -- wait for 1 ns;
            wait for (n-2)*15 ns + 40 ns; -- (n-2)*cdelay + 2*max(cdelay,sdelay)
 
            -- check results
            assert s    = expected(n-1 downto 0) report "bad sum value"   severity error;
            assert cout = expected(n)            report "bad carry value" severity error;
          
          end loop;
        end loop;
      end loop;
      
      report "done." severity note;
      wait;
  end process;          
     
end architecture;`

eine, 2020-11-18:
@Sir-Photch, that is a legit bug. Even if your code was wrong, GHDL should never crash. However, you are using a rather old version of GHDL (from 2 years ago!). Chances are that the bug was fixed since then. In fact, I just tried your sources with `GHDL 1.0-dev (v0.37.0-1076-g9db35195-dirty)` and this is the output I get:

```
# ./run.sh
nbit_adder_tb.vhd:44:11:warning: declaration of "a_dec" hides variable "a_dec" [-Whide]
nbit_adder_tb.vhd:47:13:warning: declaration of "b_dec" hides variable "b_dec" [-Whide]
nbit_adder_tb.vhd:50:15:warning: declaration of "c_dec" hides variable "c_dec" [-Whide]
fulladder.vhd:15:3: no actual for generic "delay"
fulladder.vhd:18:3: no actual for generic "delay"
fulladder.vhd:15:3: no actual for generic "delay"
fulladder.vhd:18:3: no actual for generic "delay"
C:\msys64\mingw64\bin\ghdl.exe: compilation error
C:\msys64\mingw64\bin\ghdl.exe: file 'nbit_adder_tb.exe' does not exist
C:\msys64\mingw64\bin\ghdl.exe: Please elaborate your design.
```

I used GHDL with LLVM backend, so this is the set of commands:

```
ghdl -c *.vhd
ghdl -m nbit_adder_tb
ghdl -e nbit_adder_tb
ghdl -r nbit_adder_tb
```

Therefore, the problem in your design is that generic `delay` of entity `halfadder` has no default value, and you are not providing any value when you instantiate it. I won't tell you how to fix it for now, because that's part of your learning process, and because multiple solutions might be valid. Nonetheless, this is the output I get after fixing it:

```
# ./run.sh
nbit_adder_tb.vhd:44:11:warning: declaration of "a_dec" hides variable "a_dec" [-Whide]
nbit_adder_tb.vhd:47:13:warning: declaration of "b_dec" hides variable "b_dec" [-Whide]
nbit_adder_tb.vhd:50:15:warning: declaration of "c_dec" hides variable "c_dec" [-Whide]
analyze nbit_adder_tb.vhd
nbit_adder_tb.vhd:44:11:warning: declaration of "a_dec" hides variable "a_dec" [-Whide]
nbit_adder_tb.vhd:47:13:warning: declaration of "b_dec" hides variable "b_dec" [-Whide]
nbit_adder_tb.vhd:50:15:warning: declaration of "c_dec" hides variable "c_dec" [-Whide]
analyze nbit_adder.vhd
analyze fulladder.vhd
analyze halfadder.vhd
elaborate nbit_adder_tb
nbit_adder_tb.vhd:42:7:@0ms:(report note): Starting test...
nbit_adder_tb.vhd:67:7:@7040ns:(report note): done.
```

Therefore, let's focus on helping you get an up to date GHDL. How did you install the one you are using?

Sir-Photch, 2020-11-19:
Hey @eine, thanks for your quick reply!
Huh! Yeah thats a pretty old version indeed. 
I just used `apt install ghdl`
Maybe it's because the packagelists for debian are pretty slowly updated (not like arch), in order to provide sufficient stability.

eine, 2020-11-19:
There are two different issues. On the one hand, the outdated GHDL version might have multiple bugs which are fixed in master. On the other hand, your code has/had a bug which prevented the test from building. That is, the issue with `delay` is probably unrelated to the bug that crashes the older GHDL.

My suggestion is to build latest GHDL from sources. The dependencies are `gnat gcc make zlib1g-dev`. Uninstall the one you installed through apt, and then:

```sh
git clone https://github.com/ghdl/ghdl/
cd ghdl
./configure --prefix=/usr/local
make
make install
```

Sir-Photch, 2020-11-19:
I see. Now I know what you mean with "default value".

It works now! Thank you.

closed 2020-11-19 by Sir-Photch

eine, 2020-11-19:
Nice! Have fun with your VHDL lectures and see your around 😉