ghdl/ghdl #1522: synth: Support for arrays inside records.

bl0x, 2020-11-25:
**Description**
Indexing an array inside a record with several array members via to_integer() conversion causes an error in synthesis.

**Expected behaviour**
I expect a more descriptive error message, in case this is really a user error.
The example below fails, but it works when:

  a) the record is reduced to just the 'a' member.
  b) the 'to_integer(c)' is replaced by a valid numeric constant (0 or 1).

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity ent is
	port (int_out : out integer);
end;

architecture RTL of ent is
	type int2_array is array (0 to 1) of integer;

	type my_record is record
		a : int2_array;
		b : int2_array;
	end record my_record;

	signal c : unsigned(1 downto 0) := (others => '0');

	constant sym : my_record := (a => (0=>1, 1=>2), b => (0=>2, 1=>2));
begin
	int_out <= sym.a(to_integer(c));
	c <= (others => '0');
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
ghdl --synth ent.vhd -e ent
```

**Context**
```
ghdl_bug.vhd:21:24:note: found ROM "n7", width: 32 bits, depth: 4

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1074-g31ca0d8a) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/home/bloeher/git/rataser/tmp/
Command line:
ghdl --synth ghdl_bug.vhd -e ent
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : netlists-memories.adb:336
Call stack traceback locations:
0x498be7 0x49a62e 0x56a4a4 0x565837 0x5946c2 0x4f9b77 0x4076a9 0x7f0934022150 0x40771c 0xfffffffffffffffe
******************************************************************
```

dalex78, 2026-08-20:
# Issue #1522 — Crash: array inside a record indexed via `to_integer()`

Analysis by an AI agent.

## High-level summary

Indexing an array member of a record (with more than one array member) via `to_integer()` (`sym.a(to_integer(c))`, `sym` a constant record with `a`/`b` array fields) crashed `ghdl --synth` with an internal `TYPES.INTERNAL_ERROR` at `netlists-memories.adb:336`. The reporter found it worked with a single-member record or a static index.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth ent.vhd -e ent` now produces a correct netlist, inferring a 2-entry ROM for `sym.a` and indexing it dynamically via `c`, instead of crashing. This must have been fixed as part of general improvements to record-with-multiple-array-members handling in `src/synth` since this was reported (originally against a 2021 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1522/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this ticket be closed ? (PR merged)

closed 2026-08-24 by tgingold