ghdl/ghdl #1527: Generic Types Not Being Interpreted Correctly

patelvrajn, 2020-11-29:
../Source_Code/Binary_Multiplier.vhd:97:29:error: can't associate 'operand_default_value' with generic "default_value"
../Source_Code/Binary_Multiplier.vhd:97:29:error: (type of 'operand_default_value' is a subtype of std_ulogic_vector)
../Source_Code/Flip_Flop.vhd:29:7:error: (type of generic "default_value" is template_type)
../Source_Code/Binary_Multiplier.vhd:102:29:error: can't associate 'multiplicandin' with port "datain"
../Source_Code/Binary_Multiplier.vhd:102:29:error: (type of 'multiplicandin' is a subtype of std_ulogic_vector)
../Source_Code/Flip_Flop.vhd:38:7:error: (type of port "datain" is template_type)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/patelvrajn/Development_Workspace/VHDL/My_VHDL_Designs/Scripts/
Command line:
/usr/bin/ghdl-mcode -m --std=08 --workdir=../Source_Code --work=Source_Code_Library -P=../Packages -P=../Testbenches Binary_Multiplier
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-sem_assocs.adb:1649
Call stack traceback locations:
0x7f174f6d0d12 0x55b42ccda798 0x55b42ccdc91c 0x55b42ccdd735 0x55b42ccde8aa 0x55b42ccde9a3 0x55b42ccdf03d 0x55b42cca4e83 0x55b42cca602e 0x55b42cca6272 0x55b42cca63c9 0x55b42ccde4c5 0x55b42cce5299 0x55b42cc9cb32 0x55b42cc9d9d6 0x55b42ccee971 0x55b42ccfbce0 0x55b42cd02263 0x55b42cc94557 0x55b42cdbaf31 0x55b42caa15ff 0x7f174f2800b1 0x55b42caa012c 0xfffffffffffffffe
******************************************************************
make: *** [GHDL_Makefile:18: compile] Error 2

See the following compressed folder for the makefile as well as relevant source code:
[bug.zip](https://github.com/ghdl/ghdl/files/5612593/bug.zip)

eduardosm, 2020-12-15:
Here is a minimal example that causes the "can't associate" errors, although it does not produce a "GHDL Bug occurred":
```vhd
entity ent_g is
    generic (
        type T
    );
    port (
        i: in T;
        o: out T
    );
end entity;

architecture arch of ent_g is
begin
    o <= i;
end architecture;

library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        i: in std_ulogic;
        o: out std_ulogic
    );
end entity;

architecture arch of ent is
begin
    inst: entity work.ent_g
    generic map (
        T => std_ulogic
    )
    port map (
        i => i,
        o => o
    );
end architecture;
```

```
ghdl analyze --std=08 ent.vhd
```

Xiretza, 2021-07-07:
Analyzing only the first part of the above reproducer (`ent_g` with or without architecture) now results in this crash:

```
translate_subtype_definition: cannot handle IIR_KIND_INTERFACE_TYPE_DEFINITION (ent.vhd:3:14)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r495.gcfa7ec61) [Dunoon edition]
Compiled with GNAT Version: 11.1.0
Target: x86_64-pc-linux-gnu
/home/xiretza/unison/code/spitfire/
Command line:
/usr/lib/gcc/x86_64-pc-linux-gnu/10.3.0/ghdl1 --ghdl--std=08 -P/usr/lib/ghdl/ieee/v08/ -P/usr/lib/ghdl/ -quiet -o ent.s ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x70c8fa 0x7c704c 0x7c7e94 0x7fe6f6 0x805efd 0x7f5c17 0x7f1fe0 0x828c2c 0x7a936b 0x6a2343 0xc898f2 0x69788d 0x6a7730 0x82905f 0x69546d 0x7ffbd60a4b23 0x69d19c 0xfffffffffffffffe
******************************************************************

Execution of /usr/lib/gcc/x86_64-pc-linux-gnu/10.3.0/ghdl1 terminated by unhandled exception
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Call stack traceback locations:
0x70c8fa 0x7c704c 0x7c7e94 0x7fe6f6 0x805efd 0x7f5c17 0x7f1fe0 0x828c2c 0x7a936b 0x6a2343 0xc898f2 0x69788d 0x6a7730 0x82905f 0x69546d 0x7ffbd60a4b23 0x69d19c 0xfffffffffffffffe
ghdl:error: compilation error
```

Analyzing the entire reproducer doesn't crash, but gives this error:

```
ent.vhd:33:14:error: expression associated before its type
        i => i,
             ^
ghdl:error: compilation error
```

Xiretza, 2021-07-07:
Actually, this is probably a dupe of #726.

commit ef2c99be (2023-12-17): testsuite/gna: add a test for #1527

Close #1527
closed 2023-12-17 by tgingold