ghdl/ghdl #1528: User attributes  Crash

sckoarn, 2020-12-01:
**Description**
Testing some user attributes GHDL crashes.

**Expected behaviour**
There should be some messages generated by the report statements.

**How to reproduce?**
https://www.edaplayground.com/x/76kp

**Context**
Please, provide the following information:

- OS:  Unix local,   EDA Playground ?
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2020-12-01:
Please, copy the code in the issue instead of using a link to edaplayground.com
It doesn't work immediately (a login is required) and we don't know how long it will stay.

Thanks!

sckoarn, 2020-12-01:
The projects in EDA PG  stay for ever ...   The login is minimal.   The code is not small.
But I will paste it in here, just a moment.

tgingold, 2020-12-01:
> The projects in EDA PG stay for ever ...

I am 100% sure that this is not true.  Nothing stays forever!

> But I will paste it in here, just a moment.

Thanks!

sckoarn, 2020-12-01:
I had made a mistake,  it was a privet project,  now is public.   But here is the code anyway

package attrs_pkg is

  attribute ent_name: string;
  attribute ent_type: string;
  attribute ent_stat: integer;

  attribute arc_name: string;
  attribute arc_type: string;
  attribute arc_stat: integer;

end package;

library ieee;
use ieee.std_logic_1164.all;

use work.attrs_pkg.all;

entity ent1 is
  attribute ent_name of ent1: entity is "entity ent1";
  attribute ent_type of ent1: entity is "LogicI";
  attribute ent_stat of ent1: entity is 95;
end entity ent1;

use work.attrs_pkg.all;

entity ent2 is
  attribute ent_name of ent2: entity is "entity ent2";
  attribute ent_type of ent2: entity is "LogicII";
  attribute ent_stat of ent2: entity is 75;
end entity ent2;

use work.attrs_pkg.all;

entity ent3 is
  attribute ent_name of ent3: entity is "entity ent3";
  attribute ent_type of ent3: entity is "LogicIII";
  attribute ent_stat of ent3: entity is 45;
end entity ent3;

architecture rtl of ent1 is
  attribute arc_name of rtl: architecture is "rtl of ent1";
  attribute arc_type of rtl: architecture is "RTL";
  attribute arc_stat of rtl: architecture is 100;
begin
  process
  begin
    report "I am: " & rtl'arc_name;
    report "Type: " & rtl'arc_type;
    report "Status: " & integer'image(rtl'arc_stat);
    wait;
  end process;
end rtl;

architecture bhv of ent2 is
  attribute arc_name of bhv: architecture is "bhv of ent2";
  attribute arc_type of bhv: architecture is "BHV";
  attribute arc_stat of bhv: architecture is 100;
begin
  process
  begin
    report "I am: " & bhv'arc_name;
    report "Type: " & bhv'arc_type;
    report "Status: " & integer'image(bhv'arc_stat);
    wait;
  end process;
end bhv;

architecture rtl of ent3 is
  attribute arc_name of rtl: architecture is "rtl of ent3";
  attribute arc_type of rtl: architecture is "RTL";
  attribute arc_stat of rtl: architecture is 35;
begin
  process
  begin
    report "I am: " & rtl'arc_name;
    report "Type: " & rtl'arc_type;
    report "Status: " & integer'image(rtl'arc_stat);
    wait;
  end process;
end rtl;

use work.all;
--use work.attrs_pkg.all;

entity uattr3 is
  attribute ename: string;
  attribute ename of uattr3: entity is "user attributes 3";
  attribute special: boolean;
  attribute is_clock: boolean;
  attribute sign_bit: integer;
begin
end entity uattr3;

architecture tb of uattr3 is

  attribute ename of tb: architecture is uattr3'ename & " Demonstration";
  attribute stype: integer;
  
  signal bv : bit_vector(7 downto 0) := "11000100";
  attribute stype of bv: signal is 2;
  attribute sign_bit of bv: signal is bv'left;
  signal b : bit;
  attribute stype of b: signal is 1;
  attribute special of b: signal is true;
  signal clk : bit;
  attribute is_clock of clk: signal is true;
  signal clk1 : bit;
  signal clk2 : bit;
  
  signal cnt : integer := 0;

begin
  
  start_up: process
  begin
    --report "Object: " & tb'ename & "  ... starting up ...";
    wait;
  end process start_up;

  bv <= "00001111" when cnt = 0 else
        "00011110" when cnt = 1 else
        "00111100" when cnt = 2 else
        "01111000" when cnt = 3 else
        "11110000" when cnt = 4;
        
  b <= '1' when bv = "00111100" else '0';
  
  clock: process
  begin
    while cnt < 6 loop
      wait for 1 ns;
      clk <= not clk;
    end loop;
    wait;
  end process clock;

  proc1: process
  begin
    if(clk'event and clk = '1' and clk'is_clock) then
      cnt <= cnt + 1;
      clk1 <= not clk1;
      report "Tick ... " & integer'image(cnt);
    end if;
    
    wait on clk;
  end process;
  
  bmon: process(b)
  begin
    --report "stype of b is: " & integer'image(b'stype);
    if b = '1' and b'special then
      --report "stype of bv is: " & integer'image(bv'stype);
      assert bv = "00111100" report "Error: got unexpected bv value" severity failure;
    end if;
  end process;
  
  bvmon: process(bv)
    variable i_v : integer;
    attribute stype of i_v: variable is 0;
  begin
    i_v := cnt;
    case bv is
      when "00001111" =>
        report ent1'ent_name;
        report ent1'ent_type;
        report "Completeness: " & integer'image(ent1'ent_stat);
        assert cnt = 0 report "Error0: got unexpected cnt value" severity failure;
      when "00011110" =>
        assert cnt = 1 report "Error1: got unexpected cnt value" severity failure;
      when "00111100" =>
        report ent2'ent_name;
        report ent2'ent_type;
        report  "Completeness: " & integer'image(ent2'ent_stat);
        assert cnt = 2 report "Error2: got unexpected cnt value" severity failure;
      when "01111000" =>
        assert cnt = 3 report "Error3: got unexpected cnt value" severity failure;
      when "11110000" =>
        report ent3'ent_name;
        report ent3'ent_type;
        report  "Completeness: " & integer'image(ent3'ent_stat);
        assert cnt = 4 report "Error4: got unexpected cnt value" severity failure;
      when others =>
        --report "stype of i_v is: " & integer'image(i_v'stype);
        null;
    end case;
    
    
    
  end process;
  
  d1: entity work.ent1(rtl);
  d2: entity work.ent2(bhv);
  d3: entity work.ent3(rtl);

end tb;

sckoarn, 2020-12-01:
Sorry I do not have time to boil this down to something smaller.
This is why I like to EDA PG,  to confirm any errors I find.

sckoarn, 2020-12-01:
And this is my  error output message:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (v0.37) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/project/dsim/kcampbell/superp/test/dsim/vh_tests/decl_4_93/uattr3/
Command line:
ghdl -r --std=93 uattr3
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:555
Call stack traceback locations:
0x560b7801d4d2 0x560b7801d8b1 0x560b7801db2a 0x560b78022806 0x560b781a944a 0x560b781a9360 0x560b781889eb 0x560b7814678f 0x560b781469d9 0x560b78156198 0x560b78156315 0x560b781492ae 0x560b781484d0 0x560b78148cb9 0x560b781492fc 0x560b781561a7 0x560b78156315 0x560b78159769 0x560b7815e090 0x560b781c9b7e 0x560b781353bf 0x560b781324d1 0x560b7814026e 0x560b781ce2cb 0x560b78114ed8 0x560b780aacc8 0x560b781d1a76 0x560b77ebb5c7 0x7f6473be0099 0x560b77eba108 0xfffffffffffffffe
*

sckoarn, 2020-12-01:
My compile and run script.

ghdl -a --std=93 attrs_pkg.vhdl
ghdl -a --std=93 ents.vhdl
ghdl -a --std=93 test.vhdl
ghdl -r --std=93 uattr3

sckoarn, 2020-12-02:
Also, just found out from running on another reference that
attribute ename of tb: architecture is uattr3'ename & " Demonstration";

uattr3'ename  may not be considered locally static?

sckoarn, 2020-12-02:
section  5.1   line 55  LRM 2000
If the entity name list denotes an entity interface, architecture body, or configuration declaration, then the expression is required to be locally static (see 7.4).

tgingold, 2020-12-06:
I will see what I can do, but the real problem is that the attribute is often not static in user designs.

commit 07a7f419 (2020-12-08): vhdl: handle locally static attributes on entity/architecture/configurations

Attributes of entity/architecture/configurations are expected to be locally
static so that they can be referenced from outside (so on the non-instantiated
entity).

But many designs break this assumption.  In relaxed mode, non-locally static
attributes are allowed but now cannot be referenced outside the entity.

Locally static attributes can be referenced from outside.

Fix #1528
closed 2020-12-08 by tgingold