ghdl/ghdl #1530: Crash when using PSL stable in simulation

bkzshabbaz, 2020-12-04:
**Description**
Trying to simulate code that utilizes PSL stable, but GHDL crashes when it runs the testbench.

**Expected behaviour**
Not to crash.

**How to reproduce?**

```
vhd :file: vai_reg_tb.vhd
library ieee;
use     ieee.std_logic_1164.all;
use     ieee.numeric_std.all;

entity vai_reg_tb is
-- Testbench has no ports
end entity vai_reg_tb ;

architecture tb of vai_reg_tb is
constant period         : time := 10 ns;

signal clk, reset       : std_logic := '0';
signal tb_complete      : std_logic := '0';

signal     Din_i        : std_logic_vector(7 downto 0) := (others => '0');
signal     DinValid_i   : std_logic := '0';
signal     DinStart_i   : std_logic := '0';
signal     DinStop_i    : std_logic := '0';
signal     DoutAccept_i : std_logic := '0';
signal     Dout_o       : std_logic_vector(7 downto 0);
signal     DoutValid_o  : std_logic;
signal     DoutStart_o  : std_logic;
signal     DoutStop_o   : std_logic;
signal     DinAccept_o  : std_logic;
begin

clk <= not clk after period/2 when tb_complete /= '1' else '0';

uut : entity work.vai_reg
    port map (
    reset_n_i => reset,
    clk_i       => clk,
    din_i       => din_i,
    dinvalid_i  => dinvalid_i,
    dinstart_i  => dinstart_i,
    dinstop_i   => dinstop_i,
    dinaccept_o => dinaccept_o,

    dout_o      => dout_o,
    doutvalid_o => doutvalid_o,
    doutstart_o => doutstart_o,
    doutstop_o  => doutstop_o,
    doutaccept_i=> doutaccept_i
    );
sim : process
begin
    wait for 10 us;
end process sim;
end architecture tb;
```
```
vhd :file: vai_reg.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
entity vai_reg is
  generic (
    Formal : boolean := false
  );
  port (
[... 344 lines cut ...]
  - [X] Built from sources: `3bf136140b218a3f9191ba6cef319bede7bf6425`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
translate_expression: cannot handle IIR_KIND_PSL_STABLE (vai_reg.vhd:205:47)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1099-g3bf13614) [Dunoon edition]
Compiled with GNAT Version: 8.4.0
Target: x86_64-linux-gnu
/home/dev/projects/test_psl/
Command line:
ghdl -r --std=08 -fpsl vai_reg_tb
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x5562666f7036 0x5562668d3799 0x5562668a4b34 0x5562668a4c7b 0x5562668a59ad 0x5562668a7e77 0x5562668a7ca7 0x5562668a7f85 0x556266915f26 0x55626687d709 0x55626687a81b 0x556266888d86 0x55626691a89d 0x556266859e3a 0x556266793177 0x55626691e211 0x55626654ca85 0x7f44f5001bf5 0x55626654b218 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Xiretza, 2020-12-04:
=> #1498

tmeissner, 2020-12-04:
It seems to be my vai_reg design :smile:

I always help me by using some VHDL helper logic:

```vhdl
-- VHDL helper logic
process is
begin
  wait until rising_edge(Clk_i);
    s_data_old <= s_data;
  end if;
end process;

-- Check for stable data
assert always valid and not ack -> next s_data = s_data_old;

dalex78, 2026-08-25:
# Issue #1530 — PSL builtin function in a simulated design

Analysis by an AI agent.

## High-level summary

Using a PSL builtin function in a design that is simulated — `assert always stable(x)` — ended in an internal error:

```
translate_expression: cannot handle IIR_KIND_PSL_STABLE
```

`stable()` and the rest of the family (`prev`, `rose`, `fell`, `onehot`, `onehot0`) are implemented for `--synth`, which has its own translator, but were never handled in `src/vhdl/translate`, the one used for ordinary simulation. Reaching an unimplemented case there falls through to `Error_Kind`, which is the internal-error box.

## What the fix does

Handles `Iir_Kinds_Psl_Builtin` in `Trans.Chap7.Translate_Expression`: it reports

```
PSL builtin function not supported in simulation
```

and emits a run-time program error (`Chap6.Gen_Program_Error` with the new `Prg_Err_Unsupported` code, reported by grt as `unsupported construct`) before yielding `false` for the expression so translation can continue.

The run-time trap is not belt and braces. With the gcc and llvm backends the message comes out of `ghdl -a` and compilation stops there, so nothing is ever run. With mcode the translation happens while the simulation is already being elaborated, and the error alone does not stop it: without the trap the assertion would go on being evaluated on the made-up `false` value, reporting a PSL failure at every clock edge and exiting 0. Reporting an error and then producing confidently wrong output is worse than crashing, so the generated code stops instead.

This does not implement the builtins — it turns "GHDL is broken" into "GHDL does not support this here", which is what the user needs to know.

## Where the diagnostic appears

Not at the same point on every backend, which matters for the test:

| backend | when it is reported | what then happens |
|---|---|---|
| gcc, llvm | during `ghdl -a`: these translate while analyzing | compilation stops, `ghdl -a` fails |
| mcode | during elaboration of the simulation | the trap fires, `simulation failed`, non-zero status |

That is a consequence of where translation happens, not of the fix.

## Note on the issue thread

The issue opens with a comment pointing at #1498. If these turn out to be the same underlying report, this fix and its test still stand on their own — the crash they remove is real either way — but the two issues may want linking or closing together.

## Testing

`testsuite/gna/issue1530/` analyzes and simulates a reduced equivalent of the report, asserting the clean diagnostic, the run-time trap where a simulation is reached, and the absence of the internal-error box on all three backends.

dalex78, 2026-08-27:
@tgingold the PR associated to that issue reports PSL builtins as unsupported in simulation. Does that enable to close this issue ? Or do you wish to keep it opened until PSL features are added in ?

tgingold, 2026-08-27:
Let's keep this issue open as the feature is not implemented.