ghdl/ghdl #1531: Resizing unsigned's: unhandled (static) function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_UNS_NAT

JulianKemmerer, 2020-12-06:
**Description**

It looks like GHDL might not like something like:
```
<an unsigned(31 downto 0) var> := resize(to_unsigned(0, 1), 32);
```
```
unhandled (static) function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_UNS_NAT
```

**Expected behaviour**
Synthesizes fine with other tools.

**How to reproduce?**

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process is
  variable thing : unsigned(31 downto 0);
begin
    thing := resize(to_unsigned(0, 1), 32);   -- <<<<<<< HERE
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

**Context**

- OS: Linux
- Origin: Night fpga tool forge days ago

tgingold, 2020-12-06:
Is it a synthesis issue or a simulation issue ?  Your reproducer uses `to_unsigned` but doesn't use `numeric_std` so it fails to analyse.

JulianKemmerer, 2020-12-06:
I am using the ghdl plugin for yosys so it's a synthesis issue.

My apologies, the libs I am using are 

Hmmm. for reference I am only using 
```
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;
```

I will try to update the code

tgingold, 2020-12-06:
Ah ok.  Thank you for the clarification.

eine, 2020-12-06:
I cannot reproduce with `GHDL 1.0-dev (v0.37.0-1063-gc5b094bb)`.

```sh
ghdl -s ent.vhd

ghdl -a ent.vhd
ghdl --elab-run ent
```

JulianKemmerer, 2020-12-06:
I think I need to change the example to output the value?

Had someone else reproduce it yesterday, can show their exact code too.. fixing soon...

Again sorry for the mix-up, time waste

JulianKemmerer, 2020-12-06:
OK updating the post with this from what I another person reported

```vhdl
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

entity test is
    Port(
        --a_val: in INTEGER;
        b_val: out UNSIGNED(31 downto 0)
);
end entity;

architecture Behavioral of test is
begin
    --b_val <= resize(to_unsigned(a_val, 2),32);  -- << no error w/ a_val input signal
    b_val <= resize(to_unsigned(1, 2),32);  -- Yes same error when a_val replaced with constant

end architecture;

```

They did say it was an old version of GHDL they had running

JulianKemmerer, 2020-12-06:
Added ghdl version hash
GHDL 1.0-dev (v0.37.0-1098-g92b4b6ec) [Dunoon edition] open-tool-forge.nightly-20201204

JulianKemmerer, 2020-12-06:
Confirmed my updated example with that exact version of ghdl specified produces the behavior

```
error: unhandled (static) function: IIR_PREDEFINED_IEEE_NUMERIC_STD_RESIZE_UNS_NAT
```

closed 2020-12-06 by tgingold

JulianKemmerer, 2020-12-06:
Thanks for the quick fix - on a weekend even! :) 

New to the GHDL->Yosys->NextPNR flow and loving it - thanks for great work of all yall folks

commit ee9b40b2 (2020-12-06): synth: handle static resize from std_numeric.  Fix #1531