ghdl/ghdl #1533: build failure from GCC/GNAT 11

sharkcz, 2020-12-10:
Build fails when the upcoming GCC/GNAT 11 is used due an Ada syntax error.
````
gcc -c -I./ -I./src -I./src/vhdl -I./src/synth -I./src/grt -I./src/psl -I./src/vhdl/translate -I./src/ghdldrv -I./src/ortho -I./src/ortho/llvm6 -I./src/synth -I./src/ghdldrv -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -I- /builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/src/synth/synth-insts.adb
synth-insts.adb:251:44: operator for type "Ada.Streams.Stream_Element_Offset" is not directly visible
synth-insts.adb:251:44: add with_clause and use_clause for "Streams"
gnatmake: "/builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/src/synth/synth-insts.adb" compilation error
make: *** [Makefile:344: ghdl_llvm] Error 4
````
The ada compiler is a recent snapshot in Fedora Rawhide (gcc version 11.0.0 20201204)

tgingold, 2020-12-10:
Just comment out the line 251.

sharkcz, 2020-12-10:
Thanks, it's the fix :-)

commit 42ad61bf (2020-12-12): synth-insts: comment assert for gcc/gnat-11.  Fix #1533
closed 2020-12-12 by tgingold