ghdl/ghdl #1534: unhandled (static) function: IIR_PREDEFINED_IEEE_1164_TO_STDULOGIC

eduardosm, 2020-12-12:
**Description**

Using `to_stdulogic` with synth produces the following error:
```
unhandled (static) function: IIR_PREDEFINED_IEEE_1164_TO_STDULOGIC
```

**How to reproduce?**

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        o: out std_ulogic
    );
end entity;

architecture arch of ent is
begin
    o <= to_stdulogic('0');
end architecture;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl synth ent.vhd -e ent
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `g9db35195`

eduardosm, 2020-12-12:
Something similar happens with `to_stdulogicvector`.

tgingold, 2020-12-13:
Please, provide testcases for the last two functions.

tgingold, 2020-12-13:
It looks like you are the first one to do synthesis with the bit type.

eduardosm, 2020-12-13:
For `to_stdulogicvector`:
```vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        o: out std_ulogic_vector(3 downto 0)
    );
end entity;

architecture arch of ent is
begin
    o <= to_stdulogicvector(bit_vector'("0101"));
end architecture;
```

For `"xor"(x: bit_vector, y: bit_vector) return bit_vector`:
```vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        o: out bit_vector(3 downto 0)
    );
end entity;

architecture arch of ent is
begin
    o <= bit_vector'("0000") xor bit_vector'("1111");
end architecture;
```

For `"xor"(x: bit, y: bit) return bit`:
```vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
    port (
        o: out bit
    );
end entity;

architecture arch of ent is
    constant x: bit_vector(0 to 0) := "1";
    constant y: bit_vector(0 to 0) := "0";
begin
    o <= x(0) xor y(0);
end architecture;
```
To reproduce this one, xor arguments must be indexed from vector.

commit 500ead97 (2020-12-13): synth-static_oper: handle bit xor (scalar and array).  Fix #1534
closed 2020-12-13 by tgingold