ghdl/ghdl #1535: Error building GHDL from source on Raspberry Pi

rdb9879, 2020-12-13:
**Description**
I'm trying to build GHDL from source on the Raspberry Pi 4 (Linux raspberrypi 5.4.51-v7l+ #1333 SMP Mon Aug 10 16:51:40 BST 2020 armv7l GNU/Linux), however I'm getting errors during the "make" command.

**Expected behaviour**
This is the reported errors:

```
gcc -c -g -DWITH_GNAT_RUN_TIME -o fastlz.o /home/pi/ramtmp/ghdl-master/./src/grt/fst/fastlz.c
gcc -c -g -o memsegs_c.o src/ortho/mcode/memsegs_c.c
gcc -c -g -DWITH_GNAT_RUN_TIME -o chkstk.o /home/pi/ramtmp/ghdl-master/./src/grt/config/chkstk.S
chkstk.S: Assembler messages:
chkstk.S:37: Error: unrecognized symbol type ""
chkstk.S:41: Error: bad instruction `testl %eax,%eax'
chkstk.S:42: Error: bad instruction `je 0f'
chkstk.S:43: Error: bad instruction `subl $4,%eax'
chkstk.S:44: Error: bad instruction `subl %eax,%esp'
chkstk.S:45: Error: bad instruction `jmp *(%esp,%eax)'
chkstk.S:47: Error: bad instruction `ret'
make: *** [src/grt/Makefile.inc:228: chkstk.o] Error 1

```

**How to reproduce?**
```
./configure --prefix=/usr/local
make
```

Xiretza, 2020-12-13:
The mcode backend is x86_64 only, you will have to build either the GCC or the LLVM backend. See [the docs](https://ghdl.github.io/ghdl/getting/) for more info.

eine, 2020-12-14:
This procedure shows how to build GHDL on Zynq: https://github.com/dbhi/dbhi/blob/main/scripts/setup_pynq.sh#L24-L37. You might need to tweak the version of clang/llvm. Other than that, you can `make install`, instead of installing it to a custom location.

closed 2020-12-17 by tgingold