ghdl/ghdl #1541: Add additional option in simulation for defining switching delays for basic logic operations

krjdev, 2020-12-20:
**Is your feature request related to a problem? Please describe.**
I would be nice, if there exists an additional option in GHDL to define the switching delays from "Low" to "High" and from  "High" to "Low" for basic logic operations. Don't know if this option exists in other VHDL simulators.

The following (maybe stupid) VHDL code might syntheses, but cannot be simulated in GHDL without the "after TIME" keyword. On the other side, with the "after TIME" keyword the syntheses doesn't work:

```
library IEEE;
use IEEE.STD_LOGIC_1164.all;

entity hsosc is
port (
    EN:     in std_logic;
    CLK:    out std_logic
);
end hsosc;

architecture behav of hsosc is
    signal pre_clk: std_logic := '0';

    begin
        process(EN, pre_clk)
        begin
            if (EN = '1') then
                pre_clk <= not pre_clk after 1ns;
            else
                pre_clk <= '0';
            end if;
        end process;

        CLK <= pre_clk;
end behav;

```
In this example, with the activated requested option, would be great if GHDL adds a delay on the operator `not`, before `pre_clk` change the state. 

Without this requested  feature you need two different VHDL source files. With this feature, only one VHDL source is needed except the Testbench. But it might breaks the simulation of other simulators which doesn't support this feature. Also don't know if this requested feature is compatible with the IEEE standard.

But I think this feature would be great for a nearly "real" simulation of a design in a FPGA. (Okay, there exists also other delays in the LUT's of a FPGA.)

**Describe the solution you'd like**
Please add an option in the command line to define the basic (according to the device data sheet, if this information exists) switching delays for the simulation.

**Describe alternatives you've considered**
Add generally a default switching delay (1fs as example) for basic logic operations.

tgingold, 2020-12-21:
I am surprised the code doesn't synthesize when you have `after TIME`.  Synthesizers simply ignore that.

Automatically adding after in simulation is not that obvious.  The easiest possibility is to regenerate your source without them.  That's not difficult to do.

eine, 2021-01-11:
IMHO, adding a default delay for boolean operators is kind of reinventing the language. Right after doing so, some users might want to set different delays in different places. At that point, the feature would be as complex as using `after TIME`.

closed 2021-09-14 by tgingold