ghdl/ghdl #1543: Compile failed in ARM because src/grt/config/chkstk.S

zipotron, 2020-12-21:
**Description**
A clear and concise description of what the issue is about.

**Expected behaviour**
What you expected to happen, and what is happening instead.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ent.vhd
entity ent is
end entity;

architecture a of ent is
begin
  process begin
    report "Hello world" severity note;
    wait;
  end process;
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

zipotron, 2020-12-21:
I am trying to compile in ARM, src/grt/config/chkstk.S have a X86 assembly code, I would like to make a "ifdef" for put the ARM equivalence, but I have no idea what is this code propose... Please, any one can review or help me for fix by myself? Thanks.

eine, 2020-12-21:
It is difficult to know, since you didn't provide info about the procedure you are using for building GHDL. However, my guess is that you are trying to do `./configure && make`. That won't work because it'll use the default backend (`mcode`), which is for x86 only. Using LLVM or GCC should work. See #1535.

Adding ifdefs for the mcode backend is a non-trivial task.

zipotron, 2020-12-21:
Thanks a lot Eine! I went trow #1535 but got this error:
gcc: fatal error: cannot execute ‘gnat1’: execvp: No such file or directory
compilation terminated.
gnatmake: "/home/zipotron/ghdl/src/ortho/llvm6/ortho_code_main.adb" compilation error
make[1]: *** [../src/ortho/llvm6/Makefile:14: ghdl1-llvm] Error 4

That is the same I get if I use the standard compilation with a modified chkstk.S, like: (Sorry didn't mention before " ./configure --prefix=/usr/local")
	.file		"chkstk.S"

	.text

#ifdef __APPLE__
#define __chkstk ___chkstk
#endif

	/* Function called to loop on the process.  */
	.align 4
	.globl __chkstk
__chkstk:
	cmp	r0,#0
	beq	ext_fn
	sub	r0,r0,#4		/* 4 bytes already used by call.  */
	sub	r0,r0,r13
	b	r0
ext_fn:
	bx lr
#ifdef __ELF__
	.size	__chkstk, . - __chkstk
#endif

	.ident	"Written by T.Gingold"

#if defined(__linux__) && defined(__ELF__)
	.section .note.GNU-stack,"",%progbits
#endif

Seams that I have other issue, but in x86 I had compiled...

eine, 2020-12-22:
Still, I don't understand what are you trying to achieve. Please, be explicit about the full sequence of commands that you are using. What do you do differently compared to the 13 line snippet in `setup_zynq.sh`?

zipotron, 2020-12-22:
Hi Eine, sorry, sometimes I am not so concise when I am writing, this code that I pasted correspond a my own modification of the file chkstk.S, in this modification i did the equivalent translation from x86 assembly to ARM (I hope is right done), and I got the mentioned compilation error. But going throw the post #1535 I got the same error, exactly the same, and this make me think that my assembly translation is right and I have now another issue whit something else, but I have no idea witch kind of issue is this new one, and I would like to ask you if have an a idea.
Thanks a lot!

tgingold, 2020-12-22:
From the error message from gcc, I fear you need to install the GNAT compiler.

eine, 2020-12-22:
@zipotron, if I understood correctly, you could successfully compile GHDL with LLVM backend using `CXX=clang++ ../configure --with-llvm-config=llvm-config --default-pic --disable-werror`. Therefore, you can use GHDL on ARM already (either for simulation or synthesis).

Now, are you willing to add ARM support for the mcode backend? I believe that's what `chkstk.S` is about. mcode is a backend, different from LLVM, which generates machine code.

zipotron, 2020-12-22:
No exactly... I got the same error in bought cases, (using CXX=clang++ ../configure --with-llvm-config=llvm-config --default-pic --disable-werror and modifiing chkstk.S).
And well, I wanted to give ARM support modifying chkstk.S, but I dont know if its a good idea...
The point is, I got the same compile error, I would expect to get error touching assembly code, expecially when I am not sure for what is that assembly code. But I am getting the same error using LLVM, and this make me think that there is something else bad and I don't know what.
But in resume, I would like to compile GHDL in ARM, and I don't know how...

eine, 2020-12-22:
What ARM board/device and which OS are you using?

zipotron, 2020-12-23:
Its a Raspberry pi II (Armv7h) with the last Slackware version gcc 10.0.2 llvm 10. I had no problems for compile Icestom and Yosys, and is have all the dependencies for compile GHDL I guess...

zipotron, 2020-12-23:
@tgingold Sorry, forgot answering you, really sorry, yes I have gcc-gnat 10.0.2 installed from the beggining, is not this the issue

tgingold, 2020-12-23:
So why do you have this error message:
gcc: fatal error: cannot execute ‘gnat1’: execvp: No such file or directory

zipotron, 2020-12-23:
This is my question... Lets put a highlight timeline: (correction, my gcc is 10.2.0)
In my x86 computer:
./configure error -- no gnat found
install gnat
./configure ok
make ok

In raspberry pi:
./configure error -- no gnat found
install gnat (gcc-gnat.10.2.0)
./configure ok
make error -- assembly code  chkstk.S
modify  chkstk.S for ARM
make error --  ‘gnat1’: execvp: No such file or directory
lets rollback chkstk.S and try CXX=clang++ ../configure --with-llvm-config=llvm-config --default-pic --disable-werror
make error --  ‘gnat1’: execvp: No such file or directory

eine, 2020-12-23:
@zipotron, please try, in raspberry pi:

1. Install GCC, LLVM, clang and gnat.
2. Git clone a fresh GHDL.
3. `cd ghdl`.
4. `CXX=clang++ ../configure --with-llvm-config=llvm-config --default-pic --disable-werror`
5. `make`.

tgingold, 2020-12-23:
It looks like the gcc-gnat.10.2.0 on arm is broken.

zipotron, 2020-12-23:
Guys, thanks a lot for the help, I go a SUCCESS compilation, I just update LLVM to the las version, the 11.0 and compiled using CXX=clang++ ../configure --with-llvm-config=llvm-config --default-pic --disable-werror without problem. Thanks and sorry for disturbing you because the origin was 100% in my side.
Thanks again for the project!

closed 2020-12-23 by zipotron

eine, 2020-12-23:
@tgingold, my guess is that he/she was running configure several times on the same location, first with incorrect options (mcode) and then `with-llvm-config`. I think that making it clean is what fixed it.

tgingold, 2020-12-24:
Humm, that doesn't explain the gnat1 not found issue.  But anyway...