ghdl/ghdl #1544: Different Block RAM behaviour, compared to Xilinx Vivado Simulator

carljohnsen, 2020-12-22:
**Description**
When implementing a Block RAM, following the structure in _Dual-Port Block RAM with Two Write Ports in Read-First Mode (VHDL)_ from [the Vivado Design Suite User Guide](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug901-vivado-synthesis.pdf) though with a shared clock signal, the simulation in GHDL differs from the Xilinx Vivado Simulator when reading and writing to the same address on both ports. The A port of the RAM reads the value written by the B port within the same clock cycle. 

**Expected behaviour**
I'd assume the GHDL simulation follows the Vivado simulation, as the ports share the same clock and should be synchronized.

**How to reproduce?**
An example project have been attached: 
[vhdl.zip](https://github.com/ghdl/ghdl/files/5730204/vhdl.zip)

To run with GHDL (assuming `ghdl` has been added to the `$PATH`), the `Makefile` will build and run the project by running  `make`

To run with Xilinx Vivado, all of the source files should be added to an empty project, set the `BlockRAM_tb` as top in simulation, run "behavioral  simulation", and run the TCL command `run all`. Vivado might throw an error, if the path to the csv file isn't correct. In that case: the string on line 76 in `TestBench_BlockRAM.vhdl` should reflect the correct path to the trace file: 
```
        constant filename : string := "./trace.csv";
```

The attached zip also contains two `.vcd` trace files generated by GHDL and Xilinx Vivado Simulator, called `ghdl.vcd` and `vivado.vcd` respectively. 

**Context**
- OS: Windows 10 / Ubuntu 18.04
- Origin: Docker image: ghdl/ghdl:ubuntu18-mcode

diogratia, 2020-12-22:
Your model appears to use pre-2002 shared variables and two different processes to write to the 'block RAM' avoiding the signal drivers being elaborated for the longest static prefix  in each process.  Otherwise `RAM` could be class signal where (asynchronous) reads are independent and all writes occur in the same process avoiding multiple drivers or without the `shared` reserved word as a variable in a single process which incidentally would assure order through a sequence of statements like a signal `RAM` single write process.

You appear to be counting on the order in which processes resume and subsequently suspend during simulation but that is not defined in the standard, there's an expectation processes communicate using signals. You appear to be seeing a different process execution order between ghdl and Vivado Simulator.

The use of shared variables is non-portable for this reason. The issue appears to be with your model being non-portable while choosing the behavior of one simulator to be a 'golden standard'. Matching write priority of any Xilinx model can be accomplished by having both writes in the same process.

diogratia, 2020-12-23:
>the simulation in GHDL differs from the Xilinx Vivado Simulator when reading and writing to the same address on both ports.

blockram_tb produces a lot of errors. The first is at 85 ns:

```none
ghdl -r blockram_tb --wave=blockram_tb.ghw
TestBench_BlockRAM.vhdl:218:25:@85ns:(report error): Value for TrueDualPortMemory_1_IReadResultA_UInt32_Data in cycle 8 was: 00000000000000000000000000000001 but should have been: 00000000000000000000000000000000
TestBench_BlockRAM.vhdl:218:25:@95ns:(report error): Value for TrueDualPortMemory_1_IReadResultA_UInt32_Data in cycle 9 was: 00000000000000000000000000000010 but should have been: 00000000000000000000000000000000
```
Upon examination `truedualportmemory_1_icontrola_uinit32_iswriting` is never `'1' `during the testbench simulation wherein the result must indeed be coming from a write to the b port on the same clock edge.

[image]

The reads and writes are in different processes:

```vhdl
    process (CLK, RST)
    begin
        if RST = '1' then
            ReadResultA_Data_Vector <= (others => '0');
            FIN_A <= '0';
        elsif rising_edge(CLK) then
            if (ControlA_Enabled = '1') then
                ReadResultA_Data_Vector <= RAM(to_integer(unsigned(ControlA_Address_Vector)));

                if (ControlA_IsWriting = '1') then
                    RAM(to_integer(unsigned(ControlA_Address_Vector))) := ControlA_Data_Vector;
                end if;
            end if;
            FIN_A <= not RDY;
        end if;
    end process;

    process (CLK, RST)
    begin
        if RST = '1' then
            ReadResultB_Data_Vector <= (others => '0');
            FIN_B <= '0';
        elsif rising_edge(CLK) then
            if (ControlB_Enabled = '1') then
                ReadResultB_Data_Vector <= RAM(to_integer(unsigned(ControlB_Address_Vector)));

                if (ControlB_IsWriting = '1') then
                    RAM(to_integer(unsigned(ControlB_Address_Vector))) := ControlB_Data_Vector;
                end if;
            end if;
            FIN_B <= not RDY;
        end if;
    end process;
````

However RAM is a pre-2002 shared variable:
```vhdl
type ram_type is array (0 to 15) of std_logic_vector (31 downto 0);
    shared variable RAM : ram_type := (others => x"00000000");
```

There are differences between signals and variables. Signal updates occur in a different part of the simulation cycle than where processes are resumed and subsequently suspended. A signal update due to an assignment in the current simulation cycle can't be seen in any process until the next simulation cycle.

That's not the case for variables whose value is updated immediately.

The result of using a shared variable is that you would be vulnerable to order of processes being resumed and subsequently suspended. This likely explains the difference between ghdl and Vivado Simulator behavior. A process runs until it suspends, signals and simulation cycles (here delta cycles) provide the ability to model concurrency.

VHDL does not have a defined order for process resumption. This is why using shared variables is non-portable (-1993 Annex C). Processes are expected to communicate with signals which have ordered updates.

Possible solutions here would be to make `RAM` class signal and place both writes in the same process to avoid multiple drivers, or  merge the processes and order operations - reads before writes which would also allow `RAM` to be a process variable instead of a shared variable.

diogratia, 2020-12-23:
See the marked up architecture:

```vhdl
architecture RTL_MARKUP of TrueDualPortMemory_1_UInt32 is 

type ram_type is array (0 to 15) of std_logic_vector (31 downto 0);
    -- shared variable RAM : ram_type := (others => x"00000000");
    signal ControlA_Data_Vector: std_logic_vector (31 downto 0);
    signal ControlB_Data_Vector: std_logic_vector (31 downto 0);
    signal ReadResultA_Data_Vector: std_logic_vector (31 downto 0);
    signal ReadResultB_Data_Vector: std_logic_vector (31 downto 0);
    signal ControlA_Address_Vector: std_logic_vector (3 downto 0);
    signal ControlB_Address_Vector: std_logic_vector (3 downto 0);
    signal FIN_A : std_logic;
    signal FIN_B : std_logic;
begin 

    -- process (CLK, RST)
    -- begin
    --     if RST = '1' then
    --         ReadResultA_Data_Vector <= (others => '0');
    --         FIN_A <= '0';
    --     elsif rising_edge(CLK) then
    --         if (ControlA_Enabled = '1') then
    --             ReadResultA_Data_Vector <= RAM(to_integer(unsigned(ControlA_Address_Vector)));
    --
    --             if (ControlA_IsWriting = '1') then
    --                 RAM(to_integer(unsigned(ControlA_Address_Vector))) := ControlA_Data_Vector;
    --             end if;
    --         FIN_A <= not RDY;  -- SHOULD THIS BE HERE?
    --         end if;
    --         -- FIN_A <= not RDY;
    --     end if;
    -- end process;
    --
    -- process (CLK, RST)
    -- begin
    --     if RST = '1' then
    --         ReadResultB_Data_Vector <= (others => '0');
    --         FIN_B <= '0';
    --     elsif rising_edge(CLK) then
    --         if (ControlB_Enabled = '1') then
    --             ReadResultB_Data_Vector <= RAM(to_integer(unsigned(ControlB_Address_Vector)));
    --
    --             if (ControlB_IsWriting = '1') then
    --                 RAM(to_integer(unsigned(ControlB_Address_Vector))) := ControlB_Data_Vector;
    --             end if;
    --         FIN_B <= not RDY;  -- SHOULD THIS BE HERE?
    --         end if;
    --         -- FIN_B <= not RDY;
    --     end if;
    -- end process;
    
SINGLE_ORDERED_PROCESS:
    process (CLK, RST)
        variable RAM: ram_type := (others => x"00000000");
    begin
        if RST = '1' then
            ReadResultA_Data_Vector <= (others => '0');
            ReadResultB_Data_Vector <= (others => '0');
            FIN_A <= '0';
            FIN_B <= '0';
        elsif rising_edge(CLK) then
            if ControlA_Enabled = '1' then
                ReadResultA_Data_Vector <= 
                    RAM(to_integer(unsigned(ControlA_Address_Vector)));
            end if;
            if ControlB_Enabled = '1' then
                ReadResultB_Data_Vector <= 
                    RAM(to_integer(unsigned(ControlB_Address_Vector)));
            end if;
            if ControlA_Enabled = '1' and ControlA_IsWriting = '1' then
                RAM(to_integer(unsigned(ControlA_Address_Vector))) := 
                    ControlA_Data_Vector;
            end if;
            if ControlB_Enabled = '1' and ControlB_IsWriting = '1' then
                RAM(to_integer(unsigned(ControlB_Address_Vector))) := 
                    ControlB_Data_Vector;
            end if;
            FIN_A <= not RDY;  -- SHOULD BE IN THE FIRST INNER IF STATEMENT?
            FIN_B <= not RDY;  -- SHOULD BE IN THE SECOND INNER IF STATEMENT?
        end if;
    end process;

    TrueDualPortMemory_1_UInt32_Helper: process(RST, FIN_A, FIN_B)
    begin
        if RST = '1' then
            FIN <= '0';
        elsif FIN_A = FIN_B then
            FIN <= FIN_A;
        end if;
    end process;

    ControlA_Address_Vector <= STD_LOGIC_VECTOR(resize(unsigned(ControlA_Address), 4));
    ControlB_Address_Vector <= STD_LOGIC_VECTOR(resize(unsigned(ControlB_Address), 4));
    ControlA_Data_Vector <= STD_LOGIC_VECTOR(ControlA_Data);
    ReadResultA_Data <= UNSIGNED(ReadResultA_Data_Vector);
    ControlB_Data_Vector <= STD_LOGIC_VECTOR(ControlB_Data);
    ReadResultB_Data <= UNSIGNED(ReadResultB_Data_Vector);

end RTL_MARKUP;
```

This has two changes - evaluate before assign for `RAM`, which is now a process variable because the two ports processes needed merging to allow read/write operation ordering.

Variables (including pre-2002 shared variables) assignment takes immediate effect. As long as the processes are separate the use of a (shared) variable is non-portable, requiring deterministic order of resumption and subsequent suspension of process execution. Otherwise a `RAM` write can occur before a read in the same simulation cycle.  Signals are protected from this by writing to a projected output waveform (a queue) holding future values and have signal updates applied in a different part of a simulation cycle from resumption/execution/suspension of processes.

This also says `RAM` would be fine as a signal with no change to the processes. 

`RAM` can be declared as a variable in the process instead of using a shared variable absent any other processes accessing it.

```none
make
mkdir work
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others system_types.vhdl
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others Types_BlockRAM.vhdl
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others TrueDualPortMemory_1_UInt32.vhdl
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others csv_util.vhdl
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others BlockRAM.vhdl
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others TestBench_BlockRAM.vhdl
ghdl -e --std=93c --ieee=synopsys --workdir=work --warn-no-others BlockRAM_tb
ghdl -r --std=93c --ieee=synopsys --workdir=work --warn-no-others BlockRAM_tb --vcd=trace.vcd
TestBench_BlockRAM.vhdl:253:13:@3015ns:(report note): completed successfully after 302 clockcycles
ghdl -a --std=93c --ieee=synopsys --workdir=work --warn-no-others Export_BlockRAM.vhdl
```

carljohnsen, 2020-12-23:
Makes sense! And that's also how I would like to write the RAM and how I initially wrote the RAM. The `trace.csv` comes from a [program](https://github.com/kenkendk/sme/blob/master/src/SME.Components/TrueDualPortMemory.cs)  written in C#, with a similar structure as the one you proposed:
```csharp
            if (ControlA.Enabled)
                ReadResultA.Data = m_memory[ControlA.Address];

            if (ControlB.Enabled)
                ReadResultB.Data = m_memory[ControlB.Address];

            if (ControlA.Enabled && ControlA.IsWriting)
                m_memory[ControlA.Address] = ControlA.Data;

            if (ControlB.Enabled && ControlB.IsWriting)
                m_memory[ControlB.Address] = ControlB.Data;
```

However, while the proposed vhdl produces the correct results, Vivado isn't able to infer Block RAM. Implementing the proposed single process vhdl with a signal, instead of a process variable, doesn't change the outcome of the schematic:

[image]

Which is why I was following [the Xilinx Vivado Design Suite User Guide](https://www.xilinx.com/support/documentation/sw_manuals/xilinx2020_2/ug901-vivado-synthesis.pdf), where the dual process solution correctly infers Block RAM: 

[image]

Paebbels, 2020-12-23:
Xilinx describes since years the old style of writing RAMs with variables. While this is a solution (for Xilinx tools) it's not portable as explained by @diogratia. There synthesizer also understands a better form of using signals in a single process. For dual port / dual clock RAMs, you can even use multiple clocks per process. See [PoC.mem.ocram.tdp](https://github.com/VLSI-EDA/PoC/blob/master/src/mem/ocram/ocram_tdp.vhdl?ts=2#L142-L169)

diogratia, 2020-12-23:
>**_However, while the proposed vhdl produces the correct results, Vivado isn't able to infer Block RAM._** 

```vhdl
type ram_type is array (0 to 15) of std_logic_vector (31 downto 0);
```
has too small an index range and too wide a width to to be defined as block ram automatically, and as shown instead uses registers.

Paebbels, 2020-12-23:
@diogratia the width is ok, because BlockRAM has a native width of 36 bit (32 bit data + 4 bit parity).

@carljohnsen when you read your synthesis logs, you'll find a hint from HDL Advisor, why it didn't use a BlockRAM. While RAM is to small for a BlockRAM, it should use a DistributedRAM (LUT-RAM).

Please also note: elaborated view shows what Vivado understood from your source files, not what it is using in hardware! Components identified in elaborated view are mapped to resources provided by the selected device.

carljohnsen, 2020-12-23:
> Please also note: elaborated view shows what Vivado understood from your source files, not what it is using in hardware! Components identified in elaborated view are mapped to resources provided by the selected device.

True. I used elaboration as it usually tells me whether it inferred everything correct in a few seconds, compared to several minutes of synthesis, when everything expands.

> @carljohnsen when you read your synthesis logs, you'll find a hint from HDL Advisor, why it didn't use a BlockRAM. While RAM is to small for a BlockRAM, it should use a DistributedRAM (LUT-RAM).

16 words were just to have a small example, which would run faster through Vivado. I've increased the index range, as suggested by @diogratia, to 1024 along with the address width, to come closer to a 36k RAM, which the board supports. 
```vhdl
    type ram_type is array (0 to 1023) of std_logic_vector (31 downto 0);

...    

    ControlA_Address_Vector <= STD_LOGIC_VECTOR(ControlA_Address);
    ControlB_Address_Vector <= STD_LOGIC_VECTOR(ControlB_Address);
```
The single process approach still expands into registers, resulting in 83948 LUTs and 32384 FFs. The Vivado log reports:
```
Reason is one or more of the following :
        1: RAM has multiple writes via different ports in same process. If RAM inferencing intended, write to one port per process.
        2: Unable to determine number of words or word size in RAM.
        3: No valid read/write found for RAM.
RAM "RAM_reg" dissolved into registers
```

I've attached the [runme.log](https://github.com/ghdl/ghdl/files/5734692/runme.log), in case you can see some secrets I've missed :)

> Xilinx describes since years the old style of writing RAMs with variables. While this is a solution (for Xilinx tools) it's not portable as explained by @diogratia. There synthesizer also [...]

I'll try this implementation!

carljohnsen, 2020-12-23:
Ok, following the PoC.mem.ocram.tdp approach, I had an implementation, which worked both in Vivado and GHDL. However, the RAM became write-first, and there were no output latch. I noticed the implementation used multiple `if rising_edge` inside the process block, so I tried to make a variant of the suggestion by @diogratia:

```vhdl
    type ram_type is array (0 to 15) of std_logic_vector (31 downto 0);
    signal RAM : ram_type := (others => x"00000000");
    signal ControlA_Data_Vector: std_logic_vector (31 downto 0);
    signal ControlB_Data_Vector: std_logic_vector (31 downto 0);
    signal ReadResultA_Data_Vector: std_logic_vector (31 downto 0);
    signal ReadResultB_Data_Vector: std_logic_vector (31 downto 0);
    signal ControlA_Address_Vector: std_logic_vector (3 downto 0);
    signal ControlB_Address_Vector: std_logic_vector (3 downto 0);
    signal FIN_A : std_logic;
    signal FIN_B : std_logic;
begin

    process (CLK, RST)
    begin
        if RST = '1' then
            ReadResultA_Data_Vector <= (others => '0');
        elsif rising_edge(CLK) then
            if (ControlA_Enabled = '1') then
                ReadResultA_Data_Vector 
                    <= RAM(to_integer(unsigned(ControlA_Address_Vector)));
            end if;
        end if;

        if RST = '1' then
            ReadResultB_Data_Vector <= (others => '0');
        elsif rising_edge(CLK) then
            if (ControlB_Enabled = '1') then
                ReadResultB_Data_Vector 
                    <= RAM(to_integer(unsigned(ControlB_Address_Vector)));
            end if;
        end if;

        if RST = '1' then
            FIN_A <= '0';
        elsif rising_edge(CLK) then
            if (ControlA_Enabled = '1') and (ControlA_IsWriting = '1') then
                RAM(to_integer(unsigned(ControlA_Address_Vector))) 
                    <= ControlA_Data_Vector;
            end if;
            FIN_A <= not RDY;
        end if;

        if RST = '1' then
            FIN_B <= '0';
        elsif rising_edge(CLK) then
            if (ControlB_Enabled = '1') and (ControlB_IsWriting = '1') then
                RAM(to_integer(unsigned(ControlB_Address_Vector))) 
                    <= ControlB_Data_Vector;
            end if;
            FIN_B <= not RDY;
        end if;
    end process;
```

This version behaves the same in both Vivado simulation and GHDL simulation, and it also correctly infers RAM in Vivado, both in elaboration and synthesis. Changing the order of the two reads and two writes, it can also run in write-first mode, which again behaves the same in both simulators and is inferred correctly. 

Thank you for the help, even though it seems the problem was rooted in the design guide having a non-portable implementation, not GHDL! :)

eine, 2020-12-23:
Thanks all for this very interesting dialogue.

I'm closing the issue because it's solved.

closed 2020-12-23 by eine