ghdl/ghdl #1545: --synth is not available on macOS

myzinsky, 2020-12-22:
**Description**
In the documentation it is mentioned that since 0.37 an experimental --synth feature is available. However, in the version build for macOS this is not available. 

**Expected behaviour**

The parameter --synth should exists according to documentation

```bash
ghdl --synth ifelse.vhd
ghdl:error: unknown command '--synth', try --help
```

**Context**

- OS: macOS 
- Origin:
  - [ ] Package manager: `brew`
  - [ ] Version: ` 0.37`
  - [ ] Released binaries: `https://github.com/ghdl/ghdl/releases/download/v0.37/ghdl-0.37-macosx-mcode.tgz`
  - [ ] Built from sources: `cb870085dd55167eda162b2d8b0020b30574b47fa8f8dfc7a1fd6aa3eb32ee91 `

myzinsky, 2020-12-22:
When building the tool from scratch it is there!

tgingold, 2020-12-23:
It is possible that the 0.37 release for macosx was built without the synthesis feature as it requires a slightly more recent version of the compiler.

closed 2020-12-23 by tgingold

eine, 2020-12-23:
In fact, libghdl is disabled for macOS on CI: https://github.com/ghdl/ghdl/blob/master/dist/ci-run.sh#L400. Ref #867.