ghdl/ghdl #1546: GHDL logo in README.md is poorly visible in dark mode

avinal, 2020-12-23:
Since GitHub rolled out dark theme a few days ago, more and more users are shifting to it. The **GHDL** logo is barely visible in this mode. (See below). However, this doesn't affect the project in any way but important for professional and aesthetic purposes.
[image]

Possible fix: Add an outline to the logo or change the color to gray

tgingold, 2020-12-28:
a PR is always welcome!

avinal, 2020-12-28:
> 
> 
> a PR is always welcome!

I would like to know the preference before pr, I have following suggestions:
- Change color to gray (or whatever you suggest)
- Add an outline to text (white outline)
- Add a white/textured background to the logo

eine, 2021-01-01:
A proposal: https://github.com/eine/ghdl/blob/logo/README.md

[image]

The concept: the logo is defined as four 7x5 pixel characters, spaced by one pixel. The following script generates an SVG to be used as a reference: https://github.com/eine/ghdl/blob/logo/logo/gen_logo.py.

Pixels can optionally have an outline. In these examples, there is a transparent outline (10%). Behind the pixels, there is a gaussian blur. The background is white noise behind a black layer with 95% opacity. There is a frame with a gradient.

See an square version below, which might be used for the organisation and as a favicon:

[image]

[image]

[image]

Using green "LEDs" instead of blue:

[image]

[image]

tgingold, 2021-01-01:
Up to you!

tgingold, 2021-01-01:
I'm fine with the blue logo.

eine, 2021-01-01:
I updated the README, the docs and the favicon of the docs with the blue logo proposed above. I removed the black background, since it didn't feel necessary with the blue version.

@tgingold, I cannot change the icon of the organisation in GitHub. Can you please use https://github.com/ghdl/ghdl/blob/master/logo/icon.png ?

closed 2021-01-01 by eine