ghdl/ghdl #1548: Unhandled Exception SIGFPE

Werni2A, 2020-12-25:
**Description**
GHDL throws an exception when running the example below.

I think the issue is correlated with the comparison of integer `sCtr` and a time value.

**Expected behaviour**
The exception should be caught instead of crashing GHDL. Maybe show an error that explains what the user is doing wrong.

**How to reproduce?**
Run the following example

```vhd :file: reproducer.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity reproducer is
    port(CLK : in std_logic);
end reproducer;

architecture Behavioral of reproducer is
    constant ClkFreq   : integer := 400_000_000; -- 400 MHz
    constant ClkPeriod : time    := 1 / ClkFreq * 1 sec;

    signal sCtr : integer   := 0;
    signal foo  : std_logic := '0';

begin

    process(sCtr)
    begin
        if sCtr < 1 us / ClkPeriod then
            foo <= '1';
        else
            foo <= '0';
        end if;
    end process;

end Behavioral;
```

```vhd :file: tb_reproducer.vhd
library ieee;
use ieee.std_logic_1164.all;

entity tb_reproducer is
end tb_reproducer;

architecture Behavioral of tb_reproducer is

    component reproducer is
        port(CLK : in std_logic);
    end component;

    signal sCLK : std_logic := '0';

begin

    uut: reproducer port map(CLK => sCLK);

end Behavioral;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a reproducer.vhd
ghdl -a tb_reproducer.vhd
ghdl -e tb_reproducer
ghdl -r tb_reproducer --vcd=reproducer.vcd
```

**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `8ed352778368cfbff239bb2a89fc6a937c65fc26`

Terminal output:
```
Execution of ghdl terminated by unhandled exception
raised CONSTRAINT_ERROR : SIGFPE
Call stack traceback locations:
0x7f17ce20e3be 0x7f17cdf81178
```

tgingold, 2020-12-28:
I think this is just an overflow.  The time values are represented on 64 bits.  1 sec = 10**(-15) fs, which needs ~45 bits.
400Mhz = 4*10**8 which needs ~26 bits.  So the result needs more than 70 bits.
And of course ClkPeriod is 0 which even if the first overflow is not detected will result in a division by 0.

closed 2020-12-28 by tgingold

Xiretza, 2020-12-28:
To expand on Tristan's comment: I don't see any chance of an overflow occurring here, but what you actually want is probably `1 sec / ClkFreq`, since `(1 / 400000000) * 1 sec` is just `0 * 1 sec` due to integer division.

tgingold, 2020-12-28:
Ah yes, you're right.  The overflow doesn't happen.