ghdl/ghdl #1550: Initializing libghdl (Python) causes an exception

Paebbels, 2020-12-25:
**Description**
When trying to initialize libghdl (via Python wrapper), an exception is raised.

> raised CONSTRAINT_ERROR : errorout.adb:289 access check failed

GHDL was compiled from scratch (8ed352778368cfbff239bb2a89fc6a937c65fc26) under Ubuntu 20.04 (in WSL2 on Windows 10) with GCC-9, GNAT-9 and LLVM-9. Python is 3.8.5. libghdl was installed using `pip3 install .` in GHDLs `python` directory. GHDL was installed to `/opt/ghdl/0.37-dev` via `make install`.

**Expected behavior**
As no input files was yet handed over to `libghdl`, no error is expected.

**How to reproduce?**

```python
import libghdl

class GHDL:
	def __init__(self):
		libghdl.analyze_init()

def main():
	ghdl = GHDL()

if __name__ == "__main__":
	main()
```

This code was called like this:
`GHDL_PREFIX=/opt/ghdl/0.37-dev/lib/ python pyVHDLModel/GHDL.py`

Just importing `libghdl` gives no exception.

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04 running in WSL2 on Windows 10
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `8ed352778368cfbff239bb2a89fc6a937c65fc26`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

No `GHDL Bug occurred` block was printed.

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

Paebbels, 2020-12-25:
To narrow it down, I tried to execute the `units01` test.

```bash
paebbels@Paebbels-Laptop:/mnt/c/git/GitHub/Paebbels/ghdl/testsuite/python/units01$ GHDL_PREFIX=/opt/ghdl/0.37-dev/lib/ python show_units.py

raised CONSTRAINT_ERROR : errorout.adb:289 access check failed
paebbels@Paebbels-Laptop:/mnt/c/git/GitHub/Paebbels/ghdl/testsuite/python/units01$ GHDL=/opt/ghdl/0.37-dev/bin/ghdl python show_units.py
entity e1
architecture behav
```

It fails if called with `GHDL_PREFIX`, but this was enough for `libghdl` to be loaded. And according to the source code in `__init__.py` it should be correct to specify the prefix.

The working variant uses just `GHDL` as specified in the sh file for the test.

-----------
In general the environment variable `GHDL_PREFIX` is misleading, to be the path to library directory, which by the way must end with a trailing slash. Prefix normally refers to the installation directory like it's used by the configure command.  
`CC=gcc-9 CXX=clang++-9 ../configure "--prefix=/opt/ghdl/0.37-dev" --with-llvm-config=llvm-config-9`

tgingold, 2020-12-26:
I have changed units01 tests.  Now errors are correctly reported.  What is the error message ?

Paebbels, 2020-12-26:
Now the error is this:
```
paebbels@Paebbels-Laptop:/opt/ghdl/source/testsuite/python/units01$ GHDL_PREFIX=/opt/ghdl/0.37-dev/lib/ python show_units.py
warning: ieee library directory '/opt/ghdl/0.37-dev/lib//ieee/v08/' not found
error: cannot find "std" library

raised OPTIONS.OPTION_ERROR : libraries.adb:671
```

The correct path to `std` would be: `/opt/ghdl/0.37-dev/lib/ghdl/std/...`.  
So `ghdl` is missing between the double slashes.

tgingold, 2020-12-26:
So, you just need to append '/ghdl' to your definition of GHDL_PREFIX

Paebbels, 2020-12-26:
OK, this works.

-------------
I noticed you added:
```python
# Print error messages on the console
    errorout_console.Install_Handler()
```

How to handle this in a more pythonic way?
* At best we could raise an exception.
* We could register a callback function.

tgingold, 2020-12-26:
For reporting the errors or to avoid the 'crash' ?

Paebbels, 2020-12-26:
For reporting, so the python code can decide if to use stdout, stderr, ignore it and how to format it.

tgingold, 2020-12-26:
There are hooks, so you could do whatever you want.

See for example in vhdl_langserver as an example.

Paebbels, 2020-12-26:
I see in [`workspace.py`](https://github.com/ghdl/ghdl/blob/master/python/vhdl_langserver/workspace.py?ts=2#L42) that LSP uses `errorout_memory.Install_Handler()`, which seams to capture error messages in memory instead of `stderr`.

How does my code know, if `errorout_memory` has captured warnings or errors?

There seams to be a related [line](https://github.com/ghdl/ghdl/blob/master/python/vhdl_langserver/workspace.py?ts=2#L52):  
`libghdl.thin.errorout.Enable_Warning(errorout.Msgid.Warnid_Unused, True)`

tgingold, 2020-12-26:
Correct.  You can get the number of error messages using Get_Nbr_Messages.

Errors are always reported (ie stored), while non-enabled warnings aren't reported.

tgingold, 2020-12-28:
Also, note the API has been enhanced to return a status instead of an exception if the std library is not found.

closed 2020-12-28 by tgingold

Paebbels, 2020-12-28:
Ok, I found this:
```python
class Error_Message(Structure):
    _fields_ = [
        ("id", c_int8),
        ("group", c_int8),
        ("file", c_int32),
        ("line", c_int32),
        ("offset", c_int32),
        ("length", c_int32),
    ]

# ...

Get_Nbr_Messages = libghdl.errorout__memory__get_nbr_messages

Get_Error_Record = libghdl.errorout__memory__get_error_record
Get_Error_Record.argstypes = [c_int32]
Get_Error_Record.restype = Error_Message

Get_Error_Message = libghdl.errorout__memory__get_error_message_addr
Get_Error_Message.argstype = [c_int32]
Get_Error_Message.restype = c_char_p
```

I have the following questions:
1. How to convert the captured error to a string in Python?
2. What's the difference between `Error_Record` and `Error_Message`?
3. `Error_Message` contains a position where it happend in a file, right?  
   But not all errors can be related to a position, so is `Error_Message` an element in `Error_Record`?
4. What if, multiple errors are reported?  
   How to go to the next error? I don't see an iterator/linked list or chain or accessing errors by index. Or is a `Get_*` call decrementing/clearing the structure?

tgingold, 2020-12-28:
1) Get_Error_Message returns the error message string for error N.
2) while Get_Error_Record returns additional information (location)
3) Yes: file, line and offset in the line.
4) There are Get_Nbr_Messages available, from 1 to N.  Once messages are read you can clear all the messages by calling Clear_Errors.

In general the functions are documented in the corresponding Ada package (here errorout-memory.ads).  And vhdl_langserver uses most of them.