ghdl/ghdl #1553: Higher level Python API for GHDL

Paebbels, 2020-12-26:
**Is your feature request related to a problem? Please describe.**
Related to:
* https://github.com/ghdl/ghdl/pull/1449#issuecomment-682069221

The thin wrapper around GHDL's `libghdl` is hard to use:
- no pythonic interface
- procedural API instead of OOP  
  (due to how underlying Ada works, which is used to code `libghdl` and GHDL in general)

**Describe the solution you'd like**
[`pyVHDLParser`](https://github.com/Paebbels/pyVHDLParser/) offers a generic independent [language model of VHDL](https://github.com/Paebbels/pyVHDLParser/blob/master/pyVHDLParser/VHDLModel.py?ts=2).

I'm proposing:
* split of `VHDLModel.py` from pyVHDLParser
* create a new independent repository under `https://github.com/vhdl/pyVHDL`
* register the Python package at PyPI as `pyVHDL`.
* Implement an adapter from GHDL (`libghdl`) to `pyVHDL`.

**Describe alternatives you've considered**
*None*

**Additional context**
Multiple (input) tools can create such a generic representation of VHDL.  
Currently it would be GHDL and pyVHDLParser.

Multiple (output) tools could consume such a generic representation like:
* [Symbolator](https://github.com/kevinpt/symbolator)
* Sphinx plugins
* ...

Paebbels, 2020-12-26:
This is a first proof of concept implementation for entity, architecture, package, package body, context, configuration, generics and ports.

# Output for a simple prettyPrint
[image]

# `libghdl` Adapter to `VHDLModel`
```python
from pathlib import Path
from typing import Any

import libghdl
from libghdl.thin       import name_table
from libghdl.thin       import files_map
from libghdl.thin       import errorout_console
from libghdl.thin.vhdl  import nodes
from libghdl.thin.vhdl  import pyutils
from libghdl.thin.vhdl  import sem_lib
from pydecor            import export

from pyVHDLModel.VHDLModel import Design        as VHDLModel_Design, Modes
from pyVHDLModel.VHDLModel import Library       as VHDLModel_Library
from pyVHDLModel.VHDLModel import Document      as VHDLModel_Document
from pyVHDLModel.VHDLModel import Entity        as VHDLModel_Entity
from pyVHDLModel.VHDLModel import Architecture  as VHDLModel_Architecture
from pyVHDLModel.VHDLModel import Package       as VHDLModel_Package
from pyVHDLModel.VHDLModel import PackageBody   as VHDLModel_PackageBody
from pyVHDLModel.VHDLModel import Context       as VHDLModel_Context
from pyVHDLModel.VHDLModel import Configuration as VHDLModel_Configuration
from pyVHDLModel.VHDLModel import PortSignalInterfaceItem as VHDLModel_PortSignalInterfaceItem
from pyVHDLModel.VHDLModel import GenericConstantInterfaceItem as VHDLModel_GenericConstantInterfaceItem

__all__ = []
__api__ = __all__

@export
class GHDLBaseException(Exception):
	pass

@export
class LibGHDLException(GHDLBaseException):
	pass

@export
class GHDLException(GHDLBaseException):
	pass

@export
class GHDLMixin:
	_MODE_TRANSLATION = {
		nodes.Iir_Mode.In_Mode:      Modes.In,
		nodes.Iir_Mode.Out_Mode:     Modes.Out,
		nodes.Iir_Mode.Inout_Mode:   Modes.InOut,
		nodes.Iir_Mode.Buffer_Mode:  Modes.Buffer,
		nodes.Iir_Mode.Linkage_Mode: Modes.Linkage
	}

	@classmethod
	def _ghdlNodeToName(cls, node):
		"""Return the python string from node :param:`node` identifier"""
		return name_table.Get_Name_Ptr(nodes.Get_Identifier(node)).decode("utf-8")

	@classmethod
	def _ghdlPortToMode(cls, port):
		"""Return the mode of a port."""
		try:
			return cls._MODE_TRANSLATION[nodes.Get_Mode(port)]
		except KeyError:
			raise LibGHDLException("Unknown mode.")

@export
[... 209 lines cut ...]
			print("  Packages:")
			for package in doc.Packages:
				print("  - {name}".format(name=package.Name))
			print("  Package Bodies:")
			for packageBody in doc.PackageBodies:
				print("  - {name}".format(name=packageBody.Name))
			print("  Contexts:")
			for context in doc.Contexts:
				print("  - {name}".format(name=context.Name))
			print("  Configurations:")
			for configuration in doc.Configurations:
				print("  - {name}".format(name=configuration.Name))

def main():
	try:
		app = Application()
		app.addFile(Path("demo.vhdl"), "default_lib")
	except GHDLBaseException as ex:
		print(ex)

	app.prettyPrint()

if __name__ == "__main__":
	main()
```

umarcor, 2020-12-26:
Overall, I think this is a nice feature, which we really need as a community. We should discuss how to document it in GHDL, tho. 

As discussed in #1437, my understanding is that we have (will have) 4-5 different Python APIs, each built on top of the previous (from an abstraction level perspective):

0. The Ada internals are automatically exported to Python.
1. There are some very basic iterators for traversing the procedural/node based API.
2. There are some more elaborated iterators, which try to provide a more pythonic feeling but are still procedural/node based.
3. We want to provide a higher level and more pythonic API, based on classes, enumerations, dictionaries, etc. The VHDLModel proposed in this issue can be a replacement for this pythonic API (which does not exist yet). Or, the VHDLModel and GHDL's pythonic APIs can be complementary.

As explained in https://github.com/ghdl/ghdl/issues/1437#issuecomment-682577798, I find the layout of the sources difficult to understand and map to the 4 stages/levels above. Lower level modules are parents of higher level packages. I believe we should rethink/document that before going further.

Paebbels, 2020-12-27:
So [pyVHDLModel](https://github.com/VHDL/pyVHDLModel) is now online on GitHub and PyPI as v0.7.1 and a standalone repository/project.

Paebbels, 2020-12-28:
With PR #1556 we (+ @umarcor) are proposing a restructuring of the Python sources.