ghdl/ghdl #1555: libghdl (Python): How to specify the library name a file is analyzed into?

Paebbels, 2020-12-26:
Following the example/testcase in [`units01/show_ports.py`](https://github.com/ghdl/ghdl/blob/master/testsuite/python/units01/show_ports.py?ts=2), a file is parsed.  
In the example is currently no VHDL library specified where the design units are parsed or analyzed into.

```python
def __ghdl_init(self):
	# Read input file
	self.__ghdlFileID = name_table.Get_Identifier(str(self.Path).encode("utf_8"))
	self.__ghdlSourceFileEntry = files_map.Read_Source_File(name_table.Null_Identifier, self.__ghdlFileID)
	if self.__ghdlSourceFileEntry == files_map.No_Source_File_Entry:
		raise LibGHDLException("Cannot load file '{!s}'".format(self.Path))

	# Parse input file
	self.__ghdlFile =  sem_lib.Load_File(self.__ghdlSourceFileEntry)
```

How to specify the VHDL library for a file (or it's contained design units)?

tgingold, 2020-12-26:
Load_File just load and parse the file.  If you want to load, parse and analyze you should use the python libghdl.analyze_file function.

tgingold, 2020-12-28:
Design units are added to the work library.  The variable `work_library` designates the work library.

closed 2020-12-28 by tgingold

Paebbels, 2020-12-28:
How to set this variable with `libghdl`?
So far only saw accesses to functions and procedures.

tgingold, 2020-12-28:
It is declared in libghdl.thin.libraries as:
```
# Use .value
Work_Library = c_int32.in_dll(libghdl, "libraries__work_library")
```

So, you can directly read or write this variable from python.