ghdl/ghdl #1557: pyGHDL enhancements

umarcor, 2020-12-29:
Coming from #1556. 

Checklist for further enhancements:

- [ ] imports generated by pnodes
- [ ] auto-generate from `src/edif/Makefile` and/or `src/psl/Makefile`?
- [ ] test ghdl-ls with a client
- [ ] update docs
- [ ] GHDL vs PREFIX vs GHDL_PREFIX
- [ ] run pyunit from *any* location, run individual tests, run multiple, list tests... i.e. get the previous functionality back: https://github.com/ghdl/ghdl/blob/340fc792bba2ffdb4f930bc427a39ea3a1b659b2/testsuite/python/testsuite.sh#L14-L28

Paebbels, 2020-12-29:
**Update for `pyunit`:**
* Parallel execution with `pytest` and its plugin `pytest-forked` is now possible.  
  This works around the problem for independent unit testing that `libghdl` can only be initialized once.
* `pytest`s strange/strict naming conventions were disabled via `pytest.ini` in `testsuite/pyunit` to comply with normal unittest naming rules.
* `testsuite.sh` was adjusted to run from within `testsuite` directory.
* All 5 testcases are now found and executed.

See PR #1558

umarcor, 2020-12-30:
After the latest changes, running the test suite in-place fails on MSYS2 because it cannot find the shared library:

```
  make install.vpi.local
  make CC=gcc test
```

Paebbels, 2021-01-06:
I added another task to our list: integrate language server tests into pyunit. This is handled in PR #1585.