ghdl/ghdl #1560: Install dependencies (DLLs) with MinGW64 builds

Paebbels, 2020-12-30:
# Problem
When loading the `libghdl...dll` into Python via `pyGHDL.libghdl`, some dependencies are missing in the GHDL installation directory, if GHDL is installed outside of MinGW64 system.

These dependencies need to be added to the `/lib` directory of GHDL while installation.

# Solution
I used [Dependencies](https://github.com/lucasg/Dependencies) to discover needed libraries.

## Backend 'mcode'
### For `ghdl.exe`
* `zlib1.dll`

### For `libghdl-1_0_dev.dll`
* `libgnat-10.dll`
  * `libgcc_s_seh-1.dll`
    * `libwinpthread-1.dll`
* `libgcc_s_seh-1.dll`
  * `libwinpthread-1.dll`

[image]

## Backend 'llvm'
*tbd*

## Backend 'gcc'
*tbd*

# Alternatives
* Building GHDL and libghdl with GNAT-2020 seams to be broken.  
  (Investigating ...)

Paebbels, 2020-12-30:
I can confirm, after adding the mentioned DLLs to the GHDL bin and lib paths ad described above, I was able to execute native CPython 3.8 x86-64 in PyCharm and have the Python binding of `libghdl` parse a VHDL source file and display results of the DOM in PyCharm console.

[image]

tgingold, 2020-12-30:
I am not sure how to deal with this issue.
Do we need to always install those shared libraries ?  Do we need to provide an install script ?
Note: for libgnat, we could link it in the dll.

Paebbels, 2020-12-30:
To run libghdl (not tested `ghdl.exe` itself), I need these 3 additional dlls in `<prefix>/lib` aside to `libghdl...dll`. If so, I can directly use libghdl...dll via pyGHDL in normal CPython from Python.org. No MinGW, MSYS, etc. required.

This is not needed if GHDL compiled with MinGW is delivered as a MinGW package. Here it resides in a MinGW environment with matching search paths, etc. But it also requires to run Python scripts always in MinGW itself via MinGW console (or similar).

To create the environment I have now, I used `--prefix=/c/Tools/ghdl/0.37-dev-mcode` in configure. And after building with make, I used a `make install` to copy the files into destination.

I think the solution is to extend the make script for target `install` so also copy these 3 dlls into `prefix/lib`.

--------------
If we can reduce DLLs by linking, it would be fine, but 4 DLLs like these are not so bad :).

Paebbels, 2020-12-30:
Added dependencies if build with LLVM backend.

umarcor, 2020-12-30:
Background:

- [MSYS2 vs Other Projects](https://www.msys2.org/docs/what-is-msys2/#msys2-vs-other-projects)
- [How does MSYS2 differ from Cygwin](https://www.msys2.org/wiki/How-does-MSYS2-differ-from-Cygwin/)

Therefore, MSYS2 can provide two different kinds of packages: MSYS (which depend on `msys-2.0.dll`) and MINGW (either MINGW32 or MINGW64, which don't depend on any specific environment DLL). All tools should be built as MINGW packages, unless it is not possible. In GHDL, we always build MINGW packages only.

As a result, all the binaries generated in CI and published as 'nightly' assets are relocatable/portable. As a matter of fact, the `*.zst` files generated when using PKGBUILD recipes, and the zipfiles generated by AppVeyor contain "exactly" the same files. That is because both of them are executed on MSYS2/MINGW shells. The difference is that appveyor scripts handle the build and packaging manually. However, both have the same problems with missing DLLs.

NOTE: as it was discussed in the chat, pre-build zipfiles for future releases will not be provided. Travis will be removed. We might consider dropping AppVeyor too. Or, at least, the "release" feature. Note that AppVeyor seems to be faster than GitHub Actions. However, some of the test suites are not being executed in AppVeyor (I think those are synth and pyunit).

---

Hence, the missing DLLs are GHDL's own dependencies, not added by MSYS2. There are two solutions:

- Generating static builds of GHDL. Then, there would be no external dependencies, by definition.
- Discover and gather the DLLs to be loaded dynamically. The dependencies are known. Those are specified in the dockerfiles of ghdl/docker, and also in the PKGBUILD files of subdirs `dist/msys2*` in this repo.

# Generating static builds of GHDL

There is a project which is already generating and distributing static builds of GHDL: [open-tool-forge/fpga-toolchain](https://github.com/open-tool-forge/fpga-toolchain). The build procedure includes some tests for ensuring that the binaries do not depend on external libraries.

Unfortunately, mcode backend is used, and it's built with GCC. As a result, the binaries for x86_64 are useless for simulation. Synthesis works, and I'd say that libghdl works too, but simulation is broken (see open-tool-forge/fpga-toolchain#62).

At the same time, ghdl-yosys-plugin needs to be built into Yosys, because Yosys does not accept external/dynamic plugins on Windows (see open-tool-forge/fpga-toolchain#58). In this sense, static builds from fpga-toolchain are equivalent to the recently upstreamed yosys package on MSYS2 (see [mingw-w64-yosys](https://github.com/msys2/MINGW-packages/tree/master/mingw-w64-yosys) and open-tool-forge/fpga-toolchain#67).

Generating an static and portable build implies that the absolute location of GHDL cannot be guaranteed. Therefore, setting `GHDL_PREFIX` is required. See open-tool-forge/fpga-toolchain#59.

Precisely, #1547 is work in progress for enhancing fpga-toolchain and allowing static GHDL builds on Windows which support simulation, synthesis and libghdl. As you can see, there is no acceptable solution yet.

# Existing tools for automatic discovery of MSYS2 package dependencies (files)

Generating standalone zipfiles from MSYS2 builds is a frequently asked feature. There are several references in MSYS2 docs: [Possibly useful sources of information](https://www.msys2.org/wiki/Distributing/).

In GTKWave's GitHub repo, I added CI (GitHub Actions). Upon a user's demand, I added a manual packaging script. He provided the lists of files, which I added to `*.inc` files (see https://github.com/gtkwave/gtkwave/tree/master/MSYS2). A shell script gathers the files and generates tarballs: [standalone_pkg.sh](https://github.com/gtkwave/gtkwave/blob/master/MSYS2/standalone_pkg.sh).

In MSYS2's package management plumbing, dependencies are defined explicitly. Those are found in the `depends` variable of PKGBUILD files, but can also be retrieved through `pacman`: [Finding dependencies of a package](https://www.msys2.org/docs/package-management/#finding-dependencies-of-a-package). Then, you can [Listing the content of a package](https://www.msys2.org/docs/package-management/#listing-the-content-of-a-package).

Maintaining lists of files as done in GTKWave feels cumbersome, so I'd like to avoid it. However, it seems that getting everything from package names would include many unnecessary files. I think we need an scriptable equivalent to [lucasg/Dependencies](https://github.com/lucasg/Dependencies).

I didn't look into [Styrene](https://github.com/achadwick/styrene)'s codebase, so I don't know whether it does anything different from using pacman.

# Dependency on libgnat

In the current CI jobs on GitHub Actions, GHDL is built and tested in the same environment. Build dependencies are not removed before installing the package. Therefore, tests are executed with both build and runtime dependencies available. As a result of that, we had overlooked that GHDL binaries do not depend on `*-gcc-ada`, but `libghdl*.dll` does depend on it. This is coherent with Patrick's description above.

In #1559, I split build and test jobs on Windows. That allowed me to catch the dependency on `*-gcc-ada`; otherwise, `pyunit` test would crash. That's why `depends` fields of PKGBUILD recipes are updated in #1559. BTW, if you are ok with it, I think we can merge it, as it provides a more realistic testing environment.

[... cut ...]
Overall, guessing how to build GHDL statically with LLVM backend is interesting because MSYS2 is being ported to ARM. Therefore, it might solve two targets at once: Windows x86_64 and aarch64.

> I think the solution is to extend the make script for target `install` so also copy these 3 dlls into `prefix/lib`.

@Paebbels, this is not correct. `make install` is used on many platform and it is used on MSYS2 builds where copying the DLLs is undesired. It needs to be a separated step, which the user needs to call explicitly. Say `make install-with-dlls`.

tgingold, 2020-12-31:
> Generating an static and portable build implies that the absolute location of GHDL cannot be guaranteed. Therefore, setting GHDL_PREFIX is required. See open-tool-forge/fpga-toolchain#59.

I think this is not really correct.  If the location of the libraries are relative to the executable, you don't need to set GHDL_PREFIX.

[...]

> Dependency on libgnat

[..]

> @tgingold, I'm curious about why GHDL does not depend on *-gcc-ada (for simulation), but libghdl*.dll does need it. Is it because LDFLAGS="-static" and --with-llvm-config="llvm-config --link-static" passed to configure affect the build of the binary only?

Yes, that's because it is statically compiled.  We could also link libghdl with libgnat (at least on windows) but there would be some remaining DLL: libgcc_eh and maybe libwinpthread.

umarcor, 2020-12-31:
> I think this is not really correct. If the location of the libraries are relative to the executable, you don't need to set GHDL_PREFIX. [...]

You are correct. It's not required on Windows, but on Linux or macOS only: https://github.com/open-tool-forge/fpga-toolchain#using-ghdl.

You proposed a solution already, tho: https://github.com/open-tool-forge/fpga-toolchain/pull/59#issuecomment-708907036

> We could also link libghdl with libgnat (at least on windows) but there would be some remaining DLL: libgcc_eh and maybe libwinpthread.

Would that avoid the `libghdl.link` issue (#1570)? As far as I understand, if libgnat is linked into libghdl, Yosys would not require `libghdl.link`, because using libghdl alone would suffice (just as the binary). If that is the case, I think it'd be desirable.

tgingold, 2021-01-01:
> Would that avoid the libghdl.link issue (#1570)? As far as I understand, if libgnat is linked into libghdl, Yosys would not require libghdl.link, because using libghdl alone would suffice (just as the binary). If that is the case, I think it'd be desirable.

I don't think so.  libghdl.link is needed for linking with libghdl.a, which is the case here.

umarcor, 2021-01-01:
This is the content of `libghdl.link` in my MINGW64 setup:

```
# cat /mingw64/lib/libghdl.link
D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.2.0/adalib/libgnat.a
```

As commented in #1570, that will be changed to: `/mingw64/lib/gcc/x86_64-w64-mingw32/10.2.0/adalib/libgnat.a`. Yet the point is that it contains a single line, pointing to `libgnat.a`. If `libnat.a` was built into `libghdl.a`, wouldn't the link file be empty?

tgingold, 2021-01-01:
You cannot really add an archive to an archive.
So we could remove the dependency of libgnat.dll in libghdl.dll, but not the dependency of libgnat.a in libghdl.a

Paebbels, 2021-01-11:
I extended the list above. Now I'm able to elaborate (link) a simulation with LLVM backend using a GHDL built with MinGW64 and installed outside of MinGW64 in `C:\Tools\GHDL`.

The whole installation has circa 146 MB.

Paebbels, 2025-04-27:
This has been partially implemented for GHDL v4.1.0 and GHDL v5.0.1.

closed 2025-04-27 by Paebbels