ghdl/ghdl #1561: Coverage Collection for Python APIs

Paebbels, 2020-12-30:
As we now have test cases in unit test format and pytest to run them, we can also collect line and branch coverage from Python APIs.

It requires the following Python packages (for testing/CI only):
* `coverage` and
* `pytest-cov`

Currently we achieve 50% coverage. This will increate when LSP tests are also executed.

[image]

The results can be exported as:
* command line output / ascii table
* HTML report
* XML report file ([cobertura format](https://cobertura.github.io/cobertura/))

I would like to visualize coverage results via [Codecov.io](https://codecov.io/).
We get these benefits compared to uploaded a raw HTML uploads to GH-Pages:
* Coverage badge [image]
* Coverage history/graph over time  
  [image]
* Coverage diffs (green = new code, but red parts, not yet covered by tests)  
 [image]
 

Here is an example file visualized from pyVHDLParser: [pyVHDLParser/Blocks/Generic.py:344](https://codecov.io/gh/Paebbels/pyVHDLParser/src/master/pyVHDLParser/Blocks/Generic.py#L244)  
[image]

umarcor, 2020-12-30:
Let's go for it.

umarcor, 2020-12-31:
Ref #1567.

Paebbels, 2021-01-06:
@tgingold can you register GHDL at these services?
* [ ] https://www.codecov.io/
* [ ] https://www.codacy.com/

tgingold, 2021-01-07:
Done for codacy.

tgingold, 2021-01-07:
But no success for codecov.  I don't know how to add an organization.

Paebbels, 2021-01-07:
# For codacy:

Please create a secret for GHDL, so a GH Action can use the token to push coverage results.
* Token name: `CODACY_PROJECT_TOKEN`
* To create the token:
  1. Go to the repositories overview page (should be https://app.codacy.com/gh/ghdl/ghdl/dashboard)
  2. Go to settings => Integrations => Project API
  3. If Project API does not exist, please create this entry
  4. Get the token and set this for the tokens value
 
  [image]

# For codecov
I'm logged in via GitHub and it's synchronizing with my "teams" and "orgs" after login. There are 2 possibilities to go on:
1. try so synchronize again - may take some time
2. check the settings in GitHub where codecov has access to

My login page looks like this:  
[image]

tgingold, 2021-01-07:
Done.

Paebbels, 2021-01-07:
@umarcor I think you can now extend the workflows and take the steps from pyVHDLModel as a starting point: https://github.com/VHDL/pyVHDLModel/blob/main/.github/workflows/Test.yml#L35-L77

umarcor, 2021-01-08:
Done.

umarcor, 2021-01-08:
What's is GHDL's project ID (for codacy shields/badges)?

umarcor, 2021-01-08:
Found it (I had to Join GHDL in codacy). Shields are shown in the README now.

Paebbels, 2021-01-08:
> Found it (I had to Join GHDL in codacy). Shields are shown in the README now.

Ah, now I see it too. Thanks for the hint.

I changed the PyLint checker from Python 2 to Python 3:  
[image]

There are so many checkers active ... GHDL must have a lot of languages and tools in use :).

closed 2021-01-08 by Paebbels

Paebbels, 2021-01-12:
Codecov is now also showing correct data. @umarcor fixed it.